espressif/esp-idf #6475: support for linux-arm64

DJSundog, 2021-01-29:
I have a number of ESP32-based boards that I'd like to develop against using my arm64-based linux systems.

Following the installation instructions for ESP-IDF results in an error that linux-arm64 isn't supported.

Examination of tools/tools.json shows that there are, indeed, no entries for linux-arm64, though the 32-bit linux-armel builds are there.

Is there something preventing linux-arm64 builds from being provided? The number of linux-arm64 systems in use is only going to increase now that Apple is releasing M1-based hardware.

igrr, 2021-01-29:
Hi @DJSundog, we will add arm64 build next time we release the toolchain. I can't give an exact timeline but most likely there will be a release in Q1. We are already tracking this in https://github.com/espressif/esp-idf/issues/6432, so I will close this issue. Please follow the link to that older issue and click "Subscribe" to get notified when the issue is resolved.

closed 2021-01-29 by igrr

axelcordes, 2021-08-28:
Hey, do you now have a timeline for the next release? The feature is highly appreciated.