espressif/esp-idf #6476: esp-idf-tools-setup-2.3.exe install error

gjbang, 2021-01-29:
## Environment

- Development Kit:      [ESP32-Wrover-Kit|ESP32-DevKitC|ESP32-PICO-Kit|ESP32-LyraT|ESP32-LyraTD-MSC|none]
- Kit version (for WroverKit/PicoKit/DevKitC): [v1|v2|v3|v4]
- Module or chip used:  [ESP32-WROOM-32|ESP32-WROOM-32D|ESP32-WROOM-32U|ESP32-WROVER|ESP32-WROVER-I|ESP32-WROVER-B|ESP32-WROVER-IB|ESP32-SOLO-1|ESP32-PICO-D4|ESP32]
- IDF version (run ``git describe --tags`` to find it):
  // v3.2-dev-1148-g96cd3b75c
- Build System:         [Make|CMake|idf.py]
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
  // 1.22.0-80-g6c4433a
- Operating System:     [Windows|Linux|macOS]
- (Windows only) environment type: [MSYS2 mingw32|ESP Command Prompt|Plain Command Prompt|PowerShell].
- Using an IDE?: [No|Yes (please give details)]
- Power Supply:         [USB|external 5V|external 3.3V|Battery]

## Problem Description

Running esp-idf-tools-setup-2.3.exe fails to complete the installation with exit code -2. For I met the same error like #3938, and [#3938 (comment)](https://github.com/espressif/esp-idf/issues/3938#issuecomment-524183334) cannot help me to solve this problem, I decide to clone release-v4.2 directly  and  "use and existing ESP-IDF directory" in esp-idf-tools-setup-2.3.exe to avoid that. 

Then I get the other problem which seems my python cannot run "idf-tools.py" in eps-idf-v4.2\tools\.

### Expected Behavior

Successfully install.

### Actual Behavior

 Installation has failed with exit code -2

### Steps to reproduce

1. Download esp-idf v4.2 from https://dl.espressif.com/github_assets/espressif/esp-idf/releases/download/v4.2/esp-idf-v4.2.zip and unzip it in some path.
2. Run ESP-IDF Tools 2.3.exe
3. When choose available ESP-IDF versions, I use an existing ESP-IDF directory as step1.
4. Then install with other default setting.

### Code to reproduce this issue

## Debug Logs

```
2021-01-29 14:11:21.420   Log opened. (Time zone: UTC+08:00)
2021-01-29 14:11:21.420   Setup version: Inno Setup version 6.0.2 (u)
2021-01-29 14:11:21.420   Original Setup EXE: D:\GoogleDownload\esp-idf-tools-setup-2.3.exe
2021-01-29 14:11:21.420   Setup command line: /SL5="$3E1160,350580993,721408,D:\GoogleDownload\esp-idf-tools-setup-2.3.exe" 
2021-01-29 14:11:21.420   Windows version: 10.0.19042  (NT platform: Yes)
2021-01-29 14:11:21.420   64-bit Windows: Yes
2021-01-29 14:11:21.420   Processor architecture: x64
2021-01-29 14:11:21.420   User privileges: Administrative
2021-01-29 14:11:21.428   Administrative install mode: No
2021-01-29 14:11:21.428   Install mode root key: HKEY_CURRENT_USER
2021-01-29 14:11:21.428   64-bit install mode: Yes
2021-01-29 14:11:21.435   Created temporary directory: C:\Users\axv\AppData\Local\Temp\is-ESU0O.tmp
2021-01-29 14:11:21.440   -- DLL function import --
2021-01-29 14:11:21.441   Function name: idpAddFile
2021-01-29 14:11:21.441   DLL name: files:idp.dll
2021-01-29 14:11:21.441   Extracting temporary file: C:\Users\axv\AppData\Local\Temp\is-ESU0O.tmp\idp.dll
2021-01-29 14:11:21.451   Dest DLL name: C:\Users\axv\AppData\Local\Temp\is-ESU0O.tmp\idp.dll
2021-01-29 14:11:21.451   Importing the DLL function.
2021-01-29 14:11:21.454   Successfully imported the DLL function. Delay loaded? No
2021-01-29 14:11:21.454   -- DLL function import --
2021-01-29 14:11:21.454   Function name: idpAddFileComp
2021-01-29 14:11:21.454   DLL name: files:idp.dll
2021-01-29 14:11:21.455   Dest DLL name: C:\Users\axv\AppData\Local\Temp\is-ESU0O.tmp\idp.dll
2021-01-29 14:11:21.455   Importing the DLL function.
2021-01-29 14:11:21.455   Successfully imported the DLL function. Delay loaded? No
2021-01-29 14:11:21.455   -- DLL function import --
2021-01-29 14:11:21.455   Function name: idpAddMirror
2021-01-29 14:11:21.455   DLL name: files:idp.dll
2021-01-29 14:11:21.455   Dest DLL name: C:\Users\axv\AppData\Local\Temp\is-ESU0O.tmp\idp.dll
[... 359 lines cut ...]
2021-01-29 14:12:42.013   Installation process succeeded.
2021-01-29 14:12:42.017   Setting PATH for this process: D:\CodeSoft\Python37\;D:\CodeSoft\git\Git\cmd\;D:\CodeSoft\JDK\bin;D:\CodeSoft\JDK\jre\bin;D:\CodeSoft\Python37\Scripts\;D:\CodeSoft\Python37\;C:\Windows\system32;C:\Windows;C:\Windows\System32\Wbem;C:\Windows\System32\WindowsPowerShell\v1.0\;C:\Windows\System32\OpenSSH\;C:\Program Files\Microsoft VS Code\bin;C:\Program Files\Microsoft\Web Platform Installer\;C:\Program Files (x86)\Microsoft ASP.NET\ASP.NET Web Pages\v1.0\;C:\Program Files\Microsoft SQL Server\110\Tools\Binn\;C:\Program Files\Microsoft SQL Server\130\Tools\Binn\;C:\Program Files\Microsoft SQL Server\Client SDK\ODBC\170\Tools\Binn\;C:\Program Files\Microsoft SQL Server\Client SDK\ODBC\110\Tools\Binn\;C:\Program Files (x86)\Microsoft SQL Server\120\Tools\Binn\;C:\Program Files\Microsoft SQL Server\120\Tools\Binn\;C:\Program Files\Microsoft SQL Server\120\DTS\Binn\;C:\Program Files (x86)\Windows Kits\8.1\Windows Performance Toolkit\;D:\CodeSoft\git\Git\cmd;C:\Program Files\Common Files\Autodesk Shared\;C:\Program Files\NVIDIA Corporation\NVIDIA NvDLISR;C:\Program Files (x86)\NVIDIA Corporation\PhysX\Common;C:\Program Files (x86)\WinMerge;C:\Go\bin;C:\Program Files\dotnet\;C:\Program Files\MySQL\MySQL Shell 8.0\bin\;C:\Users\axv\AppData\Local\Microsoft\WindowsApps;D:\CodeSoft\Texlive\texlive\2020\bin\win32;C:\Users\axv\go\bin;C:\Users\axv\.dotnet\tools
2021-01-29 14:12:42.017   PYTHONPATH=
2021-01-29 14:12:42.018   idf_tools.py exists in IDF directory
2021-01-29 14:12:42.018   idf_tools.py command: D:\CodeSoft\Python37\python.exe "D:\CodeSoft\esp-idf-v4.2\esp-idf-v4.2\tools\idf_tools.py" --idf-path D:\CodeSoft\esp-idf-v4.2\esp-idf-v4.2
2021-01-29 14:12:42.018   Installing tools:D:\CodeSoft\Python37\python.exe "D:\CodeSoft\esp-idf-v4.2\esp-idf-v4.2\tools\idf_tools.py" --idf-path D:\CodeSoft\esp-idf-v4.2\esp-idf-v4.2 install
2021-01-29 14:12:42.052   ProcStart failed
2021-01-29 14:12:42.052   Done, exit code=-2
2021-01-29 14:12:42.052   --------
2021-01-29 14:12:42.052   Running command: D:\CodeSoft\Python37\python.exe "D:\CodeSoft\esp-idf-v4.2\esp-idf-v4.2\tools\idf_tools.py" --idf-path D:\CodeSoft\esp-idf-v4.2\esp-idf-v4.2 install
                          
2021-01-29 14:12:42.052   --------
2021-01-29 14:12:42.053   Message box (OK):
                          Installation has failed with exit code -2
2021-01-29 14:12:45.296   User chose OK.
2021-01-29 14:12:45.314   Message box (Yes/No):
                          Installation log has been created, it may contain more information about the problem.
                          Display the installation log now?
2021-01-29 14:12:46.358   User chose Yes.
2021-01-29 14:12:46.501   CurStepChanged raised an exception.
```

## Other items if possible

I have try the solution in https://www.esp32.com/viewtopic.php?t=12481. Although we have the similar logs, that doesn't work for me.

georgik, 2021-01-29:
@gjbang Thank you for reporting the issue. It's a really interesting failure.
Please, could you try to run the following command in CMD to see the output? For some reason is not in the log:
```
D:\CodeSoft\Python37\python.exe "D:\CodeSoft\esp-idf-v4.2\esp-idf-v4.2\tools\idf_tools.py" --idf-path D:\CodeSoft\esp-idf-v4.2\esp-idf-v4.2 install
```

Other possible solutions:
We're preparing a new version of the installer which has some additional checks and corrections. The main difference is that it contains Embedded Python with prepackaged Python Wheels required by idf-tools. It can help us to verify whether the problem is still there. Downloads:
- online beta version - 13 MB - https://dl.espressif.com/dl/tmp/esp-idf-tools-setup-2.5-online-06-unsigned.exe
- offline beta version - 1 GB - https://dl.espressif.com/dl/tmp/esp-idf-tools-setup-2.5-offline-10-unsigned.exe

Kiznoh, 2021-01-29:
Hi, I noticed this error in the log `Can not find idf.py in D:\CodeSoft\esp-idf-v4.2\tools`.
Can you try to [redownload esp-idf](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/get-started/#id1)?

gjbang, 2021-01-29:
> @gjbang Thank you for reporting the issue. It's a really interesting failure.
> Please, could you try to run the following command in CMD to see the output? For some reason is not in the log:
>
> ``` [...]

Thank you very much! I successfully complete installation using the second solution. I downloaded the offline beta version, and choose the v4.2 release. It works and there is not any error. Very good work!
As for running that command, both CMD and powershell output nothing. I think that command has been excuted. Maybe there is some problems with my computer configuration.

gjbang, 2021-01-29:
> Hi, I noticed this error in the log `Can not find idf.py in D:\CodeSoft\esp-idf-v4.2\tools`.
> Can you try to [redownload esp-idf](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/get-started/#id1)?

Thank you! I check the directory, and idf.py is in '........\tools'. I think the error `Can not find idf.py in D:\CodeSoft\esp-idf-v4.2\tools` is caused by improper operation, which happened in the process of choosing the release-version, but not in the real installation. I think the key point is at the end of log. Although I redownload esp-idf and retry, setup program still gives the same error log. The new log is as followed:
`2021-01-29 23:46:06.261   Log opened. (Time zone: UTC+08:00)
2021-01-29 23:46:06.261   Setup version: Inno Setup version 6.0.2 (u)
2021-01-29 23:46:06.261   Original Setup EXE: D:\GoogleDownload\esp-idf-tools-setup-2.3.exe
2021-01-29 23:46:06.261   Setup command line: /SL5="$170E32,350580993,721408,D:\GoogleDownload\esp-idf-tools-setup-2.3.exe" 
2021-01-29 23:46:06.261   Windows version: 10.0.19042  (NT platform: Yes)
2021-01-29 23:46:06.261   64-bit Windows: Yes
2021-01-29 23:46:06.261   Processor architecture: x64
2021-01-29 23:46:06.261   User privileges: None
2021-01-29 23:46:06.263   Administrative install mode: No
2021-01-29 23:46:06.263   Install mode root key: HKEY_CURRENT_USER
2021-01-29 23:46:06.263   64-bit install mode: Yes
2021-01-29 23:46:06.270   Created temporary directory: C:\Users\axv\AppData\Local\Temp\is-LLF61.tmp
2021-01-29 23:46:06.275   -- DLL function import --
2021-01-29 23:46:06.275   Function name: idpAddFile
2021-01-29 23:46:06.275   DLL name: files:idp.dll
2021-01-29 23:46:06.276   Extracting temporary file: C:\Users\axv\AppData\Local\Temp\is-LLF61.tmp\idp.dll
2021-01-29 23:46:06.286   Dest DLL name: C:\Users\axv\AppData\Local\Temp\is-LLF61.tmp\idp.dll
2021-01-29 23:46:06.286   Importing the DLL function.
2021-01-29 23:46:06.290   Successfully imported the DLL function. Delay loaded? No
2021-01-29 23:46:06.290   -- DLL function import --
2021-01-29 23:46:06.290   Function name: idpAddFileComp
2021-01-29 23:46:06.290   DLL name: files:idp.dll
2021-01-29 23:46:06.290   Dest DLL name: C:\Users\axv\AppData\Local\Temp\is-LLF61.tmp\idp.dll
2021-01-29 23:46:06.290   Importing the DLL function.
2021-01-29 23:46:06.290   Successfully imported the DLL function. Delay loaded? No
2021-01-29 23:46:06.290   -- DLL function import --
2021-01-29 23:46:06.290   Function name: idpAddMirror
2021-01-29 23:46:06.290   DLL name: files:idp.dll
2021-01-29 23:46:06.290   Dest DLL name: C:\Users\axv\AppData\Local\Temp\is-LLF61.tmp\idp.dll
2021-01-29 23:46:06.290   Importing the DLL function.
2021-01-29 23:46:06.290   Successfully imported the DLL function. Delay loaded? No
2021-01-29 23:46:06.290   -- DLL function import --
2021-01-29 23:46:06.290   Function name: idpAddFtpDir
2021-01-29 23:46:06.290   DLL name: files:idp.dll
2021-01-29 23:46:06.291   Dest DLL name: C:\Users\axv\AppData\Local\Temp\is-LLF61.tmp\idp.dll
2021-01-29 23:46:06.291   Importing the DLL function.
2021-01-29 23:46:06.291   Successfully imported the DLL function. Delay loaded? No
2021-01-29 23:46:06.291   -- DLL function import --
2021-01-29 23:46:06.291   Function name: idpAddFtpDirComp
2021-01-29 23:46:06.291   DLL name: files:idp.dll
2021-01-29 23:46:06.291   Dest DLL name: C:\Users\axv\AppData\Local\Temp\is-LLF61.tmp\idp.dll
2021-01-29 23:46:06.291   Importing the DLL function.
2021-01-29 23:46:06.291   Successfully imported the DLL function. Delay loaded? No
2021-01-29 23:46:06.291   -- DLL function import --
2021-01-29 23:46:06.291   Function name: idpClearFiles
2021-01-29 23:46:06.291   DLL name: files:idp.dll
2021-01-29 23:46:06.291   Dest DLL name: C:\Users\axv\AppData\Local\Temp\is-LLF61.tmp\idp.dll
2021-01-29 23:46:06.291   Importing the DLL function.
2021-01-29 23:46:06.291   Successfully imported the DLL function. Delay loaded? No
2021-01-29 23:46:06.291   -- DLL function import --
2021-01-29 23:46:06.291   Function name: idpFilesCount
2021-01-29 23:46:06.291   DLL name: files:idp.dll
2021-01-29 23:46:06.291   Dest DLL name: C:\Users\axv\AppData\Local\Temp\is-LLF61.tmp\idp.dll
2021-01-29 23:46:06.291   Importing the DLL function.
2021-01-29 23:46:06.291   Successfully imported the DLL function. Delay loaded? No
2021-01-29 23:46:06.291   -- DLL function import --
2021-01-29 23:46:06.291   Function name: idpFtpDirsCount
2021-01-29 23:46:06.291   DLL name: files:idp.dll
2021-01-29 23:46:06.291   Dest DLL name: C:\Users\axv\AppData\Local\Temp\is-LLF61.tmp\idp.dll
2021-01-29 23:46:06.291   Importing the DLL function.
2021-01-29 23:46:06.291   Successfully imported the DLL function. Delay loaded? No
2021-01-29 23:46:06.291   -- DLL function import --
2021-01-29 23:46:06.291   Function name: idpFileDownloaded
2021-01-29 23:46:06.291   DLL name: files:idp.dll
2021-01-29 23:46:06.292   Dest DLL name: C:\Users\axv\AppData\Local\Temp\is-LLF61.tmp\idp.dll
2021-01-29 23:46:06.292   Importing the DLL function.
[... 394 lines cut ...]
2021-01-29 23:46:55.693   Creating new uninstall key: HKEY_CURRENT_USER\Software\Microsoft\Windows\CurrentVersion\Uninstall\{9E068D99-5C4B-4E5F-96A3-B17CF291E6BD}_is1
2021-01-29 23:46:55.693   Writing uninstall key values.
2021-01-29 23:46:55.694   Detected previous administrative 64-bit install? No
2021-01-29 23:46:55.694   Detected previous administrative 32-bit install? No
2021-01-29 23:46:55.743   Installation process succeeded.
[... cut ...]
2021-01-29 23:46:55.770   --------
2021-01-29 23:46:55.770   Message box (OK):
                          Installation has failed with exit code -2
2021-01-29 23:46:58.521   User chose OK.
2021-01-29 23:46:58.545   Message box (Yes/No):
                          Installation log has been created, it may contain more information about the problem.
                          Display the installation log now?
2021-01-29 23:46:59.611   User chose Yes.
2021-01-29 23:47:00.013   CurStepChanged raised an exception.
`

gjbang, 2021-01-31:
> @gjbang Thank you for reporting the issue. It's a really interesting failure.
> Please, could you try to run the following command in CMD to see the output? For some reason is not in the log:
>
> ``` [...]

Hi! I know why this error happened. It was indeed caused by my computer's configuration. For using different versions of python, I renamed the `python.exe` with `python3.exe` in the python's install directory(original executable named  `python.exe`), which can be distinguished from python2.7 in the CMD and powershell in this way. So although esp-idf-tools-setup-2.3.exe successfully recognized the right install path of my python3.7, it didn't find the right `exe` of python, and it still thought the executable program is `python.exe`, while the real exe has been renamed `python3.exe` by me manually. Then every command about `python.exe` in the installer won't be executed at all.

So I suggest you should give user the right to select the real python's executable program on the step named 'Python choice', but not just recognize the `Available Python versions` automatically and not give user the right to change the target real python's executable program. 

OR just use the new installer version which you have offered me to use, and it seems that this installer recognize the real python executable program automatically, so I think that it is a really great work.

At last, I'm sorry for wasting your time to solve the problem which is caused by my unconventional and stupid operation. I sincerely apologies.

georgik, 2021-02-01:
@gjbang Thank you very much for the explanation of the root cause and the test. It's very valuable input.

The handling of python.exe on Windows is very tricky because there are several different files with the name python.exe which has a different purpose. There are also some not-so-obvious things like that python.exe in virtualenv depends on the main python which created it and several other gotchas.

We're moving the new installer closer to the production state.

closed 2021-02-01 by gjbang

georgik, 2021-02-02:
Update: New installers are available
- Online installer (30 MB) : https://dl.espressif.com/dl/esp-idf-tools-setup-online-2.5.exe
- Offline installer - contains v4.2 and v4.1.1 (810 MB): https://dl.espressif.com/dl/esp-idf-tools-setup-offline-2.5.exe

gjbang, 2021-02-03:
> Update: New installers are available
> 
> * Online installer (30 MB) : https://dl.espressif.com/dl/esp-idf-tools-setup-online-2.5.exe
> * Offline installer - contains v4.2 and v4.1.1 (810 MB): https://dl.espressif.com/dl/esp-idf-tools-setup-offline-2.5.exe

Thank you for the update! Still great work! I have retried using this new installer and it still works well.
One small question: I read the lastest [document](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/get-started/windows-setup.html#get-started-windows-tools-installer) and find the installer has been updated. The version given in the doc is 2.4  while it's 2.5 in this comment,  and the fact that they are the same installer can be verified by  the hash value of sha256. So......what's the real version? I mean if real version is 2.5, then the online doc may need to be modified. Besides, when I open the installer downloaded from link given in online document, the installer's main title is `Setup - ESP-IDF Tools 2.5` though its filename is `esp-idf-tools-setup-2.4.exe`.

georgik, 2021-02-03:
@gjbang Thank you for the test.
Regarding 2.4 this is typo in the documentation. The link should point to 2.5. The correction is waiting to be merge.