espressif/esp-idf #6481: timer_isr_register() crash after esp_restart()

moggiozzi, 2021-01-30:
## Environment
- Development Kit:      ESP32-DevKit V1
- Kit version (for WroverKit/PicoKit/DevKitC): v1
- Module or chip used:  ESP32-WROOM-32
- IDF version (run ``git describe --tags`` to find it): v4.2-dev-1303-gcd4fa46f5
- Build System:         idf.py
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it): xtensa-esp32-elf-gcc (crosstool-NG esp-2020r1) 8.2.0
- Operating System:     Windows
- (Windows only) environment type: ESP Command Prompt
- Using an IDE?: No
- Power Supply:         USB

## Problem Description
timer_isr_register() crash after every second reboot by esp_restart().

d:\work\sources\esp32_spp_ledc\bt_spp_acceptor\build>xtensa-esp32-elf-addr2line -e bt_spp_acceptor_demo.elf 0x40091708:0x3ffbe350 0x40114371:0x3ffbe370 0x40082a30:0x3ffbe3a0 0x400825e1:0x3ffbe3c0 0x4000bfed:0x3ffbb400 0x40091721:0x3ffbb410 0x400d32b3:0x3ffbb430 0x40113e3d:0x3ffbb470 0x400d4669:0x3ffbb4b0 0x400d480b:0x3ffbb4f0 0x400d4518:0x3ffbb520 0x400d2b6a:0x3ffbb590 0x400914c1:0x3ffbb5c0
C:/esp32_4_1/esp-idf/components/freertos/xtensa/port.c:410
C:/esp32_4_1/esp-idf/components/freertos/xtensa/include/freertos/portmacro.h:272
d:\work\sources\esp32_spp_ledc\bt_spp_acceptor\build/../main/test.c:96
C:/esp32_4_1/esp-idf/components/freertos/xtensa/xtensa_vectors.S:1105
??:0
C:/esp32_4_1/esp-idf/components/freertos/xtensa/port.c:419
C:/esp32_4_1/esp-idf/components/esp32/intr_alloc.c:673
C:/esp32_4_1/esp-idf/components/driver/timer.c:285
d:\work\sources\esp32_spp_ledc\bt_spp_acceptor\build/../main/test.c:127 (discriminator 9)
d:\work\sources\esp32_spp_ledc\bt_spp_acceptor\build/../main/test.c:166
d:\work\sources\esp32_spp_ledc\bt_spp_acceptor\build/../main/example_spp_acceptor_demo.c:220
C:/esp32_4_1/esp-idf/components/esp32/cpu_start.c:580
C:/esp32_4_1/esp-idf/components/freertos/xtensa/port.c:143

d:\work\sources\esp32_spp_ledc\bt_spp_acceptor\build>xtensa-esp32-elf-addr2line -e bt_spp_acceptor_demo.elf 0x401338bb:0x3ffbc420 0x400d392f:0x3ffbc440 0x40091e49:0x3ffbc460 0x400914c1:0x3ffbc480
C:/esp32_4_1/esp-idf/components/esp32/pm_esp32.c:484
C:/esp32_4_1/esp-idf/components/esp_common/src/freertos_hooks.c:63
C:/esp32_4_1/esp-idf/components/freertos/tasks.c:3385 (discriminator 1)
C:/esp32_4_1/esp-idf/components/freertos/xtensa/port.c:143
### Expected Behavior
timer_isr_register NO crash system
### Actual Behavior
timer_isr_register crash system
### Steps to reproduce
run project from https://github.com/moggiozzi/esp32_spp_ledc

## Debug Logs

```
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:4
load:0x3fff0034,len:6168
load:0x40078000,len:13804
load:0x40080400,len:4000
entry 0x40080688
I (29) boot: ESP-IDF v4.2-dev-1303-gcd4fa46f5 2nd stage bootloader
I (29) boot: compile time 09:42:48
I (29) boot: chip revision: 1
I (30) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (36) boot.esp32: SPI Speed      : 40MHz
I (40) boot.esp32: SPI Mode       : DIO
I (44) boot.esp32: SPI Flash Size : 4MB
I (47) boot: Enabling RNG early entropy source...
I (52) boot: Partition Table:
I (54) boot: ## Label            Usage          Type ST Offset   Length
I (61) boot:  0 nvs              WiFi data        01 02 00009000 00004000
I (67) boot:  1 otadata          OTA data         01 00 0000d000 00002000
I (74) boot:  2 phy_init         RF data          01 01 0000f000 00001000
I (80) boot:  3 ota_0            OTA app          00 10 00010000 000f0000
I (87) boot:  4 ota_1            OTA app          00 11 00100000 000f0000
I (93) boot:  5 rawdata          Unknown data     01 ff 001f0000 00010000
[... 125 lines cut ...]
Guru Meditation Error: Core  0 panic'ed (Interrupt wdt timeout on CPU0). 

Core  0 register dump:
PC      : 0x4009170b  PS      : 0x00060034  A0      : 0x80114374  A1      : 0x3ffbe350  
A2      : 0x3ffca1e8  A3      : 0x3ffba0b8  A4      : 0x00000000  A5      : 0x00000001  
A6      : 0x0000cdcd  A7      : 0x00000000  A8      : 0x00000000  A9      : 0x00060023  
A10     : 0x0000cdcd  A11     : 0x00060023  A12     : 0x00060023  A13     : 0x00060023  
A14     : 0x00000001  A15     : 0x3ffba0d4  SAR     : 0x00000013  EXCCAUSE: 0x00000005  
EXCVADDR: 0x00000000  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xfffffffc  
Core  0 was running in ISR context:
EPC1    : 0x40133cef  EPC2    : 0x00000000  EPC3    : 0x00000000  EPC4    : 0x4009170b

Backtrace:0x40091708:0x3ffbe350 0x40114371:0x3ffbe370 0x40082a30:0x3ffbe3a0 0x400825e1:0x3ffbe3c0 0x4000bfed:0x3ffbb400 0x40091721:0x3ffbb410 0x400d32b3:0x3ffbb430 0x40113e3d:0x3ffbb470 0x400d4669:0x3ffbb4b0 0x400d480b:0x3ffbb4f0 0x400d4518:0x3ffbb520 0x400d2b6a:0x3ffbb590 0x400914c1:0x3ffbb5c0
Core  1 register dump:
PC      : 0x401338be  PS      : 0x00060b34  A0      : 0x800d3932  A1      : 0x3ffbc420  
A2      : 0x00000000  A3      : 0x00000000  A4      : 0x00000001  A5      : 0x80000001  
A6      : 0x00000003  A7      : 0x00060023  A8      : 0x800d373e  A9      : 0x3ffbc3f0  
A10     : 0x00000000  A11     : 0x00060623  A12     : 0x00060620  A13     : 0x00060623  
A14     : 0x00000001  A15     : 0x00000000  SAR     : 0x00000000  EXCCAUSE: 0x00000005  
EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000  

Backtrace:0x401338bb:0x3ffbc420 0x400d392f:0x3ffbc440 0x40091e49:0x3ffbc460 0x400914c1:0x3ffbc480

ELF file SHA256: 8305d7f0698a82d1
```

Wth-Esp, 2021-02-01:
@moggiozzi Please provide ELF file and sdkconfig. Thanks

moggiozzi, 2021-02-01:
> @moggiozzi Please provide ELF file and sdkconfig. Thanks
@Wth-Esp, *.elf file commited.

moggiozzi, 2021-06-03:
My workaround: call esp_digital_reset() instead esp_restart()
```
using static __attribute__((noreturn)) void esp_digital_reset(void)
{
    rtc_clk_cpu_freq_set_xtal();
#if CONFIG_IDF_TARGET_ESP32
    esp_cpu_unstall(PRO_CPU_NUM);
#endif
    // reset the digital part
    SET_PERI_REG_MASK(RTC_CNTL_OPTIONS0_REG, RTC_CNTL_SW_SYS_RST);
    while (true) {
        ;
    }
}
```
It is correct way? Can this lead to some bad results?

moggiozzi, 2021-07-08:
Exception not occur if call before reset:
timer_isr_callback_remove();
timer_deinit();

closed 2021-07-08 by moggiozzi