espressif/esp-idf #6482: MQTT over TLS fails for self signed certificate in a local broker

mntolia, 2021-01-30:
## Environment

- Development Kit:      ESP32-Wrover-Kit
- Module or chip used:  ESP32-WROOM-32
- IDF version : v4.0 (Using Arduino Core)
- Operating System:    Windows

## Problem Description

When I try to connect to my local broker which uses a self signed certificate, the mqtt client fails to connect to it. I checked the same configuration with a GUI client and it works fine as long as I don't verify the ssl certificates. 

Here is my MQTT configuration:

### Code to reproduce this issue

```cpp
int32_t setup_mqtt_client(mqtt_event_callback_t mqtt_event_handler_cb)
{
    ESP_LOGI(TAG, "Connecting to MQTT Broker");
    esp_mqtt_client_config_t mqtt_config = {0};
    mqtt_config.uri = "mqtts://vpnpi.local";
    mqtt_config.port = 8883;
    mqtt_config.cert_pem = cert_content;
    mqtt_config.disable_clean_session = 1;
    mqtt_config.transport = MQTT_TRANSPORT_OVER_SSL;
    mqtt_config.event_handle = mqtt_event_handler_cb;
    mqtt_config.buffer_size = 1024 * 50; //50 kb recieve buffer size
    ESP_LOGI(TAG, "MQTT Client Initilization");
    esp_mqtt_client_handle_t client = esp_mqtt_client_init(&mqtt_config);
    return esp_mqtt_client_start(client);
}
```

## Debug Logs

```
␛[0;32mI (17868) mqtt: Other event id:7␛[0m
␛[0;31mE (17868) esp-tls: couldn't get hostname for :vpnpi.local:␛[0m
␛[0;31mE (17868) esp-tls: Failed to open new connection␛[0m
␛[0;31mE (17868) TRANS_SSL: Failed to open a new connection␛[0m
␛[0;31mE (17878) MQTT_CLIENT: Error transport connect␛[0m
␛[0;32mI (17888) mqtt: MQTT_EVENT_ERROR␛[0m
␛[0;32mI (17888) mqtt: MQTT_EVENT_DISCONNECTED␛[0m
```

Is there a config option to disable ssl verification?

david-cermak, 2021-02-01:
Hi @mntolia 

This doesn't look like a certificate problem. Based on the logs above, the host name `vpnpi.local` didn't get resolved correctly. if it's a local link name, this is unfortunately not supported in `v4.0` yet. You can either
* migrate to IDF v4.1 or later
* or manually set `LWIP_DNS_SUPPORT_MDNS_QUERIES` to true (in the source code or using the build system)

mntolia, 2021-02-01:
Thanks @david-cermak . If Arduino core supported for IDF 4.1 or later?

david-cermak, 2021-02-02:
@mntolia  I'm not an expert on arduino. Based on the release page
https://github.com/espressif/arduino-esp32/releases
it uses `v3.3`, but perphas there are some ways to use IDF as a component, as described here https://github.com/espressif/arduino-esp32/blob/master/docs/esp-idf_component.md#to-use-as-a-component-of-esp-idf

david-cermak, 2021-02-22:
@mntolia Any update on using the config option or switching to more recent IDF? (`v4.1` or later)

Could you please close the issue if its resolved from IDF side?

closed 2021-02-22 by mntolia