espressif/esp-idf #6484: Ethernet Basic Example not working with W5500

ntwallace, 2021-02-01:
## Environment
- Module or chip used:  ESP32-WROVER-IE 16MB
- IDF version (run ``git describe --tags`` to find it):
    v4.3-dev-2586-g526f68239
- Build System:        CMake
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
    xtensa-esp32-elf-gcc (crosstool-NG esp-2020r3) 8.4.0
- Operating System:    macOS
- (Windows only) environment type: Plain Command Prompt
- Using an IDE?: VSCode
- Power Supply: USB

## Problem Description

I cannot get the W5500 to work successfully using the basic ethernet example. Everything installs/starts correctly, I get an Ethernet link up message and then it fails to read the registers in the mac file. As a result, I never get an IP address and cannot ping the device.

Using VSPI pins:
- MOSI - 23
- MISO - 19
- SCLK - 18
- CS - 5
- INT - 39

### Code to reproduce this issue

Using basic ethernet example found [here](https://github.com/espressif/esp-idf/blob/master/examples/ethernet/basic/main/ethernet_example_main.c)

```

## Debug Logs

```cpp
I (0) cpu_start: Starting scheduler on APP CPU.
I (916) w5500-mac: version=4
I (928) esp_eth.netif.glue: 02:00:00:12:34:56
I (928) esp_eth.netif.glue: ethernet attached to netif
I (938) eth_example: Ethernet Started
E (3601) w5500-mac: w5500_get_tx_free_size(139): read TX FSR failed
E (3601) w5500-mac: emac_w5500_transmit(509): get free size failed
I (3603) eth_example: Ethernet Link Up
I (3608) eth_example: Ethernet HW Addr 02:00:00:12:34:56
E (3673) w5500-mac: w5500_get_tx_free_size(139): read TX FSR failed
E (3673) w5500-mac: emac_w5500_transmit(509): get free size failed
E (4464) w5500-mac: w5500_get_tx_free_size(139): read TX FSR failed
E (4464) w5500-mac: emac_w5500_transmit(509): get free size failed
E (6477) w5500-mac: w5500_get_tx_free_size(139): read TX FSR failed
E (6477) w5500-mac: emac_w5500_transmit(509): get free size failed
E (10476) w5500-mac: w5500_get_tx_free_size(139): read TX FSR failed
E (10476) w5500-mac: emac_w5500_transmit(509): get free size failed
```

## Other items if possible

- [x ] sdkconfig file (attach the sdkconfig file from your project folder)
[sdkconfig.txt](https://github.com/espressif/esp-idf/files/5901502/sdkconfig.txt)

suda-morris, 2021-02-01:
I tried just now with W5500+ESP32-S2+default configuration, that example still works. Can you slow down the SPI communication speed? Have you tried with other GPIO?

felmue, 2021-02-01:
Hello @ntwallace 

is the `sdkconfig.txt` file you've attached really the one you are using? I am asking because the W5500 doesn't seem to be enabled at all:
```
#
# Example Configuration
#
CONFIG_EXAMPLE_USE_INTERNAL_ETHERNET=y
# CONFIG_EXAMPLE_USE_DM9051 is not set
# CONFIG_EXAMPLE_USE_W5500 is not set
CONFIG_EXAMPLE_ETH_PHY_IP101=y
# CONFIG_EXAMPLE_ETH_PHY_RTL8201 is not set
# CONFIG_EXAMPLE_ETH_PHY_LAN8720 is not set
# CONFIG_EXAMPLE_ETH_PHY_DP83848 is not set
# CONFIG_EXAMPLE_ETH_PHY_KSZ8041 is not set
CONFIG_EXAMPLE_ETH_MDC_GPIO=23
CONFIG_EXAMPLE_ETH_MDIO_GPIO=18
CONFIG_EXAMPLE_ETH_PHY_RST_GPIO=5
CONFIG_EXAMPLE_ETH_PHY_ADDR=1
# end of Example Configuration
```

And yes, like @suda-morris I've just tried the `basic` example myself and it works for me as well.
Note: I had to lower the SPI bus speed to 20 MHz (default is 36 MHz) else it did not work.

Thanks
Felix

ntwallace, 2021-02-01:
> is the `sdkconfig.txt` file you've attached really the one you are using? I am asking because the W5500 doesn't seem to be enabled at all:

Sorry, attached the wrong sdkconfig! Here's the correct one: [sdkconfig.txt](https://github.com/espressif/esp-idf/files/5904741/sdkconfig.txt)

> Can you slow down the SPI communication speed? Have you tried with other GPIO?

I tried at 20MHz and 16MHz, same issue. Is it possible there's an issue in IDF when using an input only pin (SENSOR_VP) for the interrupt? I'm on a custom board with no ability to change the GPIO except I could jump the interrupt pin to another pin. I'm not using the reset pin at all. The W5500 was previously working fine for me with this GPIO setup in IDF using Arduino as a component and that ethernet library.

felmue, 2021-02-01:
Hello @suda-morris 

ok, thank you. I am using the exact same GPIOs as you do except for CS where I use GPIO4 (GPIO5 is internally used on my board). And it works with a clock speed of 16MHz or 20MHz, but does not with 26MHz.

I've noticed that you've enabled `CONFIG_COMPILER_OPTIMIZATION_SIZE=y`. When I try that myself I get a watchdog error. Have you tried with the default optimization, eg. `CONFIG_COMPILER_OPTIMIZATION_DEFAULT=y`?

Thanks
Felix

ntwallace, 2021-02-01:
> I've noticed that you've enabled `CONFIG_COMPILER_OPTIMIZATION_SIZE=y`. When I try that myself I get a watchdog error. Have you tried with the default optimization, eg. `CONFIG_COMPILER_OPTIMIZATION_DEFAULT=y`?

Wow, nice catch. That fixed it! So it appears the real issue is that the W5500 driver doesn't work at higher optimization levels. Unfortunately in my main codebase I need the -Os (or -O2) flag for other things. @suda-morris any thoughts on how to proceed there?

david-cermak, 2021-05-05:
@ntwallace This issue is a duplicate of https://github.com/espressif/esp-idf/issues/6579 (and also mentions https://github.com/espressif/esp-idf/issues/6233 in you last comment). Both these issues are fixed in these commits: dfe9580, 7af358e. So we will close this, please feel free to reopen or raise another if any trouble with the driver.

closed 2021-05-06 by Alvin1Zhang