espressif/esp-idf #6485: Uart (example project) not work with 9600 baudrate

moggiozzi, 2021-02-01:
## Environment

- Development Kit:     ESP32-DevKit
- Kit version (for WroverKit/PicoKit/DevKitC): v1
- Module or chip used:  ESP32-WROOM-32
- IDF version (run ``git describe --tags`` to find it): v4.2-dev-1303-gcd4fa46f5
- Build System:         idf.py
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it): xtensa-esp32-elf-gcc (crosstool-NG esp-2020r1) 8.2.0
- Operating System:     Windows
- (Windows only) environment type: ESP Command Prompt
- Using an IDE?: No
- Power Supply:         USB

## Problem Description
Project from esp-idf\examples\peripherals\uart\uart_echo_rs485\ not work with 9600 baudrate (with 115200 uart work fine).

### Expected Behavior
Uart communicate with 9600
### Actual Behavior
Uart NOT communicate with 9600
### Steps to reproduce
Run project from https://github.com/moggiozzi/esp_uart_echo_rs485

moggiozzi, 2021-02-01:
Pins connected witch FTDI232 
[image]

closed 2021-02-01 by moggiozzi

moggiozzi, 2021-02-01:
My mistake.

alisitsyn, 2021-02-01:
Hi @moggiozzi,

Does 9600 baud this work now in your design? Just to confirm.

moggiozzi, 2021-02-01:
> Hi @moggiozzi,
> 
> Does 9600 baud work now in your design? Just to confirm.
Yes. But I found other bug: sometimes RTS pin remains in send mode. My workaround: force call uart_set_rts(uartNumber, 1) by timeout if uart driver stopped generate new events.

alisitsyn, 2021-07-12:
@moggiozzi ,
Sorry, I missed your answer. The similar issue is [here](https://github.com/espressif/esp-idf/issues/6728) and uart_wait_tx_done() fix is updated but still waiting for merge. Let me know if you still have issues with RS485 mode.