espressif/esp-idf #6486: Hands-free example shows battery level event instead on ring indication on iPhone

barkow, 2021-02-01:
## Environment

- Development Kit:      none
- Module or chip used:  ESP32-WROOM-32
- IDF version (run ``git describe --tags`` to find it):
    framework-espidf 3.40200.210118 (4.2.0)
- Build System:         CMake (PlatformIO)
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
    toolchain-xtensa32 2.80400.210114 (8.4.0)
- Operating System:     Windows
- (Windows only) environment type: PlatformIO
- Using an IDE?: VSCode/PlatformIO
- Power Supply:         USB

## Problem Description

I have tested the Bluetooth classic hands-free unit example. I want to get a ring indication from my phone. When I connect with my Android phone, I can see the ring indication event. But when I connect with my iPhone, I get battery level indication events, when my phone rings.

### Expected Behavior

Get ring indication event when connected to iPhone and incoming call

### Actual Behavior

Battery level indication event  when connected to iPhone and incoming call

### Steps to reproduce

1. Start hfp_hf example
2. Connect phone to esp32
3. Call phone

### Code to reproduce this issue

HFP Example: https://github.com/espressif/esp-idf/tree/master/examples/bluetooth/bluedroid/classic_bt/hfp_hf

## Debug Logs

### Android 8.0.0:
```
I (1044242) BT_HF: APP HFP event: CONNECTION_STATE_EVT
I (1044242) BT_HF: --connection state connected, peer feats 0x0, chld_feats 0x0
I (1044382) BT_HF: APP HFP event: CALL_IND_EVT
I (1044382) BT_HF: --Call indicator NO call in progress
I (1044382) BT_HF: APP HFP event: CALL_SETUP_IND_EVT
I (1044382) BT_HF: --Call setup indicator NONE
I (1044392) BT_HF: APP HFP event: NETWORK_STATE_EVT
I (1044402) BT_HF: --NETWORK STATE available
I (1044402) BT_HF: APP HFP event: SIGNAL_STRENGTH_IND_EVT
I (1044412) BT_HF: -- signal strength: 2
I (1044412) BT_HF: APP HFP event: ROAMING_STATUS_IND_EVT
I (1044422) BT_HF: --ROAMING: inactive
I (1044422) BT_HF: APP HFP event: BATTERY_LEVEL_IND_EVT
I (1044432) BT_HF: --battery level 5
I (1044432) BT_HF: APP HFP event: CALL_HELD_IND_EVT
I (1044442) BT_HF: --Call held indicator NONE held
I (1044532) BT_HF: APP HFP event: CONNECTION_STATE_EVT
I (1044532) BT_HF: --connection state slc_connected, peer feats 0xfe7, chld_feats 0x3f
I (1070502) BT_HF: APP HFP event: NETWORK_STATE_EVT
I (1070502) BT_HF: --NETWORK STATE available
```
Now, when I call the Android phone, I get this output
```
I (1072982) BT_HF: APP HFP event: RING_IND_EVT
I (1072982) BT_HF: APP HFP event: CLIP_EVT
I (1072982) BT_HF: --clip number ..........
```

### iPhone iOS 14.3

```
I (11612) BT_HF: APP HFP event: CONNECTION_STATE_EVT
I (11612) BT_HF: --connection state connected, peer feats 0x0, chld_feats 0x0
I (12022) BT_HF: APP HFP event: NETWORK_STATE_EVT
I (12022) BT_HF: --NETWORK STATE available
I (12022) BT_HF: APP HFP event: CALL_IND_EVT
I (12032) BT_HF: --Call indicator NO call in progress
I (12032) BT_HF: APP HFP event: CALL_SETUP_IND_EVT
I (12042) BT_HF: --Call setup indicator NONE
I (12042) BT_HF: APP HFP event: BATTERY_LEVEL_IND_EVT 
I (12052) BT_HF: --battery level 4
I (12052) BT_HF: APP HFP event: SIGNAL_STRENGTH_IND_EVT
I (12062) BT_HF: -- signal strength: 3
I (12062) BT_HF: APP HFP event: ROAMING_STATUS_IND_EVT
I (12072) BT_HF: --ROAMING: inactive
I (12082) BT_HF: APP HFP event: CALL_HELD_IND_EVT
I (12082) BT_HF: --Call held indicator NONE held
I (12092) BT_HF: APP HFP event: CONNECTION_STATE_EVT
I (12092) BT_HF: --connection state slc_connected, peer feats 0x3ef, chld_feats 0x3f
I (12102) BT_HF: APP HFP event: INBAND_RING_TONE_EVT
I (12102) BT_HF: --inband ring state Provided
```
When I call my iPhone now, I get a battery level report event. The battery is reported 1 as soon as the phone rings and 0, when the phone stops ringing.
```
I (39192) BT_HF: APP HFP event: BATTERY_LEVEL_IND_EVT
I (39192) BT_HF: --battery level 1
I (41162) BT_HF: APP HFP event: BATTERY_LEVEL_IND_EVT
I (41172) BT_HF: --battery level 0
```
## Other items if possible
n/a

munho-espressif, 2021-02-03:
Hi @barkow 
We used iPhone 7 to test. We didn't repeat your question. The result of packet capture is shown in the figure below. It can be seen that the battery level information is actively sent out by the iphone. The operating system is 12.0.1. We think it may be related to the system version.
[image]

Thanks

munho-espressif, 2021-02-03:
Hi @barkow

We used iPhone 11 pro max  to test. We didn't repeat your question. The operating system is 14.4. 

Thanks

Alvin1Zhang, 2021-02-04:
@barkow Thanks for reporting, however we could not reproduce the issue locally, would you please help share if any updates for the issue? Thanks.

barkow, 2021-02-04:
I'm trying to compile the example with more debug information active to give you more information on my issue. I will update this issue asap.

barkow, 2021-02-05:
I tried to activate more debug information, especially I activated the BTA_HF_CLIENT_AT_DUMP. Unfortunately it results in errors due to a stack overflow in BTU_TASK. I would be happy to get some support to debug the original issue.

Wth-Esp, 2021-02-08:
Hi @barkow 

I'm sorry, as @munho-espressif said, we test with iPhone 11 ProMax with iOS version 14.4, but we cannot reproduce the issue you come up with. We provided the air log, the AT command we send and the response from iPhone is right.

So, could you please update the iOS with your phone and test it again?

Thanks

barkow, 2021-02-17:
I managed to activate BTA_HF_CLIENT_AT_DUMP without crashes by massively increasing the BT stack size and changing the log level of AT command specific reports to warning. Please see the log below. According the initial answer to `AT+CIND?` the incoming call that creates the Battery Level 1 event is based on `+CIEV: 3,1` report from the iPhone. Reading the HFP spec, I'm not sure, if the index is 0 based or one based. Your implementation seems to be 0 based while the data that I see in the log seem to be 1 based. Looking forward to your assessment.

```
W (205180) BT_APPL: new conn_srvc id:27, app_id:1
W (205190) BT_APPL: bta_hf_client_send_at AT+BRSF=255
I (205200) BT_HF: APP HFP event: CONNECTION_STATE_EVT
W (205210) BT_APPL: bta_hf_client_dump_at <cr><lf>+BRSF:1007<cr><lf>
W (205210) BT_APPL: bta_hf_client_dump_at <cr><lf>OK<cr><lf>
W (205210) BT_APPL: bta_hf_client_send_at AT+BAC=1,2
I (205200) BT_HF: --connection state connected, peer feats 0x0, chld_feats 0x0
W (205230) BT_APPL: bta_hf_client_dump_at <cr><lf>OK<cr><lf>
W (205230) BT_APPL: bta_hf_client_send_at AT+CIND=?
W (205260) BT_APPL: bta_hf_client_dump_at <cr><lf>+CIND: ("service",(0-1)),("call",(0-1)),("callsetup",(0-3)),("battchg",(0-5)),("signal",(0-5)),("roam",(0-1)),("callheld",(0-2))<cr><lf><cr><lf>OK<cr><lf>
W (205270) BT_APPL: bta_hf_client_send_at AT+CIND?
W (205300) BT_APPL: bta_hf_client_dump_at <cr><lf>+CIND: 1,0,0,4,4,0,0<cr><lf><cr><lf>OK<cr><lf>
W (205300) BT_APPL: bta_hf_clie       nt_send_at AT+CMER=3,0,0,1
I (205310) BT_HF: APP HFP event: NETWORK_STATE_EVT
I (205310) BT_HF: --NETWORK STATE available
W (205310) BT_APPL: bta_hf_client_dump_at <cr><lf>OK<cr><lf>
W (205320) BT_APPL: bta_hf_client_send_at AT+CHLD=?
I (205330) BT_HF: APP HFP event: CALL_IND_EVT
I (205330) BT_HF: --Call indicator NO call in progress
I (205340) BT_HF: APP HFP event: CALL_SETUP_IND_EVT
I (205340) BT_HF: --Call setup indicator NONE
W (205350) BT_APPL: bt       a_hf_client_dump_at <cr><lf>+CHLD: (0,1,1x,2,2x,3)<cr><lf><cr><lf>OK<cr><lf>
W (205360) BT_APPL: bta_hf_client_send_at AT+BIA=       1,1,1,1,1,1,1,0,0,0,0,0,0,0,0,0,0,0,0,0
I (205370) BT_HF: APP HFP event: BATTERY_LEVEL_IND_EVT
I (205370) BT_HF: --battery level 4
I (205380) BT_HF: APP HFP event: SIGNAL_STRENGTH_IND_EVT
I (205380) BT_HF: -- signal strength: 4
I (205390) BT_HF: APP HFP event: ROAMING_STATUS_IND_EVT
I (205390) BT_HF: --ROAMING: inactive
I (205400) BT_HF: APP HFP event: CALL_HELD_IND_EVT
I (205400) BT_HF: --Call held indicator NONE held
I (205410) BT_HF: APP        HFP event: CONNECTION_STATE_EVT
I (205420) BT_HF: --connection state slc_connected, peer feats 0x3ef, ch       ld_feats 0x3f
I (205420) BT_HF: APP HFP event: INBAND_RING_TONE_EVT
I (205430) BT_HF: --inband ring state Provided
W (205490) BT_APPL: bta_hf_client_dump_at <cr><lf>OK<cr><lf>
W (205490) BT_APPL: bta_hf_client_send_at AT+CCWA=1
W (205630) BT_APPL: bta_hf_client_dump_at <cr><lf>OK<cr><lf>
W (205630) BT_APPL: bta_hf_client_send_at AT+C       MEE=1
W (205760) BT_APPL: bta_hf_client_dump_at <cr><lf>OK<cr><lf>
W (205760) BT_APPL: bta_hf_client       _send_at AT+COPS=3,0
W (205900) BT_APPL: bta_hf_client_dump_at <cr><lf>OK<cr><lf>
W (205900) BT_APPL: bta_hf_client_send_at AT+BTRH?         0m
W (206040) BT_APPL: bta_hf_client_dump_at <cr><lf>+CME ERROR: 4<cr><lf>
W (206040) BT_APPL: bta_hf_client_send_       at AT+CLIP=1
W (206170) BT_APPL: bta_hf_client_dump_at <cr><lf>OK<cr><lf>
W (212360) BT_APPL: bta_hf_client_dump_at <cr><lf>+CIEV: 5,3<cr><lf>
W (224710) BT_APPL: bta_hf_client_dump_at <cr><lf>+CIEV: 3,1<cr><lf><cr><lf>+VGS=10<cr><lf>
I (224710) BT_HF: APP HFP event: BATTERY_LEVEL_IND_EVT
I (224710) BT_HF: --battery level 1
I (224710) BT_HF: APP HFP event: VOLUME_CONTROL_EVT
I (224720) BT_HF: --volume_target: SPEAKER, volume 10
W (224750) BT_APPL: bta_hf_client_dump_at <cr><lf>+BCS:2<cr><lf>
W (224750) BT_APPL: bta_hf_client_send_at AT+BCS=2
W (224830) BT_APPL: bta_hf_client_dump_at <cr><lf>OK<cr><lf>
W (227720) BT_APPL: bta_hf_client_dump_at <cr><lf>+CIEV: 3,0<cr><lf>
I (227720) BT_HF: APP HFP event: BATTERY_LEVEL_IND_EVT
I (227720) BT_HF: --battery level 0
```

xiongweichao, 2023-04-25:
Hi @barkow ,

Sorry to reply you so late. Please try to modify this [line](https://github.com/espressif/esp-idf/blob/df9310ada26123d8d478bcfa203f874d8c21d654/components/bt/host/bluedroid/bta/hf_client/bta_hf_client_at.c#L438) of code to the following.
`realind = bta_hf_client_cb.scb.at_cb.indicator_lookup[index - 1];`

Thanks