espressif/esp-idf #6487: IDF-4.1.1 get_python_env_path( ) return error path

George-Chao, 2021-02-01:
in idf_tools.py, line number : 964

```python

match = re.match(r'^v([0-9]+\.[0-9]+).*', idf_version_str)

```

**if idf_version_str == "v4.1.1" , then match.group(1) will be "4.1", this is wrong.**
the right version is : 

```python

match = re.match(r'^v(\.,[0-9]+).*', idf_version_str)

```

igrr, 2021-02-01:
Hi @George-Chao, it is intended that get_python_env_path function only considers the major and minor components of IDF version. Since all ESP-IDF bugfix releases for the same "major.minor" version use the same python libraries, there is no need to create a separate python environment for each "major.minor.patch" version.

George-Chao, 2021-02-02:
@igrr Thanks. When I installed 4.1.1, two folders appeared under the ".espressif\python_env\" path ( I use windows 10 ), they were **idf4.1_py3.8_env** and **idf4.1.1_py3.8_env**.
Later, I uninstalled 4.1.1, because it is difficult for VS Code to install successfully, just replace it with 4.1.

George-Chao, 2021-02-02:
@igrr 
I tested the 4.1.1 version in the virtual machine today. If you install it with install.bat, **"idf4.1_py3.8_env"** will appear in the ".espressif\python_env\" path.

However, if you install 4.1.1 with the "Espressif IDF" extension of VS Code, **"idf4.1.1_py3.8_env"** will appear, and the installation fails "error: Microsoft Visual C++ 14.0 or greater is required." in building '_openssl' extension.