espressif/esp-idf #6488: Supporting multi-PHY network switches

szekelyisz, 2021-02-01:
**Is your feature request related to a problem? Please describe.**

I'm developing a board that should have Ethernet connectivity with a built-in 3-port switch so devices can be chained. I'm using the KSZ8863RLL actually. Although this chip has an SMI interface on the RMII bus, the management functionality provided through that is limited, so I want to manage it through I2C.

The problem is that ESP-IDF's current infrastructure doesn't seem to accommodate this type of configuration. I started writing a PHY driver but ran into issues. For example, the `init` function should initialize the chip itself, but the `negotiate` call should only restart autonegotiation for one port. But because `eth_phy_config_t` has only one field for the PHY address, it's unclear how to know which port should be involved. If I try to solve this by creating two PHYs with different addresses and assigning them to the same MAC (by calling `esp_eth_driver_install` twice), then both the MAC and the chip will be double-initialized.

**Describe the solution you'd like**

Open to suggestions. I'll be happy to contribute the code, but we need to come up with a solution that integrates nicely into ESP-IDF.

tionebrr, 2021-02-04:
I think it is doable with the current esp-idf/ethernet implementation. You just need to initialize one driver per port. Each port will then have its own handler. Also, it might be required to write a custom MAC to be able to use the MAC filtering functionality of the KSZ directly. The only functionality that is missing in the esp-idf for full control of the KSZ is some kind of switch management, but I don't think enough people need that yet.

tionebrr, 2021-02-04:
Look at esp_eth.c line 163.

tionebrr, 2021-02-04:
@szekelyisz did you actually got it working with I2C management for one port? I'm am stuck on a very weird I2C bug happening after the eth driver install, but the hardware and the stack are working alright (I'm not consistently getting IP via DHCP, but it happens heh :P). After re-reading your post, I see the exact same problems as you. Do you want to collaborate on this?

tionebrr, 2021-02-04:
I took another look at the code and I think it is doable with the current esp-idf by adding some branching variables in the `phy_ksz8863rll_t` struct. All the calls from the other methods are passing by the phy methods.

szekelyisz, 2021-02-04:
@tionebrr Yeah let's make this happen!

I didn't get to the I2C part yet, I gave it a shot but the I2C bus seemed to be empty, it wasn't replying. That might be because I think I read somewhere that it needs the RMII clock to function at all (or handsoldering the chip on the board didn't quite work out the way I wanted). So I started to implement a MAC and a PHY driver for this chip, mostly copy-pasting and adjusting the `emac_esp32` driver, because the MAC is essentially the same, only the SMI needs to be rewritten to I2C. The MAC is ready I think, but I ran into the above issues with the PHY.

I thought about adding state to the struct that you call `phy_ksz8863rll_t`. In the end, without changing other code you'll have to decide if your code does whatever the PHY is asked (for example, initialize), or just return without doing anything to avoid double calls. Because my guess is that double initialization, for example, will cause problems. So how do you decide if a call belongs to the same initialization sequence as the previous one, or is a legitimate call from user code for example? Just ignoring every other call or using a time window would be unsafe and ugly. Do you have any other idea?

I don't think we can get away without changing stuff around [`esp_eth.c`](https://github.com/espressif/esp-idf/blob/master/components/esp_eth/src/esp_eth.c).

tionebrr, 2021-02-05:
@szekelyisz I've found that my I2C issue is actually due to signal integrity. I need to order more ESP32 boards, I don't really trust the one I have here for high speed signals.

You're absolutely right about phy_ksz8863rll_t. I'm mistaken. The state should be stored as an instance of something that represents the chip, and not the port. So we would need some new kind of component, a switch, which would look somewhat like this:
``` C
typedef struct 
{
    bool is_initialized;
    bool port_in_use[2];
    // etc
} esw_ksz8863rll_t; // esw = ethernet switch
```
Then the phy would actually be a port of the esw. And their handlers would look like this:
``` C
typedef struct
{
    esw_ksz8863rll_t *ksz8863rll;
    esw_port_num_t int;
    esp_eth_phy_t parent;
    esp_eth_mediator_t *eth;
    uint32_t addr;
    uint32_t reset_timeout_ms;
    uint32_t autonego_timeout_ms;
    eth_link_t link_status;
    int reset_gpio_num;
} phy_ksz8863rll_port_t;
```
Now in `eth_phy_ksz8863rll.c` you can have branching in methods depending on the state of the chip.
``` C
static esp_err_t ksz8863rll_port_reset(esp_eth_phy_t *phy)
{
    ESP_LOGD(TAG, "resetting");
    phy_ksz8863rll_port_t *ksz8863rll_port = __containerof(phy, phy_ksz8863rll_port_t, parent);
    ksz8863rll_port->link_status = ETH_LINK_DOWN;
    if (phy.esw_ksz8863rll_t->is_initialized)
    {
        ESP_LOGD(TAG, "KSZ8863RLL has already been reset and initialized);
        return ESP_OK; // We do nothing because the chip has already a port working
                                 // But things could still be done in the port control registers
    }
    reset_reg_t reset_reg = {.software_reset = 1};
    PHY_CHECK(ksz8863rll_port_write(ETH_PHY_RESET_ADDR, sizeof(reset_reg), (uint8_t *)&reset_reg) == ESP_OK,
              "write Reset Reg failed", err);
    vTaskDelay(pdMS_TO_TICKS(10));
```
That's just an example tho. We'll need to find exactly how to manage the switch by testing in the long run. But I still think it is doable without touching the existing esp-idf code and logic. We only add new stuff actually.

Also, I'm still have in mind that the KSZ is able to handle layer 2. So each port could also have its own MAC address and work as a separate interface from the ESP point of view. (You could have an admin configuration port on the front and an internet access on the back of a device). I am not using a custom mac file for the moment but will surely try when everything is stable and nice with two ports.

First task for me is to actually get the hardware to work reliably. I was expecting some issues but not that much.
I'll create a repo tomorrow to share the things I've already done.

szekelyisz, 2021-02-06:
Yeah I like the idea of having two separate handles for the two ports. Your suggestion will work in the case of initialization, but you'll have a problem with de-initialization for example. `esp_eth_stop` will call `stop` on the MAC but won't tell you which port is affected. But your mac driver will need this info in order to know whether it should really stop the MAC or just return without doing anything to allow the other port to function. Same for `esp_eth_driver_uninstall`.

Moreover, the only thing `esp_eth_stop` will do as far as we are concerned is call `stop` on the MAC. This is supposed to stop all traffic and shut down that interface, but you can't do that because you might want to continue using the other port, so traffic from the stopping interface will also continue. You need to get to the phy level but `esp_eth_stop` will hide it from you.

tionebrr, 2021-02-07:
That's why we need to hold the KSZ state and methods as another component. 
I thought about another solution in the meantime: The MAC/PHY could actually only be acting on the PORT3, and the PORT1 and 2 are left managed by that new switch component. That way when you call esp_eth_stop is only stops the port attached to the ESP and leave P1 and P2 live their life and pass packets. If you want to stop P1 and P2 you would have to call for a switch stop, which makes more sense to me actually. Still doesn't solves the general problem tho. It might be a good idea to take a look at how esw are implemented in Linux.

tionebrr, 2021-03-13:
@szekelyisz any new thoughts on that? I stopped the development for the last months but I'm starting to work back on this today.

szekelyisz, 2021-03-13:
Yeah same here. I put this on hold for now because I didn't have a good idea how to move on, kinda waiting for the light bulb to turn on above my head. I realized that having two handles for the two ports might not be a good idea because when you just want to send a packet, you want the switch to decide which port to route it to based on its MAC address table, so you need one handle. But there are applications (my use case is RSTP) when you want to force a packet to be sent out on a specific port (possible with tail tagging). But the ESP-IDF APIs don't seem to provide that flexibility.

kostaond, 2021-06-15:
@szekelyisz, @tionebrr  we performed analysis and made the same conclusion as you discussed - there should be separate components to control the KSZ8863RLL. The Ethernet driver philosophy could stay the same as of now. It would only require adding some basic support to control PORT3 "PHY" which acts as counterpart to ESP32 EMAC. The other features like PORT1/2 speed/duplex/autonegotiation, filtering, MAC/VLAN tables management and other switch functionality should be controlled from separate Switch component as shown in attached figure.

There was open question of how to support e.g. Tail tagging for non-IP protocols like RTSP. 
We propose adding a new feature in the ESP-IDF to be users able to receive and send raw Ethernet frames. The user IO infrastructure to such feature would be realized using Virtual File System (VFS) type interface (open, read, write, ioctl, fcntl, select, etc…). It could work as follows: user would open raw Ethernet File Descriptor and configure it with specific Ether type to be filtered via ioctl. All Ethernet frames with that specific Ether type would be filtered by to the new Raw Ethernet VFS component from which user could pick it up by standard read function and perform additional processing based on their specific needs. Such feature would be also particularly handful for future implementation of protocols used at Ethernet layer (IEEE 802.3) such as PTP, for instance. The Ethernet frames could be sent by user application which would resolve the issue with Tail Tagging since the user (the switch application) would be responsible to assemble the whole frame. All other Ether types like IP traffic would be still processed by LwIP as shown in attached figure.
[image]

The Raw Ethernet Interface is planned to be implemented by Espressif. However, note that on the other hand, it is expected that the majority of work related to switch component is to be done by you, by users. Espressif might only implement some basic demonstrator.

Does such solution make you sense? The only drawback might be that Tail tagging would be only available for non-IP traffic but I don’t see any real use case for it since the IP traffic flow should be preferably driven by MAC tables in my opinion. Or do you have any real use case for Tail tagging in IP traffic in your mind?

kostaond, 2021-06-15:
Here's a WIP patch based on IDF 4.3-rc to demonstrate the Raw Ethernet interface. Please note that it is very first proof of concept draft.
file:///home/ondrej/Releases/esp-idf-v4.3-rc/0001-Raw-Ethernet-interface-the-first-draft.tar

kostaond, 2021-06-15:
Here's a WIP patch based on IDF 4.3-rc to demonstrate the Raw Ethernet interface. Please note that it is very first draft.

[0001-Raw-Ethernet-interface-the-first-draft.zip](https://github.com/espressif/esp-idf/files/6655069/0001-Raw-Ethernet-interface-the-first-draft.zip)

szekelyisz, 2021-06-15:
Hi @kostaond, Thanks a lot for taking a look at this. Your proposed solution sounds awesome, I think it meets our requirements. I'd be happy to work on the switch controller logic. I don't think it's a problem that tail tag information is only available for non-IP protocols, it's layer 2 information anyway, so layer 3 and above shouldn't be concerned about it. While the port information doesn't matter much, the tail tag also carries information about the frame's priority, which would be nice to have access to for IP traffic as well, but that doesn't seem like a very important feature for now, I can definitely live without it at least.

Give me a few days to take a look at the patch.

tionebrr, 2021-06-15:
Hello @kostaond,
Yes, big thanks. You got quite deep into the rabbit hole. The solution sounds like what I wanted to implement but I never found the time to do so. I think there might be a way to intercept the packets out of lwip and tag them at this point. That would require hijacking function pointers. I need to dig into the code to find back those pointers. IIRC it is in eth_mac or esp_eth.
I've made a header file some times ago with almost all KSZ registers that you can use if needed. Same as @szekelyisz , I'll need some time to get back to this as it is not my first priority but you can ask if you need help, I'll do the maximum.
[ksz8863rll_component.zip](https://github.com/espressif/esp-idf/files/6656234/ksz8863rll_component.zip)

tionebrr, 2021-06-15:
Oh wait. I've missed some files in the zip.
[ksz8863rll_component.zip](https://github.com/espressif/esp-idf/files/6656288/ksz8863rll_component.zip)
I took a quick read into what I've done. So, eth_phy is just hacked together and need to be rewritten to comply to the new strategy. eth_mac is unchanged from the esp_idf and was there to keep it handy as a reference. Basically this component was just hacked together for testing purpose.
The packet from the lwip are passing by receive and transmit in eth_mac. That could be a good point to inject a tail tag, but I'm not sure how you would go about RTSP.
Don't forget that there is also the option to make the KSZ work as a dual MAC device. So, if bluetooth is not used in the ESP32, this MAC address could theoretically be used for one port of the KSZ, and the usual eth address for the other.

kostaond, 2021-06-16:
Thank you guys for your pro-activity and your suggestions.
Here’s the plan:
1) I need some reliable hardware solution at first so I ask hardware group to develop a board with KSZ8863RLL.
2) In the mean time, I will be working on Raw Ethernet Interface since it does not require any specific hardware and it has a potential to be universally used in the future.
3) Once the board is ready, we can start finalizing Ethernet driver for KSZ8863RLL (MAC and PHY components). This activity will be ideally based on your current progress. @tionebrr I will check your code and we can coordinate the following actions.
4) Develop Switch component example. It can be just simple demonstrator from Espressif’s perspective.

IP traffic Tail tagging has the lowest priority for me so far unless you provide some strong argument :) It can be used for traffic prioritizing as @szekelyisz mentioned but the KSZ offers QoS which seems to me as better solution. I also tend to think that if Tail tagging was used for IP traffic then the IP stack should have been aware of it. Anyway, if to be injected to the IP traffic outside the IP stack than it should be done in NETIF. This is from where I get the raw frames as well. So I will keep it in mind during the Raw Ethernet interface development.

tionebrr, 2021-06-16:
I made a small board for testing. Ethernet works fine but there are some I²C issues, surely due to the long wires between the KSZ test board and the ESP dev board, and poor grounding. I was not able to pinpoint where the problem was coming from exactly. It is finicky, but it works. I still have 4 naked boards and could send one to both of you if need be.
[image]
If you are redesigning a PCB, I would advise to go straight for a 4 layers board with the ESP-32 module soldered on.

I'm not sure IP should be aware of Tail Tagging in any way. I think this is a Layer 2 thing, really specific and close to the hardware. To me, Tail tagging might be the trick to enable dual IP on the ESP32, and the applications for this are numerous. I'm not sure about how hard this would be to implement tho.

kostaond, 2021-06-17:
@tionebrr Thank you for the offer! The plan is to develop a board with ESP module soldered as you suggested to avoid any intermittent issues. I'll let you know if anything needed.
Regarding Tail tagging, you are right, it is specific L2. So I updated the draft little bit. I added hooks to NETIF by which you could manipulate data at L2 as you wish. Search for `esp_netif_transmit` and `esp_netif_receive` for more details. The hook functions can be registered by:
`void esp_netif_transmit_hook_attach(esp_netif_t *esp_netif, void *hook_fn);`
`void esp_netif_recv_hook_attach(esp_netif_t *esp_netif, void *hook_fn);`
I also updated the Raw Ethernet interface draft to use the receive hook.
The hook functions give you absolute freedom to manipulate with the Ethernet frame. It might be to much freedom at the end :) , so please keep in mind that this is just a draft and final solution may vary.
[0001-Raw-Ethernet-interface-the-first-draft2.zip](https://github.com/espressif/esp-idf/files/6671191/0001-Raw-Ethernet-interface-the-first-draft2.zip)

kostaond, 2021-10-10:
hi @mckay6842, the L2 Ethernet interface is currently in code review. Schematics of KSZ8863RLL development board is also in review and the first version of PCB is going to be ready very soon. Therefore, realistically, I presume we could start working the driver in one month time frame.

mckay6842, 2021-10-11:
Hi @kostaond,
thanks for the update, sounds great.

kostaond, 2021-11-25:
Just a quick update. We have the prototype board with KSZ8863 ready and the L2 Ethernet interface merge request has been recently approved.
[image]

Just a side note, there is really a crisis in electronic parts distribution since it was not so easy to buy KSZ8863 chips. The usual  big distributors are out of stock!

kostaond, 2021-12-15:
Hi all, prepared the very first concept of the KSZ8863 driver and updated `ethernet/basic` to demonstrate its usage. There will be several options how the KSZ8863 could be used in the ESP32 environment (please stay tuned). However, this first example demonstrates how to use it along with Tail tagging feature and so acts as two separate Ethernet interfaces with their own MAC addresses. Please check it [here](https://github.com/kostaond/esp-idf/tree/feature/ksz8863). The idea is to create three `Ethernet handlers`, the first one is a gateway from host (ESP32) to the KSZ8863 utilizing ESP's internal EMAC. The remaining two are then directly accessed via that gateway using Tail tagging. I will elaborate it better later but I hope the idea is clear from provided example code. Advantage of such approach is a fact that it can exist in the current ESP Ethernet infrastructure.

Please have a look and let me know your thoughts.

kostaond, 2021-12-15:
BTW: @tionebrr, I really started with the code which you shared. It was great help for the very start! Thank you very much for sharing! I changed a lot since then. However still, I would like to ask you what about licensing. Would you be OK to keep it as is now? (there was no licence in the provided files)

tionebrr, 2021-12-15:
Hello @kostaond ,
Very glad I could help. I don't remember what is the usual licensing of the esp-idf code. Is it MIT? CC0? If the code could retain my name that would be great, but I don't want any license to be a hassle on your side. Would an MIT license work for you?

kostaond, 2021-12-16:
@tionebrr, we use Apache-2.0. What is your full name?

tionebrr, 2021-12-16:
It's Benoît Roehr. I'm ok with the Apache license. 👍

florianhumblot, 2022-01-21:
@kostaond Do you have a timeline for this feature?

kostaond, 2022-01-21:
@florianhumblot, I am currently busy with several bugfixes and their "random" nature and that's is always pain. And even if I wasn't, it would take some time to KSZ8863 driver make it to IDF master branch (review and interface polishing usually takes time). The demo which I've already shared is functioning quite well but there are one or two big philosophical things which I need to resolve. The first is a way how to handle common control of KSZ8863 (like reset, common register configuration) since there are several PHY and MAC instances which are not aware about their state. And the second one is a way of how to update lwIP to support optimized Tx tail tagging (I already have an [idea](https://github.com/kostaond/esp-idf/blob/3aab9188729170a51f8ab6a7b79d301dbc215004/components/esp_eth/src/esp_eth_pmac_ksz8863.c#L319) but it affect lwIP implementation where goal is to keep it as close as possible to upstream). These two things, or at least the first one, I would like to resolve in next two weeks. Therefore it would be better to wait for it. Or if you have any idea of how it could be handled, you can share it and we can cooperate.

kostaond, 2022-02-04:
So I think I resolved those philosophical issues, added SPI control interface support and made several other improvements. Please check [here](https://github.com/kostaond/esp-idf/tree/feature/ksz8863). I am waiting for one other MR to be approved (it improves a way of how to init MAC layer) then I could send this KSZ8863 for preliminary internal review. In any case, it would be great if anybody from you guys tried to play with it a little bit too and provided feedback.

tionebrr, 2022-02-06:
@kostaond sorry I'm very busy with other things these times. I'm thinking about getting back to the project that needed an ESP32 switch but I don't know when I'll find the time to do so...
I took a look in the code some weeks ago and it looked good to me, but can't run it right now. I'm reading the example code right now.
Could be a bit hard for other people to run the code tho. I'm not sure there are other KSZ8863 ESP32 board running out there...

florianhumblot, 2022-02-08:
@kostaond I can run your version on our hardware, but I'm wondering what version of esp-idf it would be if I were to checkout your branch? We've recently upgraded to esp-idf v4.4, but there's a lot of changes Comparing espressif:release/v4.4...kostaond:feature/ksz8863 · espressif/esp-idf:
[image]

kostaond, 2022-02-08:
@tionebrr I understand and thanks for the feedback. I will try to comment better and I am also planning to write some description how the things are implemented from higher level point of view.

@florianhumblot my branch is based on master (commit b092fa073047c957545a0ae9504f04972a8c6d74). Regarding the testing, I've been mainly interested if the driver meets your use case. And if not, then what would you like to see. For example, the current driver example utilized the KSZ8863 as two ports endpoints. I call it "Two Ports Mode", i.e. each port acts as separate endpoint with specific MAC address, its own NETIF instance and no traffic is exchanged between P1 and P2 directly at KSZ8863 level. This mode takes advantage of Tail taging and in such configuration, the KSZ8863 can be used in networks were P1 & P2 are connected to the same switch (or in presence of redundant path). However, I am also planning to create an example to use the KSZ8863 in "Switch Mode". In this mode only, one endpoint with one NETIF exists in ESP32 and KSZ8863 acts as switch. This mode will require more management (like RTSP) to be able to be used properly in networks with redundant paths.

florianhumblot, 2022-02-08:
In our case, we have the following use-case:
 - The KSZ8863 is an unmanaged switch, where we do want to write *some* settings (mainly clock source)
 - We want to be able to detect events such as cable plugged-in on either of the ports (and if possible know which port)
 
So it seems that the use-case you've implemented doesn't quite match our requirements. 
Apart from that, I'm not entirely sure what redundant paths are, so I can't comment on those :)

kostaond, 2022-02-09:
@florianhumblot, I see, I will prepare example with KSZ8863 used as a switch. It is easy when do not care about Tail tagging. However, I would like to demonstrate usage of Tail tagging in this mode too. So please give me some time. Or I could share the example without it if you are interested.
Detection of link is available.
And regarding "redundant paths", please check [this post](https://www.networkacademy.io/ccna/ethernet/an-switching-redundant-links) to see more.

kostaond, 2022-02-18:
@florianhumblot, after exploring various of dead ends, I was able to finally manage to prepare Switch example with tail tagging. Please see my repository shared before.

kostaond, 2022-04-19:
Hi all, good news! ESP-IDF Ethernet infrastructure has been updated to be able to host KSZ8863, see [here](https://github.com/espressif/esp-idf/commit/4051b80b4dfaa961baeaa27aa3e220016c6b6eef). The driver itself will be available via component manager. The beta version is already available [here](https://github.com/espressif/esp-eth-drivers). It would be great if you had time and give it a try. All comments/suggestions are very welcomed!

tionebrr, 2022-04-19:
Hello @kostaond. That's great news, and it came right on time ! 
I was about to start working back on this. I guess in a week or so I might be able to test it. Cheers !

Alvin1Zhang, 2022-05-07:
Thanks for reporting, feel free to reopen.

closed 2022-05-07 by Alvin1Zhang

tionebrr, 2022-12-11:
Hello @kostaond ,
The driver is working perfectly for me. It would be nice to have the examples packaged with the component directly via the component manager tho, but it's another issue. Thanks !

kostaond, 2022-12-16:
@tionebrr, thank you for positive feedback! If you found any issue or had any idea for improvement, don't hesitate to report it.

Regarding the component manager and examples, each example can be downloaded as separate archive or created by e.g. `idf.py create-project-from-example "espressif/ksz8863^0.2.2:simple_switch"`, see https://components.espressif.com/components/espressif/ksz8863. This is how the component manager works...

diplfranzhoepfinger, 2023-07-19:
> [image]

is the Layout and Schematic of this Board available ? 

@kostaond ?

Thanks, 

Franz

diplfranzhoepfinger, 2023-07-19:
Next Question: can we buy one of these Boards ?

kostaond, 2023-07-19:
Hi Franz,

schematic is available [here](https://github.com/espressif/esp-eth-drivers/tree/master/ksz8863/docs/test_board/schematics). PCB layout is not available. Unfortunately, the board cannot be bought, it was developed solely for the KSZ8863 driver debug and testing purposes.

Ondrej

diplfranzhoepfinger, 2023-07-19:
> Hi Franz,
> 
> schematic is available [here](https://github.com/espressif/esp-eth-drivers/tree/master/ksz8863/docs/test_board/schematics). 

Thanks ! 

> PCB layout is not available.

is ist available under NDA ?

Alvin1Zhang, 2023-07-21:
@diplfranzhoepfinger Hi, thanks for the updates, please help write to sales@espressif.com, thanks.

mickeyl, 2024-03-28:
The KSZ8863 looks great, a much better option than the W5500, although also almost 5 times the price. If anyone of you is spotting a commercial board with an ESP32S3 for evaluation purposes, I'd be all ears. Does the beta driver work w/ both I2C and SPI interfaces or is only one these supported?

kostaond, 2024-04-02:
@mickeyl, SPI and I2C is only used for management of the KSZ8863. The data interface is RMII. Therefore it cannot be connected to ESP32S3...

relaxibus, 2025-01-20:
Is I2C now working properly? Any idea for a port to the Arduino or Platformio IDE?

kostaond, 2025-01-24:
> Is I2C now working properly? Any idea for a port to the Arduino or Platformio IDE?

The driver is available at [ESP Registry](https://components.espressif.com/components/espressif/ksz8863/versions/0.2.7). I2C works as expected. You can use it even in Arduino, just initialize the driver as shown in examples in your Arduino code. Test board schematics and Gerber files can be found at: https://github.com/espressif/esp-eth-drivers/tree/master/ksz8863/docs/test_board

relaxibus, 2025-01-24:
Using Platformio with Arduino Core, the KSZ8863 driver is not available.
Could you explain with an example or screenshots how to use correctly or
how to import the driver into the Arduino Core? That would be very
appreciated.

diplfranzhoepfinger, 2025-01-24:
@relaxibus i really recomment to switch to ESP-IDF on a Linux Workstation, 
using ESP-IDF plus Espressif IDE. 
your Experience will be MUCH MUCH better than with Arduino. 

you then have all the Package Management from Espressif and best Support. 

i is a small learn curve, but then it opens so much possibilities. 

just my personal Opinion.

relaxibus, 2025-01-24:
Hey, you may be right, however as I'm not the direct programmer, my
colleagues are using Platformio with Arduino Core. But, I will ask them :-)
Thanks  for your advice. Are you experienced with the KSZ stuff and ESP32?

diplfranzhoepfinger, 2025-01-24:
no, no experience with KSZ and ESP32, 

i am searching for a Soltution with SPE (Single Pair Ethernet) with POE (Power over Data Line) 

so you have a Device, 2 Connectors with each 2 pins, 

and you can Daisy Chain Power and Ethernet. this would be really cool. 

another attempt would be 10-BaseT1, there you can multidrop several Devices without a Switch, 
but i think the Switch attempt is better (on T1 a Short on any place makes all your Devices not communicating)

kostaond, 2025-01-24:
@relaxibus Arduino Core is just a wrapper over ESP-IDF drivers, check https://github.com/espressif/arduino-esp32/blob/master/libraries/Ethernet/src/ETH.cpp You can do something similar (simpler) by yourself.

kostaond, 2025-01-24:
> Switch attempt is better (on T1 a Short on any place makes all your Devices not communicating)

If you daisy chain, you can get to the same problem when one device in the chain fails...

diplfranzhoepfinger, 2025-01-24:
> > Switch attempt is better (on T1 a Short on any place makes all your Devices not communicating)
> 
> If you daisy chain, you can get to the same problem when one device in the chain fails...

yes, but then you see it clearly. and the Devices before are unaffected.

diplfranzhoepfinger, 2025-01-24:
> [@relaxibus](https://github.com/relaxibus) Arduino Core is just a wrapper over ESP-IDF drivers, check https://github.com/espressif/arduino-esp32/blob/master/libraries/Ethernet/src/ETH.cpp You can do something similar (simpler) by yourself.

absolutely right. 
Arduino add only Memory consumption, and does not add any Feature.