espressif/esp-idf #6489: Linking CXX executable (Project).elf FAILED: (Project).elf

Aistify, 2021-02-01:
**Environment**
Development Kit: SparkFun Thing Plus - ESP32 WROOM
Module or chip used: ESP32-D0WD-V3 Dual Core
IDF version v4.2 release (today)
Build System: idf.py
Compiler version: 
Operating System: Windows 10
(Windows only) environment type: ESP Command Prompt
Using an IDE?: No
Power Supply: USB

**Problem Description**
I'm trying to build https://github.com/Silvanosky/PositionEstimator

But I keep getting errors near the end.
I've tried multiple versions to no avail.
I've also used idf.py fullclean

esptool.py v3.0
Generated C:/ESP/examples/Position/build/bootloader/bootloader.bin
[934/935] Linking CXX executable Position.elf
FAILED: Position.elf
...... (Full Log in "Other items if possible" section)
collect2.exe: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
ninja failed with exit code 1

**Expected Behavior**
Project should finish building

**Actual Behavior**
Building fails near end 

**Other items if possible**
[Log.txt](https://github.com/espressif/esp-idf/files/5906196/Log.txt)
[Position.zip](https://github.com/espressif/esp-idf/files/5906224/Position.zip)

igrr, 2021-02-01:
@Aistify it seems that there is an error in CMakeLists.txt file of the main component:
https://github.com/Silvanosky/PositionEstimator/blob/22b0e785c00b2cc1ba031ad36e9c4b9ed0971a90/position/main/CMakeLists.txt#L1

The errors reported are for class ICM_20948, defined in ICM_20948.cpp. However the CMakeLists.txt file only compiles "Icm20948.cpp". It seems that other source files might also need to be compiled. For example, the missing `FusionBiasInitialise` function is defined in FusionBias.c.

You have mentioned,
> I've edited one of the CMakeLists.txt files to have proper extensions to fix some issues but I keep getting errors near the end.

However the file Position/main/CMakeLists.txt in the attached archive appears to be exactly the same as https://github.com/Silvanosky/PositionEstimator/blob/master/position/main/CMakeLists.txt.

Since this doesn't appear to be an issue in ESP-IDF project, I'll close this ticked. Please open the issue in the correct project for further support: https://github.com/Silvanosky/PositionEstimator/issues.

If you believe this is a bug in ESP-IDF, please feel free to provide more information and reopen the issue.

closed 2021-02-01 by igrr

szekelyisz, 2021-02-01:
Undefined references fo functions usually mean that some of your source files are not compiled. You're using the include files that declare a function so you can reference it, but the linker is unable to find their definition when it links the final executable.

I suggest you go through the functions that appear undefined, find the source files (.c and .cpp, header files don't matter here) that contain their definition and see if they are compiled before. If they are not, then you need to add those files to the list of files to be compiled. This is done through the CMakeLists.txt of the component that contains them. See the docs for details.

Another scenario that can cause this is when you're mixing C code with C++. If you're trying to call a C function from a C++ source file, that function needs to be declared `extern "C"`, otherwise the linker won't see it because of different calling conventions for C and C++. That will cause the same error message.

my-dudhwala, 2023-09-13:
> Undefined references fo functions usually mean that some of your source files are not compiled. You're using the include files that declare a function so you can reference it, but the linker is [...]

I am getting the same problem, but I am not getting the long output as shown above, and yea, I have merged 2 codes, one is IPv6 from a git repo(Link is given below) and the second code is esp32 with google sheet(Link is attached), help please, and I am not sure that the IPv6 code is working correctly or not but I have managed to get serial output. take me as a very beginner in IDF, thank you