espressif/esp-idf #6490: Bootloader Error while building the project

madmax0704, 2021-02-02:
## Environment
- Development Kit: ESP32-DevKitC
- Kit version (for DevKitC): v4
- Module or chip used: ESP32-WROOM-32
- IDF version (run ``git describe --tags`` to find it): release/v4.2
- Build System: CMake
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it): esp-2020r3-8.4.0
- Operating System: Windows (VS code extension)
- (Windows only) environment type: ESP Command Prompt
- Using an IDE?: Yes (Visual Studio Code)
- Power Supply: USB

## Problem Description
FAILED: bootloader-prefix/src/bootloader-stamp/bootloader-build bootloader/bootloader.elf bootloader/bootloader.bin bootloader/bootloader.map

I got error while buliding the project. I left the menuconfig on default as it's just a hello world code. I am using extension of VS code to install necessary requirements for ESP32. I tried on both v4.1 latest version and release v4.2. 

### Expected Behavior
To complete the build

### Actual Behavior
bootloader error

### Steps to reproduce
Loaded default template-app from Create project
Selected port number
Started the build

### Code to reproduce this issue
#include <stdio.h>
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "esp_system.h"
#include "esp_spi_flash.h"

void app_main(void)
{
    printf("Hello world!\n");

    /* Print chip information */
    esp_chip_info_t chip_info;
    esp_chip_info(&chip_info);
    printf("This is ESP32 chip with %d CPU cores, WiFi%s%s, ",
            chip_info.cores,
            (chip_info.features & CHIP_FEATURE_BT) ? "/BT" : "",
            (chip_info.features & CHIP_FEATURE_BLE) ? "/BLE" : "");

    printf("silicon revision %d, ", chip_info.revision);

    printf("%dMB %s flash\n", spi_flash_get_chip_size() / (1024 * 1024),
            (chip_info.features & CHIP_FEATURE_EMB_FLASH) ? "embedded" : "external");

    for (int i = 10; i >= 0; i--) {
        printf("Restarting in %d seconds...\n", i);
        vTaskDelay(1000 / portTICK_PERIOD_MS);
    }
    printf("Restarting now.\n");
    fflush(stdout);
    esp_restart();
}

## Debug Logs
[597/932] Linking CXX static library esp-idf\nvs_flash\libnvs_flash.a
[598/932] Performing build step for 'bootloader'
FAILED: bootloader-prefix/src/bootloader-stamp/bootloader-build bootloader/bootloader.elf bootloader/bootloader.bin bootloader/bootloader.map 
cmd.exe /C "cd /D "C:\Users\shaan\Documents\Project\ESP32 Project\template-app\build\bootloader" && C:\Users\shaan\.espressif\tools\cmake\3.16.4\bin\cmake.exe --build ."
[1/102] Generating project_elf_src.c
[2/102] Building C object esp-idf/soc/CMakeFiles/__idf_soc.dir/src/lldesc.c.obj
[3/102] Building C object esp-idf/soc/CMakeFiles/__idf_soc.dir/src/cpu_util.c.obj
[4/102] Building C object esp-idf/soc/CMakeFiles/__idf_soc.dir/src/memory_layout_utils.c.obj
[... 89 lines cut ...]
[94/102] Linking C static library esp-idf\soc\libsoc.a
[95/102] Building C object esp-idf/bootloader_support/CMakeFiles/__idf_bootloader_support.dir/src/esp_image_format.c.obj
[96/102] Linking C static library esp-idf\micro-ecc\libmicro-ecc.a
[97/102] Linking C static library esp-idf\bootloader_support\libbootloader_support.a
[98/102] Linking C static library esp-idf\efuse\libefuse.a
[99/102] Linking C static library esp-idf\spi_flash\libspi_flash.a
[100/102] Linking C static library esp-idf\main\libmain.a
[101/102] Linking C executable bootloader.elf
FAILED: bootloader.elf 
cmd.exe /C "cd . && C:\Users\shaan\.espressif\tools\xtensa-esp32-elf\esp-2020r3-8.4.0\xtensa-esp32-elf\bin\xtensa-esp32-elf-gcc.exe -mlongcalls -Wno-frame-address   CMakeFiles/bootloader.elf.dir/project_elf_src.c.obj  -o bootloader.elf  esp-idf/soc/libsoc.a  esp-idf/log/liblog.a  esp-idf/micro-ecc/libmicro-ecc.a  esp-idf/spi_flash/libspi_flash.a  esp-idf/efuse/libefuse.a  esp-idf/bootloader_support/libbootloader_support.a  esp-idf/main/libmain.a  -Wl,--cref -Wl,--Map=C:/Users/shaan/Documents/Project/ESP32 Project/template-app/build/bootloader/bootloader.map  -fno-rtti  -fno-lto  esp-idf/spi_flash/libspi_flash.a  esp-idf/efuse/libefuse.a  esp-idf/bootloader_support/libbootloader_support.a  esp-idf/spi_flash/libspi_flash.a  esp-idf/efuse/libefuse.a  esp-idf/bootloader_support/libbootloader_support.a  esp-idf/micro-ecc/libmicro-ecc.a  esp-idf/soc/libsoc.a  esp-idf/log/liblog.a  esp-idf/soc/soc/esp32/libsoc_esp32.a  esp-idf/soc/libsoc.a  esp-idf/log/liblog.a  esp-idf/soc/soc/esp32/libsoc_esp32.a  -L C:/Users/shaan/esp/esp-idf/components/esp32/ld  -T esp32.peripherals.ld  -Wl,--gc-sections  -L C:/Users/shaan/esp/esp-idf/components/esp_rom/esp32/ld  -T esp32.rom.ld  -T esp32.rom.newlib-funcs.ld  -T esp32.rom.libgcc.ld  -L C:/Users/shaan/esp/esp-idf/components/bootloader/subproject/main/ld/esp32  -T bootloader.ld  -T bootloader.rom.ld && cd ."
xtensa-esp32-elf-gcc.exe: error: Project/template-app/build/bootloader/bootloader.map: No such file or directory
ninja: build stopped: subcommand failed.
[599/932] Linking CXX static library esp-idf\wpa_supplicant\libwpa_supplicant.a
ninja: build stopped: subcommand failed.
The terminal process terminated with exit code: 1.
```
Debug log goes here, should contain the backtrace, as well as the reset source if it is a crash.
Please copy the plain text here for us to search the error log. Or attach the complete logs but leave the main part here if the log is *too* long.
```

## Other items if possible

- [ ] sdkconfig file (attach the sdkconfig file from your project folder)
- [ ] elf file in the ``build`` folder (**note this may contain all the code details and symbols of your project.**)
- [ ] coredump (This provides stacks of tasks.)

madmax0704, 2021-02-02:
Sorry I didn't provide enough details in the main issue. So adding some more points

### System Specs
AMD Ryzen 7 4800H 
16GB RAM
Nvidia Geforce 1660Ti
512GB SSD

Windows 10 build 1909
VS code v1.52.1
Espressif IDF extension v0.6.1 (preview)
Python 3.7.9 64 bit
Git 2.30.0.2 64 Bit

### Menuconfig (SDK configuration editor)
I kept all settings on default

Serial flasher config
Flash SPI mode: DIO
Flash SPI speed: 40 MHz
Flash size: 2 MB

Detect flash size when flashing bootloader
Before flashing: Reset to bootloader
After flashing: Reset after flashing
'idf.py monitor' baud rate: 115200 bps

### Some more debug logs from start of the terminal
ESP-IDF Custom Terminal
$ cmake
[0/1] Re-running CMake...
-- Project is not inside a git repository, or git repository has no commits; will not use 'git describe' to determine PROJECT_VER.
-- Building ESP-IDF components for target esp32
-- Checking Python dependencies...
Python requirements from C:\Users\shaan\esp\esp-idf\requirements.txt are satisfied.
-- Could NOT find Perl (missing: PERL_EXECUTABLE) 
-- App "template-app" version: 1
-- Adding linker script C:/Users/shaan/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.newlib-time.ld
-- Adding linker script C:/Users/shaan/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.ld
-- Adding linker script C:/Users/shaan/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.libgcc.ld
-- Adding linker script C:/Users/shaan/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.newlib-data.ld
-- Adding linker script C:/Users/shaan/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.syscalls.ld
-- Adding linker script C:/Users/shaan/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.newlib-funcs.ld
-- Adding linker script C:/Users/shaan/Documents/Project/ESP32 Project/template-app/build/esp-idf/esp32/esp32_out.ld
-- Adding linker script C:/Users/shaan/esp/esp-idf/components/esp32/ld/esp32.project.ld.in
-- Adding linker script C:/Users/shaan/esp/esp-idf/components/esp32/ld/esp32.peripherals.ld
-- Components: app_trace app_update asio bootloader bootloader_support bt cbor coap console cxx driver efuse esp-tls esp32 esp_adc_cal esp_common esp_eth esp_event esp_gdbstub esp_hid esp_http_client esp_http_server esp_https_ota esp_https_server esp_ipc esp_local_ctrl esp_netif esp_ringbuf esp_rom esp_serial_slave_link esp_system esp_timer esp_websocket_client esp_wifi espcoredump esptool_py expat fatfs freemodbus freertos heap idf_test jsmn json libsodium log lwip mbedtls mdns mqtt newlib nghttp nvs_flash openssl partition_table perfmon protobuf-c protocomm pthread sdmmc soc spi_flash spiffs tcp_transport tcpip_adapter tinyusb ulp unity vfs wear_levelling wifi_provisioning wpa_supplicant xtensa
-- Component paths: C:/Users/shaan/esp/esp-idf/components/app_trace C:/Users/shaan/esp/esp-idf/components/app_update C:/Users/shaan/esp/esp-idf/components/asio C:/Users/shaan/esp/esp-idf/components/bootloader C:/Users/shaan/esp/esp-idf/components/bootloader_support C:/Users/shaan/esp/esp-idf/components/bt C:/Users/shaan/esp/esp-idf/components/cbor C:/Users/shaan/esp/esp-idf/components/coap C:/Users/shaan/esp/esp-idf/components/console C:/Users/shaan/esp/esp-idf/components/cxx C:/Users/shaan/esp/esp-idf/components/driver C:/Users/shaan/esp/esp-idf/components/efuse C:/Users/shaan/esp/esp-idf/components/esp-tls C:/Users/shaan/esp/esp-idf/components/esp32 C:/Users/shaan/esp/esp-idf/components/esp_adc_cal C:/Users/shaan/esp/esp-idf/components/esp_common C:/Users/shaan/esp/esp-idf/components/esp_eth C:/Users/shaan/esp/esp-idf/components/esp_event C:/Users/shaan/esp/esp-idf/components/esp_gdbstub C:/Users/shaan/esp/esp-idf/components/esp_hid C:/Users/shaan/esp/esp-idf/components/esp_http_client C:/Users/shaan/esp/esp-idf/components/esp_http_server C:/Users/shaan/esp/esp-idf/components/esp_https_ota C:/Users/shaan/esp/esp-idf/components/esp_https_server C:/Users/shaan/esp/esp-idf/components/esp_ipc C:/Users/shaan/esp/esp-idf/components/esp_local_ctrl C:/Users/shaan/esp/esp-idf/components/esp_netif C:/Users/shaan/esp/esp-idf/components/esp_ringbuf C:/Users/shaan/esp/esp-idf/components/esp_rom C:/Users/shaan/esp/esp-idf/components/esp_serial_slave_link C:/Users/shaan/esp/esp-idf/components/esp_system C:/Users/shaan/esp/esp-idf/components/esp_timer C:/Users/shaan/esp/esp-idf/components/esp_websocket_client C:/Users/shaan/esp/esp-idf/components/esp_wifi C:/Users/shaan/esp/esp-idf/components/espcoredump C:/Users/shaan/esp/esp-idf/components/esptool_py C:/Users/shaan/esp/esp-idf/components/expat C:/Users/shaan/esp/esp-idf/components/fatfs C:/Users/shaan/esp/esp-idf/components/freemodbus C:/Users/shaan/esp/esp-idf/components/freertos C:/Users/shaan/esp/esp-idf/components/heap C:/Users/shaan/esp/esp-idf/components/idf_test C:/Users/shaan/esp/esp-idf/components/jsmn C:/Users/shaan/esp/esp-idf/components/json C:/Users/shaan/esp/esp-idf/components/libsodium C:/Users/shaan/esp/esp-idf/components/log C:/Users/shaan/esp/esp-idf/components/lwip C:/Users/shaan/esp/esp-idf/components/mbedtls C:/Users/shaan/esp/esp-idf/components/mdns C:/Users/shaan/esp/esp-idf/components/mqtt C:/Users/shaan/esp/esp-idf/components/newlib C:/Users/shaan/esp/esp-idf/components/nghttp C:/Users/shaan/esp/esp-idf/components/nvs_flash C:/Users/shaan/esp/esp-idf/components/openssl C:/Users/shaan/esp/esp-idf/components/partition_table C:/Users/shaan/esp/esp-idf/components/perfmon C:/Users/shaan/esp/esp-idf/components/protobuf-c C:/Users/shaan/esp/esp-idf/components/protocomm C:/Users/shaan/esp/esp-i [... cut ...] p/esp-idf/components/wear_levelling C:/Users/shaan/esp/esp-idf/components/wifi_provisioning C:/Users/shaan/esp/esp-idf/components/wpa_supplicant C:/Users/shaan/esp/esp-idf/components/xtensa
-- Configuring done
-- Generating done
-- Build files have been written to: C:/Users/shaan/Documents/Project/ESP32 Project/template-app/build
[1/932] Generating project_elf_src.c
[2/932] Generating linker script...
[3/932] Building C object esp-idf/mbedtls/mbedtls/library/CMakeFiles/mbedcrypto.dir/aesni.c.obj

7ROLI101, 2021-02-05:
I noticed that in your path to the project (C:\Users\shaan\Documents\Project\ESP32 Project\template-app\build\bootloader), you have a space between ESP32 and Project. When I looked at the location of the error(Project/template-app/build/bootloader/bootloader.map: No such file or directory), it seems that the space is cutting off the location path.

7ROLI101, 2021-02-05:
Can you try to change up your location path so that the space between ESP32 and Project is removed?
You could try ESP32Project or ESP32_Project, for example.

madmax0704, 2021-02-05:
That was a silly mistake. Thanks it worked.

closed 2021-02-05 by madmax0704