espressif/esp-idf #6493: esp32-s2   uvc

PureHing, 2021-02-03:
@igrr hello，Is it possible to support uvc driver in esp32-s2？

igrr, 2021-02-03:
It is possible. We are working on adding parts of the USB host support now. Once that is done and the isochronous transfers are supported, we will finish the port of libuvc.

PureHing, 2021-02-03:
> It is possible. We are working on adding parts of the USB host support now. Once that is done and the isochronous transfers are supported, we will finish the port of libuvc.

@igrr Greate job! BTW,When can it be opened? Because the project is more urgent, is it possible to provide relevant information or give me some advice？Thanks.

igrr, 2021-02-03:
I'm sorry I can't give you a timeline for this feature yet. We will update this issue when there is some progress.

PureHing, 2021-02-04:
ok.Thanks

closed 2021-02-04 by PureHing

weigede, 2022-04-10:
你们都是外国人码?聊个天还用英语???母语不舒服吗?

yangminglong, 2022-06-20:
> 你们都是外国人码?聊个天还用英语???母语不舒服吗?

这个芯片很多外国人用的