espressif/esp-idf #6494: System crashes after universal provisioning using Espressif's Android app

dbahrdt, 2021-02-03:
## Environment
- Development Kit:      none
- Module or chip used:  ESP32-WROOM-32D
- IDF version: v4.2-238-g8cd16b60f
- Build System:        idf.py
- Compiler version: xtensa-esp32-elf-gcc (crosstool-NG esp-2020r3) 8.4.0
- Operating System:     Arch Linux
- Using an IDE?: No
- Power Supply:    USB

## Problem Description
Using universal provisioning results in a crash. The crash only occurs when I'm using the Android app.
If I use the provisioning script (esp_prov), then everything is fine.
Apple pairing seems to work too (according to my colleagues).

The crash is caused by a null pointer dereference of `pc_httpd` at https://github.com/espressif/esp-idf/blob/8cd16b60f56dec4f319f6d2e73b6dec794588312/components/protocomm/src/transports/protocomm_httpd.c#L36
This in turn seems to be caused by a race condition where the protocomm stuff is deleted while a socket related to the provisioning is still open. The aforementioned function is the deleter for the user context of the http session and it is called when the socket is closed due to a connection abort.  I guess the later is caused by the network change.

### Expected Behavior
The system should not crash.

### Actual Behavior
The system crashes.

### Steps to reproduce
1. Use universal provisioning with the Android app

### Code to reproduce this issue
You can use the same code from issue #6474:
https://github.com/dbahrdt/esp-wolfssl-provisioning-crash

Simply edit `tools/debuggin/cfg` to your needs and rename it to `.cfg`.
The script `energize` will:
1. Create temporary factory_nvs data
2. Temporary CA+server-certificate+sign it
3. Build the project
4. Flash the project and the factory_nvs data
5. Start the app using idf.py monitor
6. Provision the device using universal provisioning using the Android app

## Debug Logs
See https://gist.github.com/dbahrdt/f0937c3a80e2126d52e9a4c71e0f70a9
```
I (102743) wifi_prov_mgr: Provisioning stopped
W (102753) httpd_txrx: httpd_sock_err: error in recv : 113
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x400daff4  PS      : 0x00060330  A0      : 0x801214d8  A1      : 0x3ffcc3a0  
0x400daff4: protocomm_httpd_session_close at esp-idf/components/protocomm/src/transports/protocomm_httpd.c:36

A2      : 0x3ffd0cdc  A3      : 0x00000033  A4      : 0x00000000  A5      : 0x3ffc8598  
A6      : 0x00000000  A7      : 0x00000000  A8      : 0x00000000  A9      : 0x3ffcc380  
A10     : 0x00000000  A11     : 0x00000033  A12     : 0x3ffc8598  A13     : 0x00000002  
A14     : 0x00000001  A15     : 0x00000034  SAR     : 0x00000004  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000008  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0xffffffff  

Backtrace:0x400daff1:0x3ffcc3a0 0x401214d5:0x3ffcc3c0 0x40120353:0x3ffcc3e0 0x401203bf:0x3ffcc420 0x40088c15:0x3ffcc440
0x400daff1: protocomm_httpd_session_close at esp-idf/components/protocomm/src/transports/protocomm_httpd.c:36

0x401214d5: httpd_sess_delete at esp-idf/components/esp_http_server/src/httpd_sess.c:233

0x40120353: httpd_server at esp-idf/components/esp_http_server/src/httpd_main.c:204 (discriminator 15)

0x401203bf: httpd_thread at esp-idf/components/esp_http_server/src/httpd_main.c:229 (discriminator 15)

0x40088c15: vPortTaskWrapper at esp-idf/components/freertos/xtensa/port.c:143
```

yuanjianmin, 2021-02-10:
@dbahrdt Please check this [commit](https://github.com/espressif/esp-idf/commit/72fa8d07507045ddd982a37f6e3b3adc254c541f#diff-aa75c71bb9b3b493841fa3f9cdb2a1717796ff520495d1ed4462407f664990e9) whether can solve the issue.

dbahrdt, 2021-02-12:
A checkout of that commit works.
A "back-port" (cherry-pick) also works.
I only get the following error message:
```
W (69586) httpd_txrx: httpd_sock_err: error in recv : 113
```

yuanjianmin, 2021-02-24:
> 
> 
> A checkout of that commit works.
> A "back-port" (cherry-pick) also works.
> I only get the following error message:
> 
> ```
> W (69586) httpd_txrx: httpd_sock_err: error in recv : 113
> ```

@dbahrdt Could you please provide the step to reproduce this issue. Such as which the example you test, which version(provisioning app) and platform(ios or Android) Phone you test, which sdk code version you use. And could you please offer the completed console log which can help us to analysis. In addition, could you use the latest master branch or release/v4.2 branch to test and check whether the issue exist.

dbahrdt, 2021-02-25:
App-Version: v2.0.3
Platform: Android 9 (Moto X4)
Example: See the one above

Your proposed commit fixed the issue. If you cannot reproduce the issue then please let me know and I'll try to allocate some time to further investigate this.

yuanjianmin, 2021-02-26:
@dbahrdt The `errno` 113 indicates that the socket is disconnected abnormally. It may be because the previous socket was not closed cleanly, which will cause the problem, but this does not affect the function.  Since the crash is no more seen and fixed by this [commit](https://github.com/espressif/esp-idf/commit/72fa8d07507045ddd982a37f6e3b3adc254c541f#diff-aa75c71bb9b3b493841fa3f9cdb2a1717796ff520495d1ed4462407f664990e9), we would close the issue :).

dbahrdt, 2021-02-26:
@yuanjianmin 
Fell free to close this issue. However I am wondering if the fix is backported to v4.2. My company is planing on releasing a product based on v4.2 since v4.3 will likely be still in beta/rc-phase for some time.
I found another bug in the SecureBoot-Code which is fixed in current master. However if stuff is not getting backported then we might as well use v4.3 instead of filing an issue against v4.2.

yuanjianmin, 2021-03-01:
@dbahrdt Yes, the fix has been backported to [release/v4.2](https://github.com/espressif/esp-idf/commit/328254c6316fcb555550d9a40cccfa4963aa69dd). You can test it with the latest branch release/v4.2. If have any problem, please let me know and now i will close this issue.

closed 2021-03-01 by Alvin1Zhang