espressif/esp-idf #6496: esp-idf-3.3   ./install.sh

zhy102, 2021-02-03:
Administrator@VAEPHO7HRJ69PK4 MINGW64 /c/esp-idf (master)
$ ./install.sh
Installing ESP-IDF tools
Installing tools: xtensa-esp32-elf, esp32ulp-elf, cmake, openocd-esp32, mconf, ninja, idf-exe, ccache
Skipping xtensa-esp32-elf@1.22.0-97-gc752ad5d-5.2.0 (already installed)
Skipping esp32ulp-elf@2.28.51.20170517 (already installed)
Skipping cmake@3.13.4 (already installed)
Skipping openocd-esp32@v0.10.0-esp32-20200709 (already installed)
Skipping mconf@v4.6.0.0-idf-20190628 (already installed)
Skipping ninja@1.9.0 (already installed)
Skipping idf-exe@1.0.1 (already installed)
Skipping ccache@3.7 (already installed)
Installing Python environment and packages
fatal: No names found, cannot describe anything.
Traceback (most recent call last):
  File "C:/esp-idf/tools/idf_tools.py", line 1358, in <module>
    main(sys.argv[1:])
  File "C:/esp-idf/tools/idf_tools.py", line 1354, in main
    action_func(args)
  File "C:/esp-idf/tools/idf_tools.py", line 1144, in action_install_python_env
    idf_python_env_path, _, virtualenv_python = get_python_env_path()
  File "C:/esp-idf/tools/idf_tools.py", line 840, in get_python_env_path
    idf_version_str = subprocess.check_output(['git', '--work-tree=' + global_idf_path, 'describe'], cwd=global_idf_path, env=os.environ).decode()
  File "C:\Users\Administrator\AppData\Local\Programs\Python\Python37\lib\subprocess.py", line 411, in check_output
    **kwargs).stdout
  File "C:\Users\Administrator\AppData\Local\Programs\Python\Python37\lib\subprocess.py", line 512, in run
    output=stdout, stderr=stderr)
subprocess.CalledProcessError: Command '['git', '--work-tree=C:/esp-idf', 'describe']' returned non-zero exit status 128.

dobairoland, 2021-02-03:
@zhy102 If you want to use ESP-IDF v3.3 then I would like to recommend the ZIPed environment from here: https://docs.espressif.com/projects/esp-idf/en/v3.3.3/get-started/windows-setup.html#toolchain-setup. It is based on MINGW32. To the best of my knowledge, MINGW64 was never supported.

The install and export scripts are intended to be run from Windows command line (there is also a Power Shell alternative). You can use that option for ESP-IDF v3.3 as well but is officially supported from v4.0.

The reported issue is related to invoking a Windows Python from MINGW64. Windows Python uses Windows paths which are invalid in MINGW64.

dobairoland, 2021-02-03:
A haven't noticed before that you are on the latest master branch.

In that case, please use the Windows Tool Installer as described here https://docs.espressif.com/projects/esp-idf/en/latest/esp32/get-started/windows-setup.html and use Windows command line or Power Shell.

closed 2021-02-03 by dobairoland