espressif/esp-idf #6497: Bluetooth classic GAP callback crashed when user selected Cancel on Android phone during pairing procedure

X-Files-ESP32, 2021-02-03:
Hi,
I'm currently trying to use the "bt_app_acceptor" code provided in esp-idf examples on a TTGO device.
Most of the times, the device crashed and rebooted if the I selected 'Cancel' during the Pairing procedure with my Android cellphone (Motorola g7).
Here one of the logs captured during the Pairing procedure when it crashed (if I clicked on Cancel on my phone):

----------------------------------------------------
[0;32mI (41946) SPP_INITIATOR_DEMO: ESP_BT_GAP_CFM_REQ_EVT Please compare the numeric value: 804822[0m
Guru Meditation Error: Core  0 panic'ed (InstructionFetchError). Exception was unhandled.

Core  0 register dump:
PC      : 0x3ffb9a34  PS      : 0x00060530  A0      : 0x800f30e8  A1      : 0x3ffcd8d0  
A2      : 0x00000001  A3      : 0x3ffb9a34  A4      : 0x3ffbac48  A5      : 0x00000000  
A6      : 0x3ffbac68  A7      : 0x3f411ef0  A8      : 0x800df8a8  A9      : 0x3ffcd790  
A10     : 0x3ffbac68  A11     : 0x00000001  A12     : 0x00000000  A13     : 0x0000000a  
A14     : 0x00000001  A15     : 0x00000001  SAR     : 0x0000001d  EXCCAUSE: 0x00000002  
EXCVADDR: 0x3ffb9a34  LBEG    : 0x4000c46c  LEND    : 0x4000c477  LCOUNT  : 0xffffffff  

Backtrace:0x3ffb9a31:0x3ffcd8d0 |<-CORRUPTED
----------------------------------------------------

Could someone verify, thanks you.

Wth-Esp, 2021-02-04:
Hi, @X-Files-ESP32 

Sorry to bring you the problem during the development. But could you please provide the basic information for this issue like other issues do?

Thanks

X-Files-ESP32, 2021-02-04:
Hi,

I’m currently trying to use the ‘bt_spp_acceptor’ code provided by esp-idf as examples for classic Bluetooth applications (located in “esp-idf\examples\bluetooth\bluedroid\classic_bt” folder) to develop a new Bluetooth project.

Here are the devices I used in my test:

- TTGO (with LCD display) development board to run ‘bt_spp_acceptor’ as spp server (acceptor).
- Motorola g7 cellphone.

- ‘Serial Bluetooth Terminal’ as Android application that I used as Bluetooth Initiator (Client) on my cellphone to pair and connect to the spp acceptor running on TTGO board.
Note that you can download this app. from Playstore if you want to do a test.

Now the problem:

- During the ‘pairing’ process between my cellphone and the ‘spp acceptor’ (TTGO running the bt_spp_acceptor’ code), when I received the ‘Pair with ESP_SPP_ACCEPTOR’ message asking me to confirm the 6-digit Blutooth pairing code:

    -> If I click on ‘Pair’ – the code worked and my cellphone was able to pair with it and I can transmit/received messages normally on both sides without problem.

    -> If I click on ‘Cancel’ then the TTGO board (bt_spp_acceptor) crashed and produced the log message below before rebooting:

[0;32mI (22296) SPP_ACCEPTOR_DEMO: ESP_BT_GAP_CFM_REQ_EVT Please compare the numeric value: 551395[0m
Guru Meditation Error: Core  0 panic'ed (InstructionFetchError). Exception was unhandled.

Core  0 register dump:
PC      : 0x3ffb639c  PS      : 0x00060530  A0      : 0x800f2984  A1      : 0x3ffcd7b0  
A2      : 0x00000001  A3      : 0x3ffb639c  A4      : 0x3ffb7f24  A5      : 0x00000000  
A6      : 0x3ffb7f44  A7      : 0x3f411dd4  A8      : 0x800df144  A9      : 0x3ffcd670  
A10     : 0x3ffb7f44  A11     : 0x00000001  A12     : 0x00000000  A13     : 0x0000000a  
A14     : 0x00000001  A15     : 0x00000001  SAR     : 0x0000001c  EXCCAUSE: 0x00000002  
EXCVADDR: 0x3ffb639c  LBEG    : 0x4000c46c  LEND    : 0x4000c477  LCOUNT  : 0xffffffff  

Backtrace:0x3ffb6399:0x3ffcd7b0 |<-CORRUPTED

ELF file SHA256: 9c5e19642c770f1a

Rebooting...
    

Note that every time during the pairing cycle if I press ‘Cancel’ on my cellphone to cancel the pairing the ‘bt_spp_acceptor’ code crashed – this is not just a random problem.

As at where it stands now, as long as this problem is not yet resolved - it is impossible for me to develop any commercial project based on the current esp-idf Bluetooth classic stack (I’m using the latest esp-idf stack).
I suspected that it’s not only just the particular problem with this TTGO development board – I tested on two diff. TTGO boards and got the same result. I hope you could test this code on a different development board just to confirm
my doubt.

Thanks you.

From: KBFEDE 
Sent: Thursday, February 4, 2021 1:42 AM
To: espressif/esp-idf 
Cc: X-Files-ESP32 ; Mention 
Subject: Re: [espressif/esp-idf] Bluetooth classic GAP callback crashed when user selected Cancel on Android phone during pairing procedure (IDFGH-4685) (#6497)

Hi, @X-Files-ESP32

Sorry to bring you the problem during the development. But could you please provide the basic information for this issue like other issues do?

Thanks

Alvin1Zhang, 2021-03-01:
Thanks for reporting and sorry for the inconvenience, we have a fix under internal reviewing, and the issue will close automatically once the fix is available on GitHub.

Alvin1Zhang, 2021-03-16:
Thanks for reporting, the fix is available https://github.com/espressif/esp-idf/commit/41ffe31cbd5354231eb2d51f5507c60ed8c283ae, feel free to reopen.

closed 2021-03-16 by Alvin1Zhang