espressif/esp-idf #6498: OTA sign verification fails on 16MB ESP32-WROOM-32

thecashit, 2021-02-03:
We have 4mb ESP32 devices and everything is ok, OTA runs like a charm, sign works and everything is well now.
Recently we launched 16MB devices with the same firmware code, only difference is partition table and how our code works with it.

The problem is that we can't run OTA on 16MB chips. The bootloader just doesn't accept the signature.
Moreover, if we check with espsecure.py, this script says that the signature is correct.

Partition table for 4mb looks likes following:
phy_init, data, phy, 0x0f000, 0x1000
otadata, data, ota, 0x10000, 0x2000
nvs, data, nvs, 0x12000, 0x20000
# data1 (384KB) 0x32000, 0x60000
# data2 (40KB) 0x92000, 0xA000
# data3 (528KB) 0x9C000, 0x84000
ota_0, app, 0x10, 0x120000, 0x170000
ota_1, app, 0x11, 0x290000, 0x170000

For the 16mb:
phy_init, data, phy, 0x00f000, 0x1000
otadata, data, ota, 0x010000, 0x2000
factory_nvs,data, nvs, 0x012000, 0x4000
nvs_keys, data, nvs_keys, 0x016000, 0x1000
coredump,data, coredump,0x017000, 0x10000
nvs, data, nvs, 0x027000, 0x49000
ota_0, app, 0x10, 0x070000, 0x300000
ota_1, app, 0x11, 0x370000, 0x300000
misc, data,, 0x670000, 0x30000
data1, data,, 0x6A0000, 0x200000
data2, data,, 0x8A0000, 0x200000
data3, data,, 0xAA0000, 0x40000
data4, data,, 0xAE0000, 0x210000
constants, data,, 0xCF0000, 0x310000

The application and the environment are the same, in the sdkconfig settings in the "Serial flasher config" section the chip size is set to 16MB and autodetection of the chip size.
Partitition table is also choose correct (for 16mb).

Security features in sdkconfig are:
CONFIG_SECURE_SIGNED_ON_UPDATE=y
CONFIG_SECURE_SIGNED_APPS=y
CONFIG_SECURE_SIGNED_APPS_NO_SECURE_BOOT=y
CONFIG_SECURE_SIGNED_APPS_ECDSA_SCHEME=y
# CONFIG_SECURE_SIGNED_ON_BOOT_NO_SECURE_BOOT is not set
CONFIG_SECURE_SIGNED_ON_UPDATE_NO_SECURE_BOOT=y
# CONFIG_SECURE_BOOT is not set
CONFIG_SECURE_BOOT_BUILD_SIGNED_BINARIES=y
CONFIG_SECURE_BOOT_SIGNING_KEY="private_ota_signing_key.pem"
# CONFIG_SECURE_FLASH_ENC_ENABLED is not set

Before flashing, we do a clean, the firmware starts and everything works except for the OTA updates. 
The update is downloaded, but then the signature verification fails.

Could you please tell me, maybe there are some tricks in working with a 16MB chip?
Because of all our attempts to make it works failed.

Additional question: several 16MB devices have been shipped to customers, unfortunately, we learned late about the problems with the update, how is it possible to update them via OTA? For example, if the key is not completely saved in the bootloader, find it out at the compilation stage in order to sign the application.

## Environment

- Module or chip used:  ESP32-WROOM-32D [16MB]
- ESP-IDF v4.2
- Build System: idf.py
- macOS

projectgus, 2021-02-08:
Hi @thecashit,

Thanks for the very detailed issue report. I have a couple of follow up questions:

* Are you able to get a serial log from the device when it fails to update?
* Just to confirm: the exact IDF version you have is v4.2 (the release tag), not a different v4.2-xyz version?

Angus

thecashit, 2021-02-09:
> Hi @thecashit,
> 
> Thanks for the very detailed issue report. I have a couple of follow up questions:
> 
> * Are you able to get a serial log from the device when it fails to update?
> * Just to confirm: the exact IDF version you have is v4.2 (the release tag), not a different v4.2-xyz version?
> 
> Angus

Sorry for the long answer, we went deep into debugging and analyzing the problem.

The bottom line is that there are no problems in esp-idf, everything works as intended. 
The problem was in the incorrect CSV file contains partition table, some of the areas when assembling the partition table were replaced by OTA (there were gaps in the original file) and this was the cause of the problem.

Thank you for prompt reply.

closed 2021-02-09 by thecashit

projectgus, 2021-02-09:
Hi @thecashit,

I'm glad you found the solution. Do you mean that you updated the partition table by OTA (erased part of it and rewrote it)?

If you think there's anything that ESP-IDF could do to help prevent the kind of error you had, please let me know  (for example, any conditions we could check when processing the partition CSV file that are not currently checked.)

thecashit, 2021-02-11:
> Hi @thecashit,
> I'm glad you found the root cause. Do you mean that you updated the partition table by OTA (erased part of it and rewrote it)? 
> If you think there's anything that ESP-IDF could do to help prevent the kind of error you had, please let me know (for example, any conditions we could check when processing the partition CSV file that are not currently checked.)

Hello and thank you for your desire to get to the point.
Yes, overwriting the partition table solved our problem. However, already on the devices sent, this is not possible, since the current version of our firmware does not allow this. We solve this issue by recalling devices to reflashing partition table.

Our main problem appeared in the fact that the following entry in the CSV of the partition table (just an example):
scache, data,, 0xAA0000, 0x40000
When building to .bin file turned into the following:
scache, data, ota, 0xAA0000, 0x40000

And after that, OTA stopped working, the new firmware is downloaded, but the bootloader cannot switch to ota1. The firmware also has the ability to force rollback to ota1 and this mechanism also does not work. Despite the fact that in the logs we see something similar to "everything is ota valid".

I think that replacing "scache, data,," to "scache, data, ota," is not very correct and either when creating a partition table, partition tool need to check for such empty values ​​and show a warning to the console, or the bootloader (?) should not take into account in "data, ota" check, only "app, ota".

Of course, our case is one-off. 
Our partition table has not changed for several years, but someone else may encounter similar behavior.