espressif/esp-idf #6499: Using ESP32 S2 FTM/TOF with external APs while connected to main AP

MS71, 2021-02-03:
I like to measure the distance to "external" APs (APx + APy) while my ESP32-S2 is connected to third Wifi (APmain) and exchanging data with that.
```
APx <= FTM => |
              | ESP32-S2 (STA) | <= data exchange => APmain
APy <= FTM => |
```

If I try this, then I get this error message:

_wifi:FTM with external AP not supported in connected state yet!_

Can I expect that functionality in near future?

sagb2015, 2021-02-04:
@MS71 , measuring distance with AP other than the connected one will require support for channel switching as well since other APs could be on other channel. We are in process of adding that support. Until then we will discuss internally if we could have ranging done at least on same channel with  APs other than the one to which ESP device is connected.

MS71, 2021-02-04:
Hi,
thank you very much for the answer.
Can I use the current FTM implementation in a ESP MESH network?

nachiketkukade, 2021-11-18:
@MS71 , this feature is supported in latest IDF v4.3.1 release. Simply issue the ftm command with SSID of ext-AP. However if the ext-AP is in a different channel, the FTM operation is limited to a short burst due to channel switching.
Also there is no plan to support FTM in an ESP MESH network as of now.

closed 2021-11-18 by nachiketkukade