espressif/esp-idf #6500: ESP32 controller not available

Qbicz, 2021-02-03:
## Environment

- Development Kit:      ESP32-PICO-Kit
- Kit version (for WroverKit/PicoKit/DevKitC): v4.1
- Module or chip used:  ESP32-PICO-D4
- IDF version (run ``git describe --tags`` to find it): v4.2-1-g88c34cc4e
- Build System:         idf.py
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it): xtensa-esp32-elf-gcc (crosstool-NG esp-2020r3) 8.4.0
- Operating System:     Linux
- (Windows only) environment type: [MSYS2 mingw32|ESP Command Prompt|Plain Command Prompt|PowerShell].
- Using an IDE?: No
- Power Supply: USB

## Problem Description

We're using an example from ESP-IDF https://github.com/espressif/esp-idf/tree/master/examples/bluetooth/hci/controller_hci_uart which should provide HCI controller firmware with UART as IO. However, when attached to a Linux computer, the UART HCI controller is not available: `No default controller available`

I would suspect this example is 4 years old an maybe the UART HCI IO was broken at some point. Or it may be user error, in this case please help me fix it.

### Expected Behavior

HCI Controller available, can be used with BlueZ over UART:
> bluetoothctl 
> Agent registered
> [CHG] Controller AB:CD:12:23:56:78 Pairable: yes

### Actual Behavior

Controller not available.
> bluetoothctl
> Agent registered
> [bluetooth]# power on
> No default controller available

### Steps to reproduce

1. Download and install ESP-IDF v4.2
2. cd examples/bluetooth/hci/controller_hci_uart
3. idf.py flash
4. idf.py monitor
5. Use BlueZ 5.55 on Linux host
6. Attach HCI controller with 921600 baudrate, HCI protocol H4:
> /usr/bin/btattach -B /dev/ttymxc1 -S 921600 --protocol h4 &
> Attaching Primary controller to /dev/ttymxc1
> Switched line discipline from 0 to 15
> Device index 0 attached
7. Start BlueZ daemon
> /usr/libexec/bluetooth/bluetoothd
8. Use Bluetoothctl to enable and use HCI UART controller
> bluetoothctl
> Agent registered
> [bluetooth]# power on
> No default controller available

### Setup/wiring

ESP32 UART 1 <----->   Linux host UART /dev/ttymxc1
- TX 5  <----->  RX
- RX 18  <----->  TX
- RTS 19  <----->  CTS
- CTS 23  <----->  RTS

ESP32 PICO D4 powered from USB.

### Code to reproduce this issue
HCI UART example from ESP-IDF, no changes
ttps://github.com/espressif/esp-idf/tree/master/examples/bluetooth/hci/controller_hci_uart

## Debug Logs

```
--- idf_monitor on /dev/ttyUSB0 115200 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x12 (SPI_FAST_FLASH_BOOT)
configsip: 188777542, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:4
load:0x3fff0034,len:7092
load:0x40078000,len:13212
load:0x40080400,len:4568
0x40080400: _init at ??:?

entry 0x400806f4
I (30) boot: ESP-IDF v4.2-1-g88c34cc4e-dirty 2nd stage bootloader
I (30) boot: compile time 22:27:05
I (30) boot: chip revision: 1
I (34) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (41) boot.esp32: SPI Speed      : 40MHz
I (46) boot.esp32: SPI Mode       : DIO
I (50) boot.esp32: SPI Flash Size : 2MB
I (55) boot: Enabling RNG early entropy source...
I (60) boot: Partition Table:
I (64) boot: ## Label            Usage          Type ST Offset   Length
I (71) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (78) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (86) boot:  2 factory          factory app      00 00 00010000 00100000
I (93) boot: End of partition table
I (98) boot_comm: chip revision: 1, min. application chip revision: 0
I (105) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x091e4 ( 37348) map
I (128) esp_image: segment 1: paddr=0x0001920c vaddr=0x3ffbdb60 size=0x03308 ( 13064) load
I (134) esp_image: segment 2: paddr=0x0001c51c vaddr=0x40080000 size=0x00404 (  1028) load
0x40080000: _WindowOverflow4 at /home/filip/esp/esp-idf/components/freertos/xtensa/xtensa_vectors.S:1730

I (135) esp_image: segment 3: paddr=0x0001c928 vaddr=0x40080404 size=0x036f0 ( 14064) load
I (150) esp_image: segment 4: paddr=0x00020020 vaddr=0x400d0020 size=0x32d00 (208128) map
0x400d0020: _stext at ??:?

I (232) esp_image: segment 5: paddr=0x00052d28 vaddr=0x40083af4 size=0x11654 ( 71252) load
0x40083af4: r_huart_finish_transfers at ??:?

I (275) boot: Loaded app from partition at offset 0x10000
I (275) boot: Disabling RNG early entropy source...
I (275) cpu_start: Pro cpu up.
I (279) cpu_start: Application information:
I (284) cpu_start: Project name:     controller_hci_uart
I (290) cpu_start: App version:      1
I (294) cpu_start: Compile time:     Feb  3 2021 22:27:02
I (300) cpu_start: ELF file SHA256:  4322a7df4085b3fe...
I (306) cpu_start: ESP-IDF:          v4.2-1-g88c34cc4e-dirty
I (312) cpu_start: Starting app cpu, entry point is 0x40081854
0x40081854: call_start_cpu1 at /home/filip/esp/esp-idf/components/esp32/cpu_start.c:287

I (0) cpu_start: App cpu up.
I (323) heap_init: Initializing. RAM available for dynamic allocation:
I (330) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM
I (336) heap_init: At 3FFB7CD8 len 00000328 (0 KiB): DRAM
I (342) heap_init: At 3FFB9A20 len 00004108 (16 KiB): DRAM
I (348) heap_init: At 3FFC2348 len 0001DCB8 (119 KiB): DRAM
I (354) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (361) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (367) heap_init: At 40095148 len 0000AEB8 (43 KiB): IRAM
I (373) cpu_start: Pro cpu start user code
I (392) spi_flash: detected chip: gd
I (392) spi_flash: flash io: dio
W (392) spi_flash: Detected size(4096k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (402) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (437) BTDM_INIT: BT controller compile version [3723d5b]
I (447) system_api: Base MAC address is not set
I (447) system_api: read default base MAC address from EFUSE
I (537) phy: phy_version: 4500, 0cd6843, Sep 17 2020, 15:37:07, 0, 0

```

## Other items if possible

- [ Attached ] sdkconfig file (attach the sdkconfig file from your project folder)
[sdkconfig.txt](https://github.com/espressif/esp-idf/files/5921657/sdkconfig.txt)

Qbicz, 2021-02-10:
After a few days, I have found a solution of this problem.

The problem was on the Linux host side, which was missing Linux kernel crypto interface in user-space

After starting BlueZ daemon with more debug info with 
`bluetoothd -n`
it showed more details:

    bluetoothd[593]: src/adapter.c:get_static_addr() Failed to open crypto
    bluetoothd[593]: No Bluetooth address for index 0

Which led me to this topic: https://github.com/raspberrypi/linux/issues/3628

It turns out, when attaching the UART HCI controller with `btattach`, BlueZ was not able to create a static address, because of missing crypto interface in user-space.

I compiled the kernel with additional configs:

    CONFIG_CRYPTO_USER
    CONFIG_CRYPTO_USER_API
    CONFIG_CRYPTO_USER_API_AEAD
    CONFIG_CRYPTO_USER_API_HASH
    
    CONFIG_CRYPTO_AES
    CONFIG_CRYPTO_CCM
    CONFIG_CRYPTO_AEAD
    CONFIG_CRYPTO_CMAC

After that change, I was able to attach the Bluetooth controller to my Embedded Linux host.

closed 2021-02-10 by Qbicz

danergo, 2021-08-17:
I have the same problem, with one addition: I want to do the same with a host computer with Ubuntu 20.04LTS and an FTDI232 USB-UART controller.

ESP32 is not providing any usable HCI interface over UART to my Linux machine.

Symptoms are the same as for @Qbicz.

My kernel is pretty new:

    Linux computer 5.4.0-80-generic #90-Ubuntu SMP Fri Jul 9 22:49:44 UTC 2021 x86_64 x86_64 x86_64 GNU/Linux

So I believe I have those required options compiled into this, but have no idea on what's the problem.

Running bluetoothd with -n -d doesn't show me any warnings or errors.