espressif/esp-idf #6501: Simultaneous `esp_http_client` causes increase in unexplained errors

andySigler, 2021-02-04:
## Environment

- Development Kit: `ESP32-DevKitC`
- Kit version: `v4`
- Module or chip used: `ESP32-WROVER-E`
- IDF version: `v4.1.1` (commit `4ebe0d8`)
- Build System: `CMake`
- Compiler version: `xtensa-esp32-elf-gcc (crosstool-NG esp-2020r3) 8.4.0`
- Operating System: `macOS`
- Using an IDE?: `No`
- Power Supply: `USB`

## Problem Description

HTTP errors increase when running two `esp_http_client` simultaneously, each in a separate tasks. When not running simultaneously, error rate is about `1/5000`. When running simultaneously, error rate is about `1/500`.

NOTE: errors seem intermittent and randomized. These error rates were found through a few hours of testing.

A working example is attached to this issue.

Here are some snippets to explain.

Two tasks are created, which simply run `GET` requests to `example.com`:

```c
xTaskCreatePinnedToCore(&http_test_task, "http_test_task_0", 8192, (void *)0, 5, NULL, 0);
xTaskCreatePinnedToCore(&http_test_task, "http_test_task_1", 8192, (void *)1, 6, NULL, 1);
```

The `http_test_task` essentially is this:
```c
static void http_test_task(void *pvParameters) {
  while (true) {
    http_test_get("http://example.com");
    vTaskDelay(50 / portTICK_PERIOD_MS);
  }
}
```
And `http_test_get()` essentially is this:
```c
static esp_err_t http_test_get(char *url) {
  esp_err_t err = ESP_OK;
  char *response_body = NULL;
  int status_code = 0;
  int response_length = 0;

  esp_http_client_config_t config = {.url = url};
  esp_http_client_handle_t client = esp_http_client_init(&config);

  // OPEN & FETCH HEADERS
  err = esp_http_client_open(client, 0);
  if (err != ESP_OK) {
    goto _done;
  }
  if (esp_http_client_fetch_headers(client) == ESP_FAIL) {
    err = ESP_FAIL;
    goto _done;
  }

  // READ STATUS CODE & RESPONSE BODY
  status_code = esp_http_client_get_status_code(client);
  int data_length = esp_http_client_get_content_length(client);
  if (status_code != 200) {
    err = ESP_FAIL;
  }
  if (data_length > 0) {
    response_body = calloc(data_length + 1, sizeof(char));
    response_length =
        esp_http_client_read(client, response_body, data_length);
    if (response_length < 0) {
      goto _done;
    }
  }

_done:
  if (response_body != NULL) {
    free(response_body);
  }
  esp_http_client_cleanup(client);
  return err;
}
```

As you can see, each task is simultaneously running `GET` requests to the same URL. This setup will cause errors to happen about `1` out of every `500` requests.

Running tests, I find that if I use a semaphore to prevent the two clients from running requests simultaneously, then this setup will cause errors to happen about `1` out of every `5000` requests.

The updated `http_test_task` with semaphores essentially is this:
```c
static void http_test_task(void *pvParameters) {
  while (true) {
    xSemaphoreTake(_s_http_semphr, portMAX_DELAY);
    http_test_get("http://example.com");
    xSemaphoreGive(_s_http_semphr);
    vTaskDelay(50 / portTICK_PERIOD_MS);
  }
}
```

### Expected Behavior

Running two `esp_http_client` simultaneously, in separate tasks, does not greatly increase the number of errors.

The number of errors that occur when running HTTP clients simultaneously is equal to when running them serially.

### Actual Behavior

The number of errors that occur when running simultaneously greatly increases.

These errors are always coming from one of the following IDF functions:

#### `esp_http_client_open()`

Calling `open()` will increasingly raise the following error while two clients are running:
```
TRANS_TCP: Error create socket
HTTP_CLIENT: Connection failed, sock < 0
```
#### `esp_http_client_fetch_headers()`

Calling `fetch_headers()` will increasingly return `ESP_FAIL`. [I have traced the failure location to this line in `esp_http_client.c`](https://github.com/espressif/esp-idf/blob/6e776946d01ec0d081d09000c36d23ec1d318c06/components/esp_http_client/esp_http_client.c#L1076).

### Steps to reproduce

1. Install IDF version `v4.1.1` (or `v4.0.2`)
2. Download attached example code
2. Run `idf.py menuconfig` to set wifi SSID and password
3. Flash to device
4. Run for several minutes, counting errors
5. Modify source code to enable the semaphore
6. Rebuild and flash to device
7. Run for several minutes, counting errors
8. Compare error count before and after modifying source code

The section of the source code to modify is here:
```c
// uncomment to force HTTP clients to not run simultaneously
// #define HTTP_CLIENTS_BLOCK_EACH_OTHER (1)
```
Uncommenting that line will create a semaphore, and force each HTTP client to **not** run simultaneously.

### Code to reproduce this issue

See attached file.

## Debug Logs

```
W (1531884) HTTP_TEST: Core 1: Total=13100, Fails=15
W (1533014) HTTP_TEST: Core 0: Total=13600, Fails=21
W (1543114) HTTP_TEST: Core 1: Total=13200, Fails=15
W (1543114) HTTP_TEST: Core 0: Total=13700, Fails=21
E (1549344) HTTP_TEST: Failed to fetch headers
E (1549344) HTTP_TEST: HTTP GET request failed: ESP_FAIL (-1)
W (1557624) HTTP_TEST: Core 0: Total=13800, Fails=21
W (1569604) HTTP_TEST: Core 0: Total=13900, Fails=21
E (1572524) HTTP_CLIENT: Connection failed, sock < 0
E (1572524) HTTP_TEST: Failed to open http client
E (1572524) HTTP_TEST: HTTP GET request failed: ESP_ERR_HTTP_CONNECT (28674)
W (1577894) HTTP_TEST: Core 1: Total=13300, Fails=17
E (1587194) HTTP_TEST: Failed to fetch headers
E (1587194) HTTP_TEST: HTTP GET request failed: ESP_FAIL (-1)
E (1596524) HTTP_CLIENT: Connection failed, sock < 0
E (1596524) HTTP_TEST: Failed to open http client
E (1596524) HTTP_TEST: HTTP GET request failed: ESP_ERR_HTTP_CONNECT (28674)
W (1598484) HTTP_TEST: Core 0: Total=14000, Fails=22
E (1605524) HTTP_CLIENT: Connection failed, sock < 0
E (1605524) HTTP_TEST: Failed to open http client
E (1605524) HTTP_TEST: HTTP GET request failed: ESP_ERR_HTTP_CONNECT (28674)
W (1608954) HTTP_TEST: Core 0: Total=14100, Fails=22
W (1612724) HTTP_TEST: Core 1: Total=13400, Fails=19
W (1619774) HTTP_TEST: Core 0: Total=14200, Fails=22
E (1623784) TRANS_TCP: Error create socket
E (1623784) HTTP_CLIENT: Connection failed, sock < 0
E (1623784) HTTP_TEST: Failed to open http client
E (1623794) HTTP_TEST: HTTP GET request failed: ESP_ERR_HTTP_CONNECT (28674)
W (1623794) HTTP_TEST: Core 1: Total=13500, Fails=20
W (1631014) HTTP_TEST: Core 0: Total=14300, Fails=22
W (1635904) HTTP_TEST: Core 1: Total=13600, Fails=20
```

## Other items if possible

[http-error-test.elf.zip](https://github.com/espressif/esp-idf/files/5922367/http-error-test.elf.zip)

[http_error_test_app.zip](https://github.com/espressif/esp-idf/files/5922369/http_error_test_app.zip)

yuanjianmin, 2021-02-04:
hi @andySigler , Thank you for reporting this question, and attach a very detailed reproduction step. I have followed your step and attached code to reproduce it. And i found the problem the same with you. Next, I will analyze and answer your questions one by one.
## Question 1: 
### esp_http_client_open() will appear `TRANS_TCP: Error create socket`
There is indeed a problem with this place, please follow the patch below for modification and verification.
[0001-tcp-fix-tcp-alloc-may-fail.zip](https://github.com/espressif/esp-idf/files/5925340/0001-tcp-fix-tcp-alloc-may-fail.zip)
### esp_http_client_open() will not appear `TRANS_TCP: Error create socket`, only appear `Connection failed, sock < 0` 
In the v4.1.1, tcp_connect in transport_tcp.c is blocking, while in master branch, it has been set to non-blocking, and call `getsockopt(tcp->sock, SOL_SOCKET, SO_ERROR, (void*)(&sockerr), &len)` to get connect error. I think you can use this code to debug.In short, I think it is possible that the server was disconnected when we connected to the server.
### esp_http_client_fetch_headers() will appear `Failed to fetch headers`
This problem occurred because the server responded to us with FIN. Before calling the function esp_http_client_fetch_headers(), we had successfully conducted a TCP three-way handshake with the server. When we obtained the data through the http request, the server responded to us with FIN, resulting in failing to get data. In my own tests, I have also seen that the server directly responds to RST, so this problem is basically a server problem.
### Why use `HTTP_CLIENTS_BLOCK_EACH_OTHER` and not use, the failure rate is very different.
I think, when we not use `HTTP_CLIENTS_BLOCK_EACH_OTHER`, the http request per unit time is twice that of using `HTTP_CLIENTS_BLOCK_EACH_OTHER`. After all, when `HTTP_CLIENTS_BLOCK_EACH_OTHER` is not used, the two tasks are executed synchronously. Therefore, such a large access load will cause the server to exhibit abnormal problems, such as replying to RST and FIN. I suggest you can choose two different servers for testing.

There are a lot of explanations, but in summary, appear `Error create socket` is indeed a problem, you can modify it according the patch. And other wrong log, I think it's a server problem.

andySigler, 2021-02-05:
Thank you @yuanjianmin for your detailed reply. I will review the information you gave me and test some more, then share my findings.

yuanjianmin, 2021-02-25:
@andySigler Is this issue any updates or newly findings. If yes, please let me know.

yuanjianmin, 2021-03-15:
@andySigler Is there any update? If not, please help to close this issue. Thanks.

closed 2021-04-26 by Alvin1Zhang