espressif/esp-idf #6502: check_python_dependencies.py detected by Comodo as a trojan

rouzbeh-afrasiabi, 2021-02-04:
After using https://dl.espressif.com/dl/esp-idf-tools-setup-2.4.exe to install the tools, when esp-idf cmd or powershell is opened comodo detects check_python_dependencies.py as a trojan(Generic.Trojan.C@6@1 )

dobairoland, 2021-02-04:
Thank you @rouzbeh-afrasiabi for reporting this. We will look into this.

georgik, 2021-02-05:
@rouzbeh-afrasiabi Thank you for reporting the problem. This is false positive identification of potential threat. You can check the .py file source code or run the full scan on folder.

Comodo has feature Auto-Containment which is enabled by default. This feature causes that anything which is not known to Commodo is running in virtualized environment. This will cause pop-up of the message on the first sight. The file can be then found in the list in Commodo - Advanced View - Unrecognized files.

Unfortunately, this feature will also cause problem using any idf.py command. The build process won't run, command line will just idly wait for Commodo to finish containerization. Solution: Turn off the Auto-Containment mode.

Go to Commodo Settings - Advanced View - Auto-Containment - Enabled and change the value to Disabled. Click Ok.
Now the build will run.

antmak, 2021-02-08:
@rouzbeh-afrasiabi Hi! Also you can submit the detection as a false positive, as described in https://help.comodo.com/topic-394-1-767-12121-.html , it could help other comodo users.

georgik, 2021-02-08:
Thank you @rouzbeh-afrasiabi for the link. It looks like a viable solution.

rouzbeh-afrasiabi, 2021-02-09:
Thank you @dobairoland and @georgik. I am going to close this since the issue is due to Comodo.

closed 2021-02-09 by rouzbeh-afrasiabi