espressif/esp-idf #6503: HFP AG unable to open SCO data path.

ypiotrp, 2021-02-04:
## Environment

- Development Kit:      none
- Kit version:          does not apply
- Module or chip used:  ESP32-WROOM-32D
- IDF version          v4.2-rc with patch from #6218 
- Build System:         idf.py
- Compiler version:     8.4.0
- Operating System:     Windows
- Environment type:     PowerShell
- Using an IDE?:        No (only PowerShell with ESP-IDF scripts)
- Power Supply:         USB

## Problem Description
My project is based on HFP-AG example from the ESP-IDF. After applying the patch from #6218 on ESP-IDF v4.2-rc I have a problem with opening SCO data path after `connect_audio()` request. I checked if the problem exists in HFP AG example and it does. I use HCI as a SCO data path, turned off WBS and I changed the `BR/EDR Sync(SCO/eSCO) Max Connections` to 1. 

### Expected Behavior
Audio data being seemlessly transferred to the HFP Client device.

### Actual Behavior
I open and close audio several times. After some time the example can't open the SCO data path and audio closes just after opening. See debug logs below.

### Steps to reproduce

1. Pull the ESP-IDF from the repo in rev 4.2-rc.
2. Apply the patch from #6218.
3. Enter `idf.py menuconfig` and:
    1. In Component config > Bluetooth > Bluetooth controller:
        * Set BR/EDR Sync (SCO/eSCO) Max Connections to 1,
        * Set BR/EDR Sync (SCO/eSCO) default data path to HCI,
    2. In Component config > Bluetooth > Bluedroid Options:
        * Set Hands Free/Handset Profile audio(SCO) data path to HCI,
        * Disable Wide Band Speech
    3. Save the changes and exit `menuconfig`. 
4. In `app_hf_msg_set.c` uncomment line 16 and provide the Bluetooth address of the device you want to connect with.
5. Build the firmware and flash the ESP32.
6. Connect to your headset using `hf con;` command in UART interface.
7. Connect and disconnect audio using `hf cona` and `hf disa`. After some time the issue should take place.

### Code to reproduce this issue
Just original HFP-AG example with changes in `menuconfig` described above

## Debug Logs
```
[0;31mE (305965) CNSL: Command [hf cona;][0m                                                                            
Connect Audio                                                                                                           
[0;33mW (305965) BT_BTM: BTM Remote does not support 2-EDR eSCO[0m                                                      
[0;33mW (305965) BT_BTM: BTM Remote does not support 3-EDR eSCO[0m                                                      
[0;32mI (305965) BT_APP_HF: APP HFP event: AUDIO_STATE_EVT[0m                                                           
[0;31mE (305975) BT_BTM: btm_sco_connected, handle 80[0m                                                                
[0;33mW (305975) BT_BTM: BTM Remote does not support 2-EDR eSCO[0m                                                      
[0;33mW (305985) BT_BTM: BTM Remote does not support 3-EDR eSCO[0m                                                      
[0;32mI (305995) BT_APP_HF: --Audio State connecting[0m                                                                 
[0;31mE (305995) BT_BTM: btm_sco_connected, handle 80[0m                                                                
[0;32mI (306005) BT_APP_HF: APP HFP event: AUDIO_STATE_EVT[0m                                                           
[0;32mI (306005) BT_APP_HF: --Audio State disconnected[0m                                                               
[0;32mI (306015) BT_APP_HF: --ESP AG Audio Connection Disconnected.[0m 
```
Before this happened the handle of the connection was 180 or 181. Can you tell me what it means? 

Regards, Piotr

ypiotrp, 2021-02-11:
Hi @Alvin1Zhang, @Wth-Esp, @munho-espressif.

It's been a week since I posed this issue. Is there something going on with it. Has someone been assigned to it?

Regards,
Piotr

Wth-Esp, 2021-02-18:
Hi @ypiotrp 

The issue is assigned to @munho-espressif and he has been looking into it before the Chinese Spring Festival. He has found something suspicious but not very sure about the root cause. We have back to work now and I think he will share the findings ASAP.

Thanks

munho-espressif, 2021-02-23:
Hi @ypiotrp
Please try this
[hfp_ag.zip](https://github.com/espressif/esp-idf/files/6029188/hfp_ag.zip)

on v4.2-rc.
Thanks

chwwu, 2021-03-02:
Hi @ypiotrp

I facing the same [issue](https://github.com/espressif/esp-idf/issues/6218#issuecomment-753948294) you report
the [patch](https://github.com/espressif/esp-idf/issues/6218#issuecomment-765174168) didn't work for me, is there any change you made?

Thanks

ypiotrp, 2021-03-02:
Hi @chwwu,

this patch actually seems to be working well and solves my problem. The changes I made to 4.2-rc were more or less the two patches - one from another of my issues #6218 and second form the above post. 

What is the problem you faced? Show me some logs and I will try to help you.

Regards

chwwu, 2021-03-02:
Hi @ypiotrp 

The use case is the ESP32 acts as Audio gateway and can receive/sent audio data (pcm?) to handsfree devices
but hf cona; will lead core dump
This is my post [issue](https://github.com/espressif/esp-idf/issues/6649)

Best Regards

Alvin1Zhang, 2021-03-10:
@ypiotrp Thanks for reporting, and glad to hear that the issue resolved, will close the ticket now. For the issue by @chwwu , will track in https://github.com/espressif/esp-idf/issues/6649. Feel free to reopen.

closed 2021-03-10 by Alvin1Zhang