espressif/esp-idf #6504: Starting DHCPC not possible after reinitializing wifi

istokm, 2021-02-04:
## Environment

- Module or chip used:  ESP32-WROOM-32D
- IDF version: v4.3-dev-2950-gb414b139a
- Build System: idf.py
- Compiler version: esp-2020r3 8.4.0
- Power Supply: USB

## Problem Description

If one deinitializes the wifi, and then initializes it again, DHCP client will never be able to start again. I've confirmed this by checking the dhcpc status like this, on the STA_CONNECTED event:
```c++
/**
 * @bug The DHCP client doesn't start if the wifi was previously 
 * deinitialized. I've tried starting it manually if this happens, 
 * but without success. It doesn't return any errors either.
 */
esp_netif_dhcp_status_t status;
esp_netif_dhcpc_get_status(m_esp_netif_sta, &status);

if (status == 0x0) {
  ESP_LOGW(TAG, "[%s]: DHCPC not started! (state=%u)",
    __func__, status);
  ESP_LOGI(TAG, "[%s]: Starting DHCPC", __func__);
  ESP_ERROR_CHECK(esp_netif_dhcpc_start(m_esp_netif_sta));
}
```

To reproduce this, you of course have to initialize the wifi stack, I do that like this:
```c++
// Initialize NVS
esp_err_t err = nvs_flash_init();

if (err == ESP_ERR_NVS_NO_FREE_PAGES || err == ESP_ERR_NVS_NEW_VERSION_FOUND) {
  ESP_LOGW(TAG, "[%s]: Erasing NVS flash due to initialization failure (%s)",
    __func__, esp_err_to_name(err));
  ESP_ERROR_CHECK(nvs_flash_erase());
  ESP_ERROR_CHECK(nvs_flash_init());
}
ESP_ERROR_CHECK(err);

// Create & initialize default netif configuration
ESP_ERROR_CHECK(esp_netif_init());
ESP_ERROR_CHECK(esp_event_loop_create_default());

m_esp_netif_sta = esp_netif_create_default_wifi_sta();

// Setup default wifi config
wifi_init_config_t wifi_init_config = WIFI_INIT_CONFIG_DEFAULT();
ESP_ERROR_CHECK(esp_wifi_init( & wifi_init_config));
ESP_ERROR_CHECK(esp_wifi_set_storage(WIFI_STORAGE_RAM));

// Register esp_wifi_handler() as a handler for wifi & IP events
ESP_ERROR_CHECK(esp_event_handler_instance_register(WIFI_EVENT,
  ESP_EVENT_ANY_ID, & esp_event_handler, this, & m_wifi_events_instance));
ESP_ERROR_CHECK(esp_event_handler_instance_register(IP_EVENT,
  ESP_EVENT_ANY_ID, & esp_event_handler, this, & m_ip_events_instance));

// Set STA mode & start the wifi driver
ESP_ERROR_CHECK(esp_wifi_set_mode(WIFI_MODE_STA));
ESP_ERROR_CHECK(esp_wifi_start());
```

And then simply deinitialize it, I do that in this manner:
```c++
ESP_ERROR_CHECK(esp_wifi_disconnect());
ESP_ERROR_CHECK(esp_wifi_stop());

ESP_ERROR_CHECK(esp_wifi_deinit());

// Unregister internal event handlers
esp_event_handler_instance_unregister(WIFI_EVENT, ESP_EVENT_ANY_ID,
  m_wifi_events_instance);
esp_event_handler_instance_unregister(IP_EVENT, ESP_EVENT_ANY_ID,
  m_wifi_events_instance);

// Destroy the default netif object
esp_netif_destroy(m_esp_netif_sta);
// Not implemented yet
//ESP_ERROR_CHECK(esp_netif_deinit());

ESP_ERROR_CHECK(esp_event_loop_delete_default());
```

If you, after doing this, initialize the wifi again & connect to an AP, not only will the DHCPC will not start automatically as it is supposed to, you won't even be able to start it manually with the `esp_netif_dhcpc_start()` function - it won't even return an error as the `ESP_ERROR_CHECK` macro doesn't trigger.

Let me know If I'm doing anything wrong with the deinitialization, but I haven't observed any memory leaks, so I believe that it should be fine.	

### Actual Behavior

Unable to start DHCP client if the wifi was previously deinitialized.

## Debug Logs

https://gist.github.com/istokm/339140322ce7feadbc3a89ec6ae78bbd

istokm, 2021-02-04:
Note: It is also impossible to use the `esp_netif_get_hostname` function after reinitialization. It fails with the error: `ESP_ERR_ESP_NETIF_IF_NOT_READY`
It does not fail, if you manually set the hostname via `esp_netif_set_hostname` after initialization.

david-cermak, 2021-02-04:
Hi @istokm 

Your code misses deinitialization of wifi-netif handlers. This is needed because you initialize the netif with simplified wifi default interface `esp_netif_create_default_wifi_sta()`, which not only creates the netif, but also registers handlers and attaches the netif to wifi.

Please add this line before `esp_netif_destroy()`
```c
    ESP_ERROR_CHECK(esp_wifi_clear_default_wifi_driver_and_handlers(m_esp_netif_sta));
```
similarly as it's done here: https://github.com/espressif/esp-idf/blob/master/examples/common_components/protocol_examples_common/connect.c#L293

istokm, 2021-02-05:
@david-cermak 

You got it, totally my mistake. Both issues were resolved by adding this statement. Though, to my credit, the documentation for [wifi initialization/esp_netif_create_default_wifi_sta](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/network/esp_netif.html#wifi-default-initialization) doesn't mention it ;)

Alvin1Zhang, 2021-02-05:
@istokm Thanks for the report! Reopening and will close once documentation gets updated.

closed 2021-03-12 by espressif-bot