espressif/esp-idf #6506: Program Build Error (error: operation on '<variable>' may be undefined [-Werror=sequence-point])

carsonRadtke, 2021-02-04:
## Environment

- Development Kit: e-Paper ESP32 Driver Board ([link](https://www.waveshare.com/e-paper-esp32-driver-board.htm))
- Kit version (for WroverKit/PicoKit/DevKitC): N/A
- Module or chip used:  ESP32-WROOM-32
- IDF version (run ``git describe --tags`` to find it): 4.1
- Build System: unsure...I use the vscode-esp-idf-extension
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it): xtensa-esp32-elf-gcc.exe (crosstool-NG esp-2020r2) 8.2.0
- Operating System: Windows
- (Windows only) environment type: vscode-esp-idf-extension
- Using an IDE?: vscode w/ vscode-esp-idf-extension
- Power Supply: USB

## Problem Description

When building my project, the build fails due to an undefined operation on a variable. I compile a similar program (change `void app_main(void)` to `int main(void)`) using gcc (version 9.3.0).

### Expected Behavior

I expect the project to compile (and print 25). As it does using gcc.

### Actual Behavior

The project does not compile.

### Steps to reproduce

I use the extension
1. `<Ctrl> + <E> <B>`

### Code to reproduce this issue

```cpp
#include "esp_log.h"

static const char *TAG = "error_example";

void app_main()
{

    int i = 0;
    int p = (i = 5) * i;

    ESP_LOGI(TAG, "should be 25: %d", p);

}

```

## Debug Logs

```

> Executing task: espressif.esp-idf-extension: ESP-IDF Compile <

ESP-IDF Custom Terminal
$ cmake
-- Project version: 9eaaf33-dirty
-- Building ESP-IDF components for target esp32
-- Checking Python dependencies...
Python requirements from C:\esp-idf\requirements.txt are satisfied.
-- Could NOT find Perl (missing: PERL_EXECUTABLE) 
-- Adding linker script C:/Users/CarsonRadtke/source/repos/EPaper/Esp32Firmware/firmware/template-app/build/esp-idf/esp32/esp32_out.ld
-- Adding linker script C:/esp-idf/components/esp32/ld/esp32.project.ld.in
-- Adding linker script C:/esp-idf/components/esp32/ld/esp32.peripherals.ld
-- Adding linker script C:/esp-idf/components/esp_rom/esp32/ld/esp32.rom.newlib-time.ld
-- Adding linker script C:/esp-idf/components/esp_rom/esp32/ld/esp32.rom.ld
-- Adding linker script C:/esp-idf/components/esp_rom/esp32/ld/esp32.rom.libgcc.ld
-- Adding linker script C:/esp-idf/components/esp_rom/esp32/ld/esp32.rom.newlib-data.ld
-- Adding linker script C:/esp-idf/components/esp_rom/esp32/ld/esp32.rom.syscalls.ld
[... 11 lines cut ...]
$ ESP-IDF Custom Terminal
$ cmake
[1/9] Building C object esp-idf/main/CMakeFiles/__idf_main.dir/main.c.obj
FAILED: esp-idf/main/CMakeFiles/__idf_main.dir/main.c.obj 
C:\Users\CarsonRadtke\.espressif\tools\xtensa-esp32-elf\esp-2020r2-8.2.0\xtensa-esp32-elf\bin\xtensa-esp32-elf-gcc.exe -DHAVE_CONFIG_H -DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\" -DUNITY_INCLUDE_CONFIG_H -DWITH_POSIX -Iconfig -I../main/include -IC:/esp-idf/components/newlib/platform_include -IC:/esp-idf/components/freertos/include -IC:/esp-idf/components/heap/include -IC:/esp-idf/components/log/include -IC:/esp-idf/components/lwip/include/apps -IC:/esp-idf/components/lwip/include/apps/sntp -IC:/esp-idf/components/lwip/lwip/src/include -IC:/esp-idf/components/lwip/port/esp32/include -IC:/esp-idf/components/lwip/port/esp32/include/arch -IC:/esp-idf/components/soc/esp32/include -IC:/esp-idf/components/soc/include -IC:/esp-idf/components/esp_rom/include -IC:/esp-idf/components/esp_common/include -IC:/esp-idf/components/xtensa/include -IC:/esp-idf/components/xtensa/esp32/include -IC:/esp-idf/components/esp32/include -IC:/esp-idf/components/driver/include -IC:/esp-idf/components/driver/esp32/include -IC:/esp-idf/components/esp_ringbuf/include -IC:/esp-idf/components/efuse/include -IC:/esp-idf/components/efuse/esp32/include -IC:/esp-idf/components/vfs/include -IC:/esp-idf/components/esp_wifi/include -IC:/esp-idf/components/esp_wifi/esp32/include -IC:/esp-idf/components/esp_event/include -IC:/esp-idf/components/esp_netif/include -IC:/esp-idf/components/esp_eth/include -IC:/esp-idf/components/tcpip_adapter/include -IC:/esp-idf/components/app_trace/include -IC:/esp-idf/components/app_update/include -IC:/esp-idf/components/spi_flash/include -IC:/esp-idf/components/bootloader_support/include -IC:/esp-idf/components/mbedtls/port/include -IC:/esp-idf/components/mbedtls/mbedtls/include -IC:/esp-idf/components/nvs_flash/include -IC:/esp-idf/components/pthread/include -IC:/esp-idf/components/espcoredump/include -IC:/esp-idf/components/perfmon/include -IC:/esp-idf/components/wpa_supplicant/include -IC:/esp-idf/components/wpa_supplicant/port/include -IC:/esp-idf/components/wpa_supplicant/include/esp_supplicant -IC:/esp-idf/components/asio/asio/asio/include -IC:/esp-idf/components/asio/port/include -IC:/esp-idf/components/cbor/port/include -IC:/esp-idf/components/coap/port/include -IC:/esp-idf/components/coap/port/include/coap -IC:/esp-idf/components/coap/libcoap/include -IC:/esp-idf/components/coap/libcoap/include/coap2 -IC:/esp-idf/components/console -IC:/esp-idf/components/nghttp/port/include -IC:/esp-idf/components/nghttp/nghttp2/lib/includes -IC:/esp-idf/components/esp-tls -IC:/esp-idf/components/esp_adc_cal/include -IC:/esp-idf/components/esp_gdbstub/include -IC:/esp-idf/components/tcp_transport/include -IC:/esp-idf/components/esp_http_client/include -IC:/esp-idf/components/esp_http_server/include -IC:/esp-idf/components/esp_https_ota/include -IC:/esp-idf/components/protobuf-c/protobuf-c -IC:/esp-idf/components/protocomm/include/common -IC:/esp-idf/components/protocomm/include/security -IC:/esp-idf/components/protocomm/include/transports -IC:/esp-idf/components/mdns/include -IC:/esp-idf/components/esp_local_ctrl/include -IC:/esp-idf/components/sdmmc/include -IC:/esp-idf/components/esp_serial_slave_link/include -IC:/esp-idf/components/esp_websocket_client/include -IC:/esp-idf/components/expat/expat/expat/lib -IC:/esp-idf/components/expat/port/include -IC:/esp-idf/components/wear_levelling/include -IC:/esp-idf/components/fatfs/diskio -IC:/esp-idf/components/fatfs/vfs -IC:/esp-idf/components/fatfs/src -IC:/esp-idf/components/freemodbus/common/include -IC:/esp-idf/components/idf_test/include -IC:/esp-idf/components/jsmn/include -IC:/esp-idf/components/json/cJSON -IC:/esp-idf/components/libsodium/libsodium/src/libsodium/include -IC:/esp-idf/components/libsodium/port_include -IC:/esp-idf/components/mqtt/esp-mqtt/include -IC:/esp-idf/components/openssl/include -IC:/esp-idf/components/spiffs/include -IC:/esp-idf/components/ulp/include -IC:/esp-idf/components/unity/include -IC:/esp-idf/components/unity/unity/src -IC:/esp-idf/components/wifi_provisioning/include -mlongcalls -Wno-frame-address   -ffunction-sections -fdata-sections -fstrict-volatile-bitfields -Wall -Werror=all -Wno-error=unused-function -Wno-error=unused-but-set-variable -Wno-error=unused-variable -Wno-error=deprecated-declarations -Wextra -Wno-unused-parameter -Wno-sign-compare -ggdb -O0 -std=gnu99 -Wno-old-style-declaration -D_GNU_SOURCE -DIDF_VER=\"v4.1-dirty\" -DESP_PLATFORM -MD -MT esp-idf/main/CMakeFiles/__idf_main.dir/main.c.obj -MF esp-idf\main\CMakeFiles\__idf_main.dir\main.c.obj.d -o esp-idf/main/CMakeFiles/__idf_main.dir/main.c.obj   -c ../main/main.c
../main/main.c: In function 'app_main':
../main/main.c:10:16: error: operation on 'i' may be undefined [-Werror=sequence-point]
     int p = (i = 5) * i;
             ~~~^~~~
cc1.exe: some warnings being treated as errors
[2/9] cmd.exe /C "cd /D C:\Users\CarsonRadtke\source\repos\EPaper\Esp32Firmware\firmware\template-app\build\esp-idf\partition_table && C:\Users\CarsonRadtke\.espressif\tools\cmake\3.13.4\bin\cmake.exe -E echo "Partition table binary generated. Contents:" && C:\Users\CarsonRadtke\.espressif\tools\cmake\3.13.4\bin\cmake.exe -E echo ******************************************************************************* && C:\Users\CarsonRadtke\.espressif\python_env\idf4.1_py3.9_env\Scripts\python.exe C:/esp-idf/components/partition_table/gen_esp32part.py -q --offset 0x10000 --flash-size 2MB C:/Users/CarsonRadtke/source/repos/EPaper/Esp32Firmware/firmware/template-app/build/partition_table/partition-table.bin && C:\Users\CarsonRadtke\.espressif\tools\cmake\3.13.4\bin\cmake.exe -E echo *******************************************************************************"
Partition table binary generated. Contents:
*******************************************************************************
# Espressif ESP32 Partition Table
# Name, Type, SubType, Offset, Size, Flags
nvs,data,nvs,0x11000,24K,
phy_init,data,phy,0x17000,4K,
factory,app,factory,0x20000,1M,
*******************************************************************************
ninja: build stopped: subcommand failed.
The terminal process failed to launch (exit code: 1).

Terminal will be reused by tasks, press any key to close it.

```

Alvin1Zhang, 2021-02-05:
Thanks for reporting.

igrr, 2021-02-05:
@carsonRadtke when compiling with gcc on the host, have you added `-Wall -Werror`? It should fail if you do: https://godbolt.org/z/E8K4bq.

projectgus, 2021-02-05:
To expand @igrr's point, the reason the compiler doesn't like this code is that this line is undefined behaviour:

```
int p = (i = 5) * i;
```

Should the compiler assign the value of `i` before evaluating the right hand side of the multiplication (`p = 25`), or before it (result: `p = 0`)? Turns out this is explicitly undefined by the C standard, there's no answer to what should happen in this case.

The thing to look up if you want to learn more about this is [sequence points](http://c-faq.com/expr/seqpoints.html).

It is possible to tell the ESP-IDF build system that you don't want gcc to issue this warning, add something to the end of your component CMakeLists.txt file like `target_compile_options(${COMPONENT_LIB} PRIVATE -Wno-sequent-point)`. But note that
the compiler is warning you because it doesn't guarantee that output of this expression will be valid - it may not do what you want, and the result may even change depending on random and unexpected factors.

Suggest changing the code so that any assignment statement is on a separate line to any code that relies on that variable's old or new value.

closed 2021-02-05 by projectgus

carsonRadtke, 2021-02-05:
I understand now. I see `-Werror=all` as a flag the extension uses. Thanks for the help!