espressif/esp-idf #6507: examples/bluetooth/nimble/blehr: shared variable safety

DurandA, 2021-02-04:
https://github.com/espressif/esp-idf/blob/6e776946d01ec0d081d09000c36d23ec1d318c06/examples/bluetooth/nimble/blehr/main/main.c#L37

Probably not an actual bug, but shouldn't `notify_state` theoretically use an `atomic_bool` or another locking mechanism?

It is written in `blehr_gap_event` callback while it is read in the `blehr_tx_hrate` from a timer task.

Alvin1Zhang, 2021-02-05:
Thanks for reporting.

prasad-alatkar, 2021-02-05:
@DurandA Thank you for raising the issue. Could you please help raise this issue on mynewt/nimble github : [mynewt_nimble](https://github.com/apache/mynewt-nimble/issues) ?

The `blehr` example is based on: [blehr](https://github.com/apache/mynewt-nimble/blob/master/apps/blehr/src/main.c#L33).

DurandA, 2021-02-06:
Opened https://github.com/apache/mynewt-nimble/issues/911

DurandA, 2021-02-07:
@prasad-alatkar @Alvin1Zhang As a side question, why does the ESP-IDF example call `ble_svc_gap_init()` and `ble_svc_gatt_init()`?

https://github.com/espressif/esp-idf/blob/6e776946d01ec0d081d09000c36d23ec1d318c06/examples/bluetooth/nimble/blehr/main/gatt_svr.c#L171-L172

These calls are not present in the original NimBLE example.

rahult-github, 2021-08-04:
> @prasad-alatkar @Alvin1Zhang As a side question, why does the ESP-IDF example call `ble_svc_gap_init()` and `ble_svc_gatt_init()`? [...]

Hello @DurandA ,

 The calls are used to register GAP and GATT as a service. This is because:

1. This is a demo application. So, usage of these APIs can show or exhibit end user how a service can be added.
2. SDK also has bluedroid stack support and examples in bluedroid show ways to register services. In order to showcase the functionality in Nimble, some services were chosen to be added. It is just that these services got selected as example.
3. Also , some chararacteristic permissions in those services are controlled from config options selected during compile time. This helps user to control certain permissions as per need via configuration ( without need to modify code ).

Hope this clarifies.

Thanks,
 Rahul

rahult-github, 2022-05-23:
Hi @DurandA ,

Am closing this issue on esp-idf. Hope your query for profile init is resolved by above comment. 

Thanks,
 Rahul

closed 2022-05-23 by rahult-github