espressif/esp-idf #6508: esp_ble_gatts_send_service_change_indication bug on IOS idf 4.2

filzek, 2021-02-05:
Hi

We are trying to send a service change notification within the BLE as 2a05 using esp_ble_gatts_send_service_change_indication but it is not working.

What is the correct way to send a service change notification within the BLE?

Please advise.

filzek, 2021-02-05:
SERVICE CHANGED appears on BLE SCAN on android, but it doesnt appears on IOs, this is a very serious problem.
If you advertise a new name using esp_ble_gap_set_device_name("XXX") IOs will not show this new name, no matter what.

So, we need to clarify correctly the name change using esp_ble_gap_set_device_name and how to set a flag to IOs BLE scan update the cache on it.

Please advise.

xiewenxiang, 2021-03-21:
@filzek 

[image]

Service change notification can be configured to send manually or automatically, From your description, it is recommended that you configure it to send manually.

Alvin1Zhang, 2021-08-04:
@filzek Thanks for reporting, would you please help share if any updates for the issue? Thanks.

filzek, 2022-04-29:
Apple doesn't allow the BLE name change for an already scanned Mac address while the iPhone still powered on, so, no matter you clean cache in iPhone it always return the first name found.

We had change way of doing it by using exclusive recursive macaddress change to be able to allow it to be found and match to our purpose!

closed 2022-04-29 by filzek