espressif/esp-idf #6509: FileNotFoundError .espressif\\tools\\idf-python\\3.8.7\\DLLs'

franz-ISOMAX, 2021-02-05:
using the latest installer: esp-idf-tools-setup-2.4.exe
i get this Error: 

FileNotFoundError .espressif\\tools\\idf-python\\3.8.7\\DLLs'

franz-ISOMAX, 2021-02-05:
here some portion of the Log: 

  
2021-02-05 15:47:06.772   Dest filename: C:\\Users\\mail\\.espressif\\tools\\idf-python\\3.8.7\\Scripts\\pip3.exe  
2021-02-05 15:47:06.773   Time stamp of our file: 2021-01-21 14:22:48.000  
2021-02-05 15:47:06.773   Installing the file.  
2021-02-05 15:47:06.778   Successfully installed the file.  
2021-02-05 15:47:06.779   -- File entry --  
2021-02-05 15:47:06.779   Dest filename: C:\\Users\\mail\\.espressif\\tools\\idf-python\\3.8.7\\Scripts\\virtualenv.exe  
2021-02-05 15:47:06.779   Time stamp of our file: 2021-01-21 14:22:52.000  
2021-02-05 15:47:06.779   Installing the file.  
2021-02-05 15:47:06.787   Successfully installed the file.  
2021-02-05 15:47:06.787   -- File entry --  
2021-02-05 15:47:06.787   Dest filename: C:\\Users\\mail\\.espressif\\tools\\idf-python\\3.8.7\\Scripts\\wheel.exe  
2021-02-05 15:47:06.787   Time stamp of our file: 2021-01-21 14:22:48.000  
2021-02-05 15:47:06.787   Installing the file.  
2021-02-05 15:47:06.795   Successfully installed the file.  
2021-02-05 15:47:06.796   -- File entry --  
2021-02-05 15:47:06.797   Dest filename: C:\\Users\\mail\\.espressif\\dist\\tools\_WD\_excl.ps1  
2021-02-05 15:47:06.797   Time stamp of our file: 2021-01-15 13:43:14.000  
2021-02-05 15:47:06.797   Installing the file.  
2021-02-05 15:47:06.798   Successfully installed the file.  
2021-02-05 15:47:06.799   -- File entry --  
2021-02-05 15:47:06.799   Dest filename: C:\\Users\\mail\\.espressif\\dist\\tools\_WD\_clean.ps1  
2021-02-05 15:47:06.799   Time stamp of our file: 2021-01-15 13:43:14.000  
2021-02-05 15:47:06.799   Installing the file.  
2021-02-05 15:47:06.801   Successfully installed the file.  
2021-02-05 15:47:06.801   -- Registry entry --  
2021-02-05 15:47:06.801   Key: HKEY\_CURRENT\_USER\\Environment  
2021-02-05 15:47:06.801   Value name: IDF\_TOOLS\_PATH  
2021-02-05 15:47:06.801   Creating or opening the key.  
2021-02-05 15:47:06.801   Deleting the value.  
2021-02-05 15:47:06.801   Creating or setting the value.  
2021-02-05 15:47:06.801   Successfully created or set the value.  
2021-02-05 15:47:06.802   Saving uninstall information.  
2021-02-05 15:47:06.802   Deleting uninstall key left over from previous non administrative install.  
2021-02-05 15:47:06.804   Creating new uninstall key: HKEY\_CURRENT\_USER\\Software\\Microsoft\\Windows\\CurrentVersion\\Uninstall\\{9E068D99-5C4B-4E5F-96A3-B17CF291E6BD}\_is1  
2021-02-05 15:47:06.804   Writing uninstall key values.  
2021-02-05 15:47:06.805   Detected previous administrative 64-bit install? No  
2021-02-05 15:47:06.805   Detected previous administrative 32-bit install? No  
2021-02-05 15:47:06.812   Installation process succeeded.  
2021-02-05 15:47:06.814   Extracting temporary file: C:\\Users\\mail\\AppData\\Local\\Temp\\is-3SEK2.tmp\\7za.exe  
2021-02-05 15:47:06.855   PythonExecutablePathUpdateAfterInstall: PythonPath=C:\\Users\\mail\\.espressif\\tools\\idf-python\\3.8.7\\ PythonExecutablePath=C:\\Users\\mail\\.espressif\\tools\\idf-python\\3.8.7\\python.exe  
2021-02-05 15:47:06.855   Checking existence of Embedded Python: C:\\Users\\mail\\.espressif\\tools\\idf-python\\3.8.7\\python.exe  
2021-02-05 15:47:06.855   Embedded Python found.  
2021-02-05 15:47:06.855   Setting PATH for this process: C:\\Users\\mail\\.espressif\\tools\\idf-python\\3.8.7\\;C:\\Program Files\\Git\\cmd\\;C:\\Program Files\\Python39\\Scripts\\;C:\\Program Files\\Python39\\;C:\\Program Files (x86)\\VMware\\VMware Workstation\\bin\\;C:\\Program Files\\Python38\\Scripts\\;C:\\Program Files\\Python38\\;C:\\WINDOWS\\system32;C:\\WINDOWS;C:\\WINDOWS\\System32\\Wbem;C:\\WINDOWS\\System32\\WindowsPowerShell\\v1.0\\;C:\\WINDOWS\\System32\\OpenSSH\\;C:\\Program Files\\nodejs\\;C:\\Program Files\\Google\\Google Apps Sync\\;C:\\Program Files\\Google\\Google Apps Migration\\;C:\\Program Files\\srecord-1.64-win32;C:\\Program Files\\doxygen\\bin;C:\\Go\\bin;C:\\Program Files (x86)\\NVIDIA Corporation\\PhysX\\Common;C:\\Program Files\\NVIDIA Corporation\\NVIDIA NvDLISR;d:\\ProgramFiles\\wureset;C:\\Program Files\\Git\\cmd;C:\\Users\\mail\\AppData\\Local\\Microsoft\\WindowsApps;C:\\Users\\mail\\AppData\\Roaming\\npm;C:\\Program Files\\JetBrains\\CLion 2020.2.1\\bin;C:\\Users\\mail\\AppData\\Local\\Programs\\Microsoft VS Code\\bin;C:\\Users\\mail\\go\\bin;C:\\Users\\mail\\AppData\\Local\\gitkraken\\bin  
2021-02-05 15:47:06.855   PYTHONPATH=  
2021-02-05 15:47:06.855   PYTHONHOME=  
2021-02-05 15:47:06.856   idf\_tools.py exists in IDF directory  
2021-02-05 15:47:06.856   Adding extra Python wheels location. Setting environment variable PIP\_EXTRA\_INDEX\_URL=https://dl.espressif.com/pypi  
2021-02-05 15:47:06.856   idf\_tools.py command: C:\\Users\\mail\\.espressif\\tools\\idf-python\\3.8.7\\python.exe "C:\\Users\\mail\\esp-idf-v4.2\\tools\\idf\_tools.py" --idf-path C:\\Users\\mail\\esp-idf-v4.2  
2021-02-05 15:47:06.856   Installing tools:C:\\Users\\mail\\.espressif\\tools\\idf-python\\3.8.7\\python.exe "C:\\Users\\mail\\esp-idf-v4.2\\tools\\idf\_tools.py" --idf-path C:\\Users\\mail\\esp-idf-v4.2 install  
2021-02-05 15:48:36.655   Done, exit code=0  
2021-02-05 15:48:36.655   --------  
2021-02-05 15:48:36.655   Running command: C:\\Users\\mail\\.espressif\\tools\\idf-python\\3.8.7\\python.exe "C:\\Users\\mail\\esp-idf-v4.2\\tools\\idf\_tools.py" --idf-path C:\\Users\\mail\\esp-idf-v4.2 install  
[... cut ...]
2021-02-05 15:48:37.400   --------  
2021-02-05 15:48:37.400   Message box (OK):  
                         Installation has failed with exit code 1  
2021-02-05 15:48:39.022   User chose OK.  
2021-02-05 15:48:39.034   Message box (Yes/No):  
                         Installation log has been created, it may contain more information about the problem.  
                         Display the installation log now?  
2021-02-05 15:48:40.150   User chose Yes.  
2021-02-05 15:48:40.176   CurStepChanged raised an exception.

DarekLin, 2021-02-07:
I have the same:

2021-02-07 00:48:28.242   Installing Python environment:C:\esp\.espressif\tools\idf-python\3.8.7\python.exe "C:\esp\esp-idf\tools\idf_tools.py" --idf-path C:\esp\esp-idf install-python-env
2021-02-07 00:48:28.719   Done, exit code=1
2021-02-07 00:48:28.719   --------
2021-02-07 00:48:28.719   Running command: C:\esp\.espressif\tools\idf-python\3.8.7\python.exe "C:\esp\esp-idf\tools\idf_tools.py" --idf-path C:\esp\esp-idf install-python-env
                          Creating a new Python environment in C:\esp\.espressif\python_env\idf4.2_py3.8_env
                          FileNotFoundError: [WinError 3] System nie może odnaleźć określonej ścieżki: 'C:\\esp\\.espressif\\tools\\idf-python\\3.8.7\\DLLs'
                          Traceback (most recent call last):
                            File "C:\esp\esp-idf\tools\idf_tools.py", line 1492, in <module>
                              main(sys.argv[1:])
                            File "C:\esp\esp-idf\tools\idf_tools.py", line 1488, in main
                              action_func(args)
                            File "C:\esp\esp-idf\tools\idf_tools.py", line 1207, in action_install_python_env
                              subprocess.check_call([sys.executable, '-m', 'virtualenv', idf_python_env_path],
                            File "subprocess.py", line 364, in check_call
                          subprocess.CalledProcessError: Command '['C:\\esp\\.espressif\\tools\\idf-python\\3.8.7\\python.exe', '-m', 'virtualenv', 'C:\\esp\\.espressif\\python_env\\idf4.2_py3.8_env']' returned non-zero exit status 1.

bot1131357, 2021-02-07:
Me too:
````
5   --------
2021-02-07 21:37:29.033   Installing Python environment:C:\Users\y.tan\.espressif\tools\idf-python\3.8.7\python.exe "C:\Users\y.tan\Desktop\esp-idf-2\tools\idf_tools.py" --idf-path C:\Users\y.tan\Desktop\esp-idf-2 install-python-env
2021-02-07 21:37:30.213   Done, exit code=1
2021-02-07 21:37:30.213   --------
2021-02-07 21:37:30.213   Running command: C:\Users\y.tan\.espressif\tools\idf-python\3.8.7\python.exe "C:\Users\y.tan\Desktop\esp-idf-2\tools\idf_tools.py" --idf-path C:\Users\y.tan\Desktop\esp-idf-2 install-python-env
                          Creating a new Python environment in C:\Users\y.tan\.espressif\python_env\idf4.3_py3.8_env
                          FileNotFoundError: [WinError 3] The system cannot find the path specified: 'C:\\Users\\y.tan\\.espressif\\tools\\idf-python\\3.8.7\\DLLs'
                          Traceback (most recent call last):
                            File "C:\Users\y.tan\Desktop\esp-idf-2\tools\idf_tools.py", line 1655, in <module>
                              main(sys.argv[1:])
                            File "C:\Users\y.tan\Desktop\esp-idf-2\tools\idf_tools.py", line 1651, in main
                              action_func(args)
                            File "C:\Users\y.tan\Desktop\esp-idf-2\tools\idf_tools.py", line 1359, in action_install_python_env
                              subprocess.check_call([sys.executable, '-m', 'virtualenv', idf_python_env_path],
                            File "subprocess.py", line 364, in check_call
                          subprocess.CalledProcessError: Command '['C:\\Users\\y.tan\\.espressif\\tools\\idf-python\\3.8.7\\python.exe', '-m', 'virtualenv', 'C:\\Users\\y.tan\\.espressif\\python_env\\idf4.3_py3.8_env']' returned non-zero exit status 1.
                          
2021-02-07 21:37:30.213   --------
2021-02-07 21:37:30.213   Message box (OK):
                          Installation has failed with exit code 1
2021-02-07 21:37:43.499   User chose OK.
2021-02-07 21:37:43.505   Message box (Yes/No):
                          Installation log has been created, it may contain more information about the problem.
                          Display the installation log now?
2021-02-07 21:37:46.215   User chose Yes.
2021-02-07 21:37:46.273   CurStepChanged raised an exception.
````

DarkRabbit19, 2021-02-07:
Hi I'm having the same probleem.
I dit try to use "esp-idf-tools-setup-online-2.5.exe" and "esp-idf-tools-setup-online-2.4.exe"
To be certain it's not python, I instald the latest version of python (3.9.1) from the website but that was not the solution.
(I can use python to create a new workspace)

when the esp idf tool 2.5 is used I get the following error:
```
2021-02-07 19:12:43.694   --------
2021-02-07 19:12:46.669   idf_tools.py exists in IDF directory
2021-02-07 19:12:46.669   Adding extra Python wheels location. Setting environment variable PIP_EXTRA_INDEX_URL=https://dl.espressif.com/pypi
2021-02-07 19:12:46.669   idf_tools.py command: E:\esp\tools\.espressif\tools\idf-python\3.8.7\python.exe "E:\esp\esp-idf\tools\idf_tools.py" --idf-path E:\esp\esp-idf
2021-02-07 19:12:46.669   Installing tools:E:\esp\tools\.espressif\tools\idf-python\3.8.7\python.exe "E:\esp\esp-idf\tools\idf_tools.py" --idf-path E:\esp\esp-idf install
2021-02-07 19:14:03.975   Done, exit code=0
2021-02-07 19:14:03.975   --------
2021-02-07 19:14:03.975   Running command: E:\esp\tools\.espressif\tools\idf-python\3.8.7\python.exe "E:\esp\esp-idf\tools\idf_tools.py" --idf-path E:\esp\esp-idf install
                          Installing tools: xtensa-esp32-elf, xtensa-esp32s2-elf, esp32ulp-elf, esp32s2ulp-elf, cmake, openocd-esp32, ninja, idf-exe, ccache, dfu-util
                          Installing xtensa-esp32-elf@esp-2020r3-8.4.0
                          Downloading xtensa-esp32-elf-gcc8_4_0-esp-2020r3-win64.zip to E:\esp\tools\.espressif\dist\xtensa-esp32-elf-gcc8_4_0-esp-2020r3-win64.zip.tmp
                          Done
                          Extracting E:\esp\tools\.espressif\dist\xtensa-esp32-elf-gcc8_4_0-esp-2020r3-win64.zip to E:\esp\tools\.espressif\tools\xtensa-esp32-elf\esp-2020r3-8.4.0
                          Installing xtensa-esp32s2-elf@esp-2020r3-8.4.0
                          Downloading xtensa-esp32s2-elf-gcc8_4_0-esp-2020r3-win64.zip to E:\esp\tools\.espressif\dist\xtensa-esp32s2-elf-gcc8_4_0-esp-2020r3-win64.zip.tmp
                          Done
                          Extracting E:\esp\tools\.espressif\dist\xtensa-esp32s2-elf-gcc8_4_0-esp-2020r3-win64.zip to E:\esp\tools\.espressif\tools\xtensa-esp32s2-elf\esp-2020r3-8.4.0
                          Installing esp32ulp-elf@2.28.51-esp-20191205
                          Downloading binutils-esp32ulp-win32-2.28.51-esp-20191205.zip to E:\esp\tools\.espressif\dist\binutils-esp32ulp-win32-2.28.51-esp-20191205.zip.tmp
                          Done
                          Extracting E:\esp\tools\.espressif\dist\binutils-esp32ulp-win32-2.28.51-esp-20191205.zip to E:\esp\tools\.espressif\tools\esp32ulp-elf\2.28.51-esp-20191205
                          Installing esp32s2ulp-elf@2.28.51-esp-20191205
                          Downloading binutils-esp32s2ulp-win32-2.28.51-esp-20191205.zip to E:\esp\tools\.espressif\dist\binutils-esp32s2ulp-win32-2.28.51-esp-20191205.zip.tmp
                          Done
                          Extracting E:\esp\tools\.espressif\dist\binutils-esp32s2ulp-win32-2.28.51-esp-20191205.zip to E:\esp\tools\.espressif\tools\esp32s2ulp-elf\2.28.51-esp-20191205
                          Installing cmake@3.16.4
                          Downloading cmake-3.16.4-win64-x64.zip to E:\esp\tools\.espressif\dist\cmake-3.16.4-win64-x64.zip.tmp
                          Done
                          Extracting E:\esp\tools\.espressif\dist\cmake-3.16.4-win64-x64.zip to E:\esp\tools\.espressif\tools\cmake\3.16.4
                          Installing openocd-esp32@v0.10.0-esp32-20200709
                          Downloading openocd-esp32-win32-0.10.0-esp32-20200709.zip to E:\esp\tools\.espressif\dist\openocd-esp32-win32-0.10.0-esp32-20200709.zip.tmp
                          Done
                          Extracting E:\esp\tools\.espressif\dist\openocd-esp32-win32-0.10.0-esp32-20200709.zip to E:\esp\tools\.espressif\tools\openocd-esp32\v0.10.0-esp32-20200709
                          Installing ninja@1.10.0
                          Downloading ninja-1.10.0-win64.zip to E:\esp\tools\.espressif\dist\ninja-1.10.0-win64.zip.tmp
                          Done
                          Extracting E:\esp\tools\.espressif\dist\ninja-1.10.0-win64.zip to E:\esp\tools\.espressif\tools\ninja\1.10.0
                          Installing idf-exe@1.0.1
                          Downloading idf-exe-v1.0.1.zip to E:\esp\tools\.espressif\dist\idf-exe-v1.0.1.zip.tmp
                          Done
                          Extracting E:\esp\tools\.espressif\dist\idf-exe-v1.0.1.zip to E:\esp\tools\.espressif\tools\idf-exe\1.0.1
                          Installing ccache@3.7
                          Downloading ccache-3.7-w64.zip to E:\esp\tools\.espressif\dist\ccache-3.7-w64.zip.tmp
                          Done
                          Extracting E:\esp\tools\.espressif\dist\ccache-3.7-w64.zip to E:\esp\tools\.espressif\tools\ccache\3.7
                          Installing dfu-util@0.9
                          Downloading dfu-util-0.9-win64.zip to E:\esp\tools\.espressif\dist\dfu-util-0.9-win64.zip.tmp
                          Done
                          Extracting E:\esp\tools\.espressif\dist\dfu-util-0.9-win64.zip to E:\esp\tools\.espressif\tools\dfu-util\0.9
                          
2021-02-07 19:14:03.975   --------
[... cut ...]
2021-02-07 19:14:04.704   --------
2021-02-07 19:14:04.704   Message box (OK):
                          Installation has failed with exit code 1
2021-02-07 19:14:11.652   User chose OK.
2021-02-07 19:14:11.661   Message box (Yes/No):
                          Installation log has been created, it may contain more information about the problem.
                          Display the installation log now?
2021-02-07 19:14:13.940   User chose Yes.
2021-02-07 19:14:13.983   CurStepChanged raised an exception.
```

georgik, 2021-02-08:
@franz-ISOMAX  @bot1131357 @DarkRabbit19 Thank you for reporting the problem.

Please could you provide a content of file ```C:\Users\mail\.espressif\python_env\idf4.2_py3.8_env\pyvenv.cfg``` ?

The installer comes with Embedded Python which does not contain that DLLs directory and the Python should not need it. It seems there is a problem that we need to diagnose.

To workaround the problem, please launch the installer from command line with one additional parameter:
```
esp-idf-tools-setup-2.4.exe /USEEMBEDDEDPYTHON=no
```
The option will turn on Python diagnostic in System Pre-Check and it will let you pick different version of Python.

georgik, 2021-02-08:
Update: 
Please try to download updated version of installer: http://dl.espressif.com/dl/esp-idf-tools-setup-online-2.5.exe?update=2021-08-02
The installer contains one extra correction for creating virtualenv. The PATH to Python for virtualenv is explicitly specified by -p parameter. This should avoid the problem when virtualenv discovers other Python as default.
The command is then:
```
E:\esp\tools\.espressif\tools\idf-python\3.8.7\python.exe -m virtualenv "E:\esp\tools\.espressif\python_env\idf4.2_py3.8_env" -p "E:\esp\tools\.espressif\tools\idf-python\3.8.7\python.exe"
```

DarkRabbit19, 2021-02-08:
@georgik Thank you for your fast response.

Using  esp-idf-tools-setup-online-2.5.exe does not work on it's own but  
`esp-idf-tools-setup-2.5.exe /USEEMBEDDEDPYTHON=no `
does work. 
I can compile an exaple and blink a led

DarekLin, 2021-02-08:
For me, both options did not work. In the attachment you will find the installation log.

[Setup Log 2021-02-08 #002.txt](https://github.com/espressif/esp-idf/files/5946192/Setup.Log.2021-02-08.002.txt)

georgik, 2021-02-09:
@DarekLin Thank you for detailed log, it's very helpful for diagnostics.
The setup 2.4 was able to discover several Anaconda environments which may be cause of the problem. The embedded Python should be able to handle it well.
 
The update is available. We run several tests with latest updated installer and various Anaconda installations. Installer version 2.5 is passing without problem.
Please, download the fresh version of the installer from: http://dl.espressif.com/dl/esp-idf-tools-setup-online-2.5.exe
If the problem persists, please try to delete ```C:\esp\esp-idf install-python-env``` and repeat the installation process.
Please, let me know about the result.

DarekLin, 2021-02-09:
Thanks for you  investigation. I made a new install with fresh installer but problem ocurred:

```
Running command: C:\esp\.espressif\tools\idf-python\3.8.7\python.exe -m virtualenv "C:\esp\.espressif\python_env\idf4.2_py3.8_env" -p "C:\esp\.espressif\tools\idf-python\3.8.7\python.exe"
                          FileNotFoundError: [WinError 3] System nie może odnaleźć określonej ścieżki: 'C:\\esp\\.espressif\\tools\\idf-python\\3.8.7\\DLLs'
```
                          
       [Setup Log 2021-02-09 #002.txt](https://github.com/espressif/esp-idf/files/5949802/Setup.Log.2021-02-09.002.txt)

georgik, 2021-02-09:
@DarekLin Thank you very much for the test. This is very interesting. It seems like virtualenv is loading different Python.

May I ask you to perform following test and post the output:
```
C:\esp\.espressif\tools\idf-python\3.8.7\python.exe -m virtualenv --version
C:\esp\.espressif\tools\idf-python\3.8.7\python.exe -m pip uninstall virtualenv
C:\esp\.espressif\tools\idf-python\3.8.7\python.exe -m pip install virtualenv
C:\esp\.espressif\tools\idf-python\3.8.7\python.exe -m virtualenv --version
C:\esp\.espressif\tools\idf-python\3.8.7\python.exe -m virtualenv  "C:\esp\.espressif\python_env\idf4.2_py3.8_env" -p "C:\esp\.espressif\tools\idf-python\3.8.7\python.exe"
```

DarekLin, 2021-02-09:
```
c:\esp>C:\esp\.espressif\tools\idf-python\3.8.7\python.exe -m virtualenv --version
virtualenv 20.0.31 from C:\Users\darlin\AppData\Roaming\Python\Python38\site-packages\virtualenv\__init__.py

c:\esp>C:\esp\.espressif\tools\idf-python\3.8.7\python.exe -m pip uninstall virtualenv
Found existing installation: virtualenv 20.0.31
Uninstalling virtualenv-20.0.31:
  Would remove:
    c:\users\darlin\appdata\roaming\python\python38\scripts\virtualenv.exe
    c:\users\darlin\appdata\roaming\python\python38\site-packages\virtualenv-20.0.31.dist-info\*
    c:\users\darlin\appdata\roaming\python\python38\site-packages\virtualenv\*
Proceed (y/n)? y
Successfully uninstalled virtualenv-20.0.31

c:\esp>C:\esp\.espressif\tools\idf-python\3.8.7\python.exe -m pip install virtualenv
Requirement already satisfied: virtualenv in .\.espressif\tools\idf-python\3.8.7\lib\site-packages (20.4.0)
Requirement already satisfied: distlib<1,>=0.3.1 in c:\users\darlin\appdata\roaming\python\python38\site-packages (from virtualenv) (0.3.1)
Requirement already satisfied: six<2,>=1.9.0 in c:\users\darlin\appdata\roaming\python\python38\site-packages (from virtualenv) (1.15.0)
Requirement already satisfied: filelock<4,>=3.0.0 in c:\users\darlin\appdata\roaming\python\python38\site-packages (from virtualenv) (3.0.12)
Requirement already satisfied: appdirs<2,>=1.4.3 in c:\users\darlin\appdata\roaming\python\python38\site-packages (from virtualenv) (1.4.4)
WARNING: You are using pip version 20.3.3; however, version 21.0.1 is available.
You should consider upgrading via the 'C:\esp\.espressif\tools\idf-python\3.8.7\python.exe -m pip install --upgrade pip' command.

c:\esp>C:\esp\.espressif\tools\idf-python\3.8.7\python.exe -m virtualenv --version
virtualenv 20.4.0 from C:\esp\.espressif\tools\idf-python\3.8.7\lib\site-packages\virtualenv\__init__.py

c:\esp>C:\esp\.espressif\tools\idf-python\3.8.7\python.exe -m virtualenv  "C:\esp\.espressif\python_env\idf4.2_py3.8_env" -p "C:\esp\.espressif\tools\idf-python\3.8.7\python.exe"
created virtual environment CPython3.8.7.final.0-64 in 3229ms
  creator CPython3Windows(dest=C:\esp\.espressif\python_env\idf4.2_py3.8_env, clear=False, no_vcs_ignore=False, global=False)
  seeder FromAppData(download=False, pip=bundle, setuptools=bundle, wheel=bundle, via=copy, app_data_dir=C:\Users\darlin\AppData\Local\pypa\virtualenv)
    added seed packages: pip==20.2.3, setuptools==51.3.3, wheel==0.36.2
  activators BashActivator,BatchActivator,FishActivator,PowerShellActivator,PythonActivator,XonshActivator

```

georgik, 2021-02-09:
@DarekLin Thank you very much. This is exactly information that I needed.

BTW: Now it should be possible to run the installer because it will reuse the environment created by command line.

DarekLin, 2021-02-09:
@georgik , Thank you. It works! for projects from examples. But when I try to compile another project, I get the following message:
```
CMake Error at CMakeLists.txt:2 (project):
 No CMAKE_C_COMPILER could be found.
 
```
although path is correct.

```
D:\projects\esp32\idf\esphttpd-freertos>echo %IDF_PATH%
C:\esp\esp-idf
```
[buildError.txt](https://github.com/espressif/esp-idf/files/5951114/buildError.txt)

georgik, 2021-02-09:
@DarekLin Thank you very much for all the information and testing. Finally, we were able to simulate and find the root cause of the problem.

Here are steps to reproduce problematic installation. It's necessary to install Python for all users with elevated privileges and pick a directory that won't be writable by the user e.g. C:\Python38. The Python will work. When a user installs anything with pip, the pip will determine that it's not possible to write to C:\Python38 and it will store files to the user's Roaming profile.
The packages in the Roaming profile have priority over those in the local directory. So even when the user removes Python from the system, the roaming data will remain, and the data can be imported by accident by another Python.

There are two solutions:
- remove Python Roaming profile
- set variable PYTHONNOUSERSITE to the directory where site-packages of Python are stored or set to the not-existing directory

DarekLin, 2021-02-09:
still the same error
[buildError2.txt](https://github.com/espressif/esp-idf/files/5951544/buildError2.txt)

georgik, 2021-02-09:
@DarekLin , I'm sorry for confusion. My last comment was not clear. The comment was about solution for the installation process.
The build error seems to be caused by different thing, not necessarily related to Python.
So, let's close to Python part.

Installers 2.4 and 2.5 were updated. They contain fix for installation process and IDF launchers - PowerShell and CMD contains the correction for variable PYTHONNOUSERSITE.

After this post, we can continue in checking the problem with esphttpd-freertos.

bot1131357, 2021-02-09:
Thanks @georgik 

Removing `%APPDATA%\Python\Python38` allowed me to install successfully.

DarekLin, 2021-02-10:
@georgik So, if possible, I would come back to the problem of not finding cmake. This is a somewhat strange problem, possibly not installer related as other programs compile without problem. So if you could take the time and see the log (3 posts above) I would be grateful.
```
CMake Error at CMakeLists.txt:2 (project):
  No CMAKE_C_COMPILER could be found.
```
CMakeLists.txt:

```
cmake_minimum_required(VERSION 3.5)
include("$ENV{IDF_PATH}/tools/cmake/project.cmake")
project(esphttpd-freertos)

```
```
D:\projects\esp32\idf\esphttpd-freertos>echo %IDF_PATH%
C:\esp\esp-idf
```

georgik, 2021-02-10:
@DarekLin Unfortunately even after investing several hours to investigation of esphttpd-freertos Windows build we are not able to resolve the problem. The build was tested even with several versions of CMake.
It seems that build of esphttpd-freertos on Windows https://github.com/chmorgan/esphttpd-freertos is broken. The problem is caused by submodules libespfs. The build works well on Linux.

Please, report the problem to maintainers of those projects.

georgik, 2021-02-10:
@franz-ISOMAX Please, download updated version of 2.5 installer
http://dl.espressif.com/dl/esp-idf-tools-setup-online-2.5.exe
Please confirm whether new installer solves the reported problem.

DarekLin, 2021-02-10:
@georgik, Thank you very much for solving the first problem and taking the time to try to solve the second problem. Especially when it turned out that it was not the installer's problem.

georgik, 2021-03-11:
@DarekLin I made further investigation of the problem of esphttp-freertos on Windows. The root cause is not obvious from the log, but it's pretty interesting.

The fail is caused by submodule espfs which contains C code of makefsimage. This application is used during build process in tools/build-image.py to package files produced by NodeJS. The C code of makefsimage is not compiled for ESP, it's compiled for the platform running the build process. E.g. in case of Windows. This is done by declaration in espfs/CMakeLists.txt:
```
externalproject_add(mkespfsimage...
```
Logically CMake tries to find toolchain for compiling for Windows. It won't find any tool chain so it will fail. In case of Linux build it will find some local gcc which will work.

Possible untested solutions:
- install Visual C++ and run build process
- replace dependency on binary makefsimage by something spiffs_create_partition_image which uses Python implementation already present in ESP-IDF

More information about externalproject_add: https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/build-system.html#fully-overriding-the-component-build-process

georgik, 2021-03-11:
Here is way how to patch the project to be buildable under Windows:
```
choco install mingw
choco install make
```
Open file components/espfs/CMakeLists.txt and change definition of external project to: 
```
externalproject_add(mkespfsimage
    SOURCE_DIR "${COMPONENT_DIR}/mkespfsimage"
    BUILD_IN_SOURCE 1
    BUILD_COMMAND make CC=gcc
    CMAKE_ARGS
        "-DCMAKE_INSTALL_PREFIX=${CMAKE_CURRENT_BINARY_DIR}"
        "-DUSE_HEATSHRINK=${CONFIG_ESPFS_USE_HEATSHRINK}"
        "-DUSE_GZIP_COMPRESSION=${CONFIG_ESPFS_USE_GZIP}"
)
```
In the same file search for ```ln -s```, change it to ``` echo ln -s```. That's because Windows has no notion of symlinks. You'll need to copy files from NodeJS/Babel manually or replace it by relevant command.

Run idf.py build. 
Majority of build will pass, you need just to get xxd working.

Basically esphttpd-freertos is not written to support Windows build. Please open issue on the project with suggestion to implement corrections mentioned above.

franz-ms-muc, 2021-03-19:
this is solved in esp-idf-tools-setup-online-2.5.exe

franz-ISOdevECU, 2021-03-19:
CLOSE!

closed 2021-03-19 by franz-ISOdevECU