espressif/esp-idf #6513: Classic Bluetooth[esp32]  In Iphone to not scanning

sowmimalu, 2021-02-07:
Hai ,
ESP32 WROVER [16mb]
espidf - 4.0 dev
 I have used cLassic Bluetooth in esp32 to scan my esp32 device then connected to send a data , it was successfully working . on the same iphone  unable to scan the device itself , which example is preferrable to conenct esp32 device bluetooth to pair with iphone?

boblane1, 2021-02-08:
Hi @sowmimalu,

Do you want to use iPhone to connect SPP acceptor on ESP32? If so, I am sorry to tell you that SPP is not supported on iPhone. You can use a2dp for audio or use BLE gatt to send data.

Thanks.

sowmimalu, 2021-02-09:
Thanks for the reply , i need to write  via bluetooth , possible to connect with Bluetooth in iphone ?

boblane1, 2021-02-09:
Hi @sowmimalu，

As I said before, I suggest you can use BLE instead. ESP32 offers demo like [gatt_client](https://github.com/espressif/esp-idf/tree/master/examples/bluetooth/bluedroid/ble/gatt_client) and [gatt_server](https://github.com/espressif/esp-idf/tree/master/examples/bluetooth/bluedroid/ble/gatt_server), iPhone has Bluetooth debug tools like 'LightBlue', you can have a try.

Thanks!

StavrosD, 2021-02-09:
If you want to connect esp32 with iPhone using bluetooth, you have two options:
1. Use BLE and CoreBluetooth
2. Use classic bluetooth and enroll the Apple MFi program https://mfi.apple.com . You have to certify your device for production, otherwise your app will not have access to bluetooth classic.

sowmimalu, 2021-02-23:
Hai ,  tried with a2dp sink example BT has enabled i can able to scan and pair with ios and aslo andriod but the example is for audio transfer is this possible to transfer data after connection ?

sowmimalu, 2021-02-26:
hello ,
  Found BLufi enable to connect with ios and writing data through BLE but in my application scanning also applicable so used GATTC client example to connect with other BLE devices but now after adding Blufi unable to connect gattc client , same time its not working , anyway to work both ?

sowmimalu, 2021-03-01:
its working, closing this issue. thanks

closed 2021-03-01 by sowmimalu

LakshaySahni, 2022-07-30:
@sowmimalu I'm facing the same issue, how did you connect esp32 bluetooth to iphone?

sowmimalu, 2022-08-05:
@LakshaySahni  i am using BLE to connect with Iphone Blufi example .

summerfind, 2023-03-16:
Hi, I am using iphone 14 with Bluetooth 5.3 still not be able to find ESP32S3 which is BLE, how to fix? Thanks.