espressif/esp-idf #6514: RTC slow memory is not automatically powered down

boarchuz, 2021-02-07:
## Environment

- Target: ESP32
- IDF version: master (v4.3-dev-2940-g6e776946d)
- Build System: CMake
- Compiler version: 8.4.0
- Operating System: Linux
- IDE: VSCode + Extension

## Problem Description

RTC slow memory is always powered up in sleep when PD option is AUTO, even when conditions are met for powering down.

`(size_t) &_rtc_slow_length > 0` always evaluates to true in get_power_down_flags here:

https://github.com/espressif/esp-idf/blob/6e776946d01ec0d081d09000c36d23ec1d318c06/components/esp_system/sleep_modes.c#L1146-L1152

even when it logically shouldn't. This seems to be a compiler optimisation (&anything != 0?) or undefined behaviour? 

See this simple demo: 
```
#include <stdint.h>
#include <stdio.h>

#include "esp_sleep.h"

extern int _rtc_slow_length;

void app_main()
{
    printf("_rtc_slow_length: %u\n", (size_t) &_rtc_slow_length);
    if((size_t) &_rtc_slow_length > 0)
    {
        printf("_rtc_slow_length > 0\n");
    }
    esp_deep_sleep_start();
}
```
Output:

> _rtc_slow_length: 0
> _rtc_slow_length > 0
> D (704) sleep: RTC_PERIPH: AUTO(OFF)
> D (714) sleep: RTC_SLOW_MEM: ON
> D (714) sleep: RTC_FAST_MEM: ON

## Other
We could probably auto power down fast memory too if s_config.wakeup_triggers == 0 (as in the above)?

lennyJL, 2022-04-07:
Hi boarchuz. I can't reproduce this issue on the latest release/v4.3 using esp32. Can you try it again on the latest release/v4.3 with your hardware？

boarchuz, 2022-04-07:
Hi @lennyJL,
Thanks for looking at this. 
IDF now always places a couple variables in RTC memory so this has no effect any more. ie. 'hibernation' is impossible with an unmodified IDF.

https://github.com/espressif/esp-idf/blob/d9be879a768d798961d3ea9aeb0cbd75a76d4a58/components/newlib/port/esp_time_impl.c
For the purposes of testing, I removed the `RTC_DATA_ATTR` attribute from the above so that &_rtc_slow_length==0, and confirmed that this still occurs with v4.3.2-473-gd9be879a76.

I think it would be necessary to do something like `&_rtc_force_slow_end - &_rtc_force_slow_start` to ensure that a length of 0 is handled correctly.

lennyJL, 2022-04-11:
Hi @boarchuz. You are right. I checked the disassembly file and found that compiler considers `(size_t) &_rtc_slow_length > 0` to always be true. We can use a volatile variable to hold the value of `&_rtc_slow_length` and prevent the compiler from doing this optimization.

Alvin1Zhang, 2022-05-16:
Thanks for reporting, 

* Fix on master branch is available https://github.com/espressif/esp-idf/commit/e503ebf5f99198ae90c7b131dffdf1f1a4b64cc4
* Fix on release/4.4 is available https://github.com/espressif/esp-idf/commit/b6f4629e119fa18714d5274682bd0aa5f5a3b89f
* Fix on release/4.3 is available https://github.com/espressif/esp-idf/commit/3207993eb78951ceb3bf011f8a04bd62771c8ab4. 

Feel free to reopen.

closed 2022-05-16 by Alvin1Zhang