espressif/esp-idf #6515: RTC slow memory issues

fpgamaster, 2021-02-07:
----------------------------- Delete below -----------------------------

- Module or chip used:  ESP32-WROOM-32D
- IDF version: v4.1
- Build System: Make
- Compiler version : xtensa-esp32-elf-gcc (crosstool-NG esp-2020r2) 8.2.0
- Operating System: Linux
- Power Supply: external 3.3V

## Problem Description
As reported here: [10526](https://esp32.com/viewtopic.php?t=10526) the length of the rtc_slow_reg looks like is incorrect!
The Igor's reply does not make sense to me since the RTC slow memory length is 8k which is 0x2000 in hex.

In our config we have CONFIG_ULP_COPROC_RESERVE_MEM=7936 since the ULP code stores a lot of data.
Also, we have variables marked with RTC_SLOW_ATTR which needs to be preserved during the deep sleep.

So, which is incorrect KConfig or ld script?

Best regards

negativekelvin, 2021-02-08:
https://github.com/espressif/esp-idf/commit/ef10c2576ff14afa033ef22105406399abc570af

AxelLin, 2022-02-27:
@igrr 
This report is against v4.1, but v4.1 (and v4.2) does not include the fix.

ESP-Marius, 2023-04-26:
Fix is available on 4.2 now.

closed 2023-04-26 by ESP-Marius