espressif/esp-idf #6517: ASSERT_PARAM(-218959118 0), in arch_main.c at line 343

ROYMURLI, 2021-02-08:
IDFv- v4.3-dev-2586-g526f68239
OS- Ubuntu 

Application- I am using GATTS and GATTC in my profile along with with Wifi STA mode also. In GATTS my device should advertise its name all the time and any client can connect to that device through BLE using app (BLE Scanner). In GATTC my device will connect to a BLE server sensor in every 45 second for a second and will take sensor data and disconnect. Wifi is used to upload the data on server. I have created my own code using the help of IDF examples.

Issue- After some time (may be after 1 hour/3 hour/12 hoours/ a day some of my device got this error ```ASSERT_PARAM(-218959118 0), in arch_main.c at line 343``` and everything stops working. 

I have tried by changing this ```.scan_duplicate = BLE_SCAN_DUPLICATE_ENABLE``` from DISABLE to ENABLE, but it is giving same error. 

My SDKCONFIG is attched below.

[sdkconfig.txt](https://github.com/espressif/esp-idf/files/5942225/sdkconfig.txt)

fhelie, 2021-02-09:
I have same error, also looking for a fix

eriknorth, 2021-02-11:
Can you please post crash log with stracktrace.

Digitelektro, 2021-02-17:
Similar problem here! I am doing an OTA firmware update through BLE and when I initialize the flash, sometimes I get this error:

```
ASSERT_PARAM(-218959118 0), in arch_main.c at line 343
Guru Meditation Error: Core  0 panic'ed (Interrupt wdt timeout on CPU0).

Core  0 register dump:
PC      : 0x4008342b  PS      : 0x00060134  A0      : 0x8013d2fc  A1      : 0x3ffbe260
0x4008342b: r_assert_param at ??:?

A2      : 0x00000001  A3      : 0x00000000  A4      : 0x60008048  A5      : 0x00000000
A6      : 0x00000004  A7      : 0x3ffbdbe4  A8      : 0x80083428  A9      : 0x3ffbe240
A10     : 0x00000000  A11     : 0x00000037  A12     : 0x00000014  A13     : 0xffffffff
A14     : 0x00000000  A15     : 0xfffffffc  SAR     : 0x00000004  EXCCAUSE: 0x00000005
EXCVADDR: 0x00000000  LBEG    : 0x40083360  LEND    : 0x40083367  LCOUNT  : 0x00000000
0x40083360: r_assert_param at ??:?

0x40083367: r_assert_param at ??:?

Core  0 was running in ISR context:
EPC1    : 0x401a0423  EPC2    : 0x00000000  EPC3    : 0x00000000  EPC4    : 0x4008342b
0x401a0423: uart_hal_write_txfifo at D:/Programozas/Nowtech/esp-idf/components/soc/src/hal/uart_hal_iram.c:35

0x4008342b: r_assert_param at ??:?

Backtrace:0x40083428:0x3ffbe260 0x4013d2f9:0x3ffbe280 0x40019fb5:0x3ffbe2a0 0x40087a5a:0x3ffbe2d0 0x400863d9:0x3ffbe310 0x400860f5:0x3ffbe330 0x400887e6:0x3ffbe350 0x40089927:0x3ffbe370 0x40082a81:0x3ffbe390 0x4013d734:0x3ffb5750 0x4013dd01:0x3ffb5770
0x40083428: r_assert_param at ??:?

0x4013d2f9: r_platform_reset at ??:?

0x40087a5a: r_lld_pdu_rx_handler at ??:?

0x400863d9: r_lld_evt_end at ??:?

0x400860f5: r_lld_evt_end_isr at ??:?

0x400887e6: r_rwble_isr at ??:?

0x40089927: r_rwbtdm_isr_wrapper at intc.c:?

0x40082a81: _xt_lowint1 at D:/Programozas/Nowtech/esp-idf/components/freertos/xtensa/xtensa_vectors.S:1105

0x4013d734: r_rw_schedule at ??:?

0x4013dd01: btdm_controller_task at ??:?

Core  1 register dump:
PC      : 0x4019d0a6  PS      : 0x00060034  A0      : 0x800d46e0  A1      : 0x3ffbc7d0
0x4019d0a6: esp_pm_impl_waiti at D:/Programozas/Nowtech/esp-idf/components/esp32/pm_esp32.c:486

A2      : 0x00000000  A3      : 0x00000001  A4      : 0x00000014  A5      : 0x000000a5
A6      : 0x000000a5  A7      : 0x00060023  A8      : 0x800d45a2  A9      : 0x3ffbc7b0
A10     : 0x00000000  A11     : 0x00000001  A12     : 0x00000014  A13     : 0x000000a5
A14     : 0x000000a5  A15     : 0x00060023  SAR     : 0x00000000  EXCCAUSE: 0x00000005
EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000

Backtrace:0x4019d0a3:0x3ffbc7d0 0x400d46dd:0x3ffbc7f0 0x4009159b:0x3ffbc810
0x4019d0a3: esp_pm_impl_waiti at D:/Programozas/Nowtech/esp-idf/components/esp32/pm_esp32.c:484

0x400d46dd: esp_vApplicationIdleHook at D:/Programozas/Nowtech/esp-idf/components/esp_common/src/freertos_hooks.c:63

0x4009159b: prvIdleTask at D:/Programozas/Nowtech/esp-idf/components/freertos/tasks.c:3386

ELF file SHA256: 006fbba9b4e7e058

Rebooting...

```

eriknorth, 2021-02-18:
Hi,

After extensive testing and bisecting, we might have pin pointed the fault.
It seems that crashing started at BT Controller commit:
espressif/esp32-bt-lib@85be5e8
This led us to check flags related to sleep modes. After disabling CONFIG_BTDM_MODEM_SLEEP, we were not able to replicate the crash anymore.

It is unfortunate that BT Lib is closed source which makes debugging issues like this very time consuming. And even now after finding the issue we are not sure why this flag exactly fixes the issue.

Please, whoever is experiencing this issue, check if this flag also mitigates/fixes the crash.

/Erik

Digitelektro, 2021-02-18:
Thank you so much! Will test it, and see, I will get back to you, as it takes several test to see it is completely solved!

eriknorth, 2021-02-24:
@Digitelektro can you please update us if CONFIG_BTDM_MODEM_SLEEP fixed the issue for you.

ROYMURLI, 2021-02-24:
For me I am not getting this error for now:-
1- Upgrade your master to v4.3-dev-2586-g526f68239
2- Change ```.scan_duplicate = BLE_SCAN_DUPLICATE_ENABLE``` from Disable to Enable. 

Hope it will work.

Digitelektro, 2021-02-24:
> @Digitelektro can you please update us if CONFIG_BTDM_MODEM_SLEEP fixed the issue for you.

Yes, it solved the problem!

closed 2021-02-24 by ROYMURLI

alexmantaut, 2021-03-05:
Hi, we have the same problem... The same assert fails and disabling CONFIG_BTDM_MODEM_SLEEP stops the problem from occurring...

However, battery consumption increases dramatically by disabling modem sleep, which makes this workaround not suitable for our battery powered application... do you have any other workarounds/ways to fix?

This error started occurring after we updated to the 4.2 version of the idf, previous versions used to work ok, but we needed to update the IDF version because of a memory leak...

alexmantaut, 2021-04-15:
Looking into the esp32-bt-lib, it seems they fixed there is a [fix](https://github.com/espressif/esp32-bt-lib/commit/1541ed995ba21efd898c84dd95660d59d6746fd7#diff-e2952694dec272e5ab5bbdc6e69c11d9927b76d8d18eedaf3562826c8a812f90) for the issue on the BT library.

Looking into the current releases, the code has been merged into master [here](https://github.com/espressif/esp-idf/commit/e5f8e12db1202921c85067db937d6a198976ba95#diff-42c8b4c57b09d720ffed3ffc3df6b69e6ad15271d00714bb4d904c1df92d7264), but the changes don't seem to have been ported to 4.2 and 4.3.

I tried the changes from master, and I don't get the crashes anymore, even with CONFIG_BTDM_MODEM_SLEEP enabled.

@eriknorth Any chance the changes can be ported into 4.2/4.3 releases?

eriknorth, 2021-04-15:
Hi, that is good news that they have made a fix for the issue. I do not work for Espressif so I do not have any knowledge of them back-porting the fix to 4.2, but I really hope they do.

jmaha, 2021-08-02:
@eriknorth The changelog reports this issue being fixed in the 4.2 releases as of 4.2.1, but I just experienced this crash using ESP-IDF 4.2.2. So apparently this bug isn't fixed, or perhaps it was reintroduced in 4.2.2?

```
ASSERT_PARAM(-218959118 0), in arch_main.c at line 343
Guru Meditation Error: Core  0 panic'ed (Interrupt wdt timeout on CPU0).

Core  0 register dump:
PC      : 0x40083701  PS      : 0x00060134  A0      : 0x800e8c14  A1      : 0x3ffbb590
A2      : 0x00000001  A3      : 0x00000000  A4      : 0x60008048  A5      : 0x00000000
A6      : 0x00000004  A7      : 0x3ffbdcb8  A8      : 0x800836fc  A9      : 0x3ffbb570
A10     : 0x0000f2f2  A11     : 0x00000037  A12     : 0x00000014  A13     : 0xffffffff
A14     : 0x00000000  A15     : 0xfffffffc  SAR     : 0x00000004  EXCCAUSE: 0x00000005
EXCVADDR: 0x00000000  LBEG    : 0x40083634  LEND    : 0x4008363b  LCOUNT  : 0x00000000

Backtrace:0x400836fe:0x3ffbb590 0x400e8c11:0x3ffbb5b0 0x40019fb5:0x3ffbb5d0 0x4001a1f2:0x3ffbb600 0x40019471:0x3ffbb620 0x400175a7:0x3ffbb660 0x40019d11:0x3ffbb680 0x40055b4d:0x3ffbb6a0 0x400e8fd8:0x3ffbb6c0 0x400e95cd:0x3ffbb6e0 0x40090485:0x3ffbb710

ELF file SHA256: 903b50dd5d46b7dc

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:7164
ho 0 tail 12 room 4
load:0x40078000,len:13212
load:0x40080400,len:4564
entry 0x400806f4
I (29) boot: ESP-IDF v4.2.2 2nd stage bootloader
I (29) boot: compile time 11:52:06
I (29) boot: chip revision: 1
I (32) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (39) boot.esp32: SPI Speed      : 40MHz
I (43) boot.esp32: SPI Mode       : DIO
I (48) boot.esp32: SPI Flash Size : 4MB
I (52) boot: Enabling RNG early entropy source...
I (58) boot: Partition Table:
I (61) boot: ## Label            Usage          Type ST Offset   Length
I (69) boot:  0 nvs              WiFi data        01 02 00009000 00004000
I (76) boot:  1 otadata          OTA data         01 00 0000d000 00002000
I (84) boot:  2 phy_init         RF data          01 01 0000f000 00001000
I (91) boot:  3 factory          factory app      00 00 00010000 00100000
I (99) boot:  4 ota_0            OTA app          00 10 00110000 00100000
I (106) boot:  5 ota_1            OTA app          00 11 00210000 00100000
I (114) boot:  6 storage          WiFi data        01 02 00310000 0007d000
I (121) boot: End of partition table
E (126) boot: ota data partition invalid, falling back to factory
```

jmaha, 2021-08-02:
If it helps, here's some additional information about my test case:

- ESP32-based product sitting next to nRF52833-based device. ESP32 connects to the nRF and retrieves data by subscribing to a characteristic. The data arrives 1/sec.
- I had the two devices sitting ~1 inch apart. The ESP32 ran for over 72 hours without resetting, and I was optimistic.
- I moved the ESP32 ~8 inches away from the nRF device, and suddenly it began resetting with this assert. It's intermittent but seems to last only 2-3 hours before resetting.

AxelLin, 2021-08-06:
> @eriknorth The changelog reports this issue being fixed in the 4.2 releases as of 4.2.1, but I just experienced this crash using ESP-IDF 4.2.2. So apparently this bug isn't fixed, or perhaps it was reintroduced in 4.2.2?

@Alvin1Zhang 
No idea why the changelog of 4.2.1 mentioned the fix.
I don't find the fix in release/v4.2 branch of esp32-bt-lib.

matt001k, 2021-09-30:
So I'm curious if I can still get this issue after CONFIG_BTDM_MODEM_SLEEP is disabled. I did this and it for sure still caused the issue.

AxelLin, 2021-09-30:
> So I'm curious if I can still get this issue after CONFIG_BTDM_MODEM_SLEEP is disabled. I did this and it for sure still caused the issue.

Which esp-idf version are you using?

matt001k, 2021-09-30:
@AxelLin esp-idf 4.2.0

AxelLin, 2021-09-30:
> @AxelLin esp-idf 4.2.0

It should be fixed in v4.2.2. Check if v4.2.2 works for you.

AxelLin, 2021-10-08:
> @AxelLin Still actually getting this error in 4.3.1 traceback:
> 

I think you need to report a new issue since this one was closed.

alexmantaut, 2021-11-01:
We are also seeing this on 4.3.1... It is definitely less frequent than it used to be, but it still happens

AxelLin, 2021-11-01:
> We are also seeing this on 4.3.1... It is definitely less frequent than it used to be, but it still happens

@Alvin1Zhang 
Do you suggest to create a new issue or re-open this issue?

alexmantaut, 2021-11-01:
I think this issue was created for that https://github.com/espressif/esp-idf/issues/7805