espressif/esp-idf #6518: ESP-Mesh nodes loose connection with root when (fixed) root is moved

sinope-db, 2021-02-08:
## Environment
- Development Kit:      none
- Module or chip used:  ESP32-WROOM-32D
- IDF version:          v3.3.4 
                        w/ ESP-MDF Release v1.0 
- Build System:         Make
- Compiler version:     1.22.0-80-g6c4433a5

## Problem Description

When using a fixed root node, some nodes will loose the connection with the root after the root was moved to a new physical location (within range of the nodes). The missing nodes are within range of other nodes that have connection with the root, and have free children capabilities.

This will cause the nodes to be unavailable to the external world (web server) and to the nodes that kept their connection with the root.

### Expected Behavior

When the root returns, the mesh network rebuilds itself. 
The branches that lost connection to the root should 

### Actual Behavior

Some nodes are stuck in "dead" branches (without connection to the root) while they could connect to neighbors that have a connection to the "new" root.
I have to power cycles the unavailable nodes for them to reconnect properly.

### Steps to reproduce

setup:
- Mesh is configured to use a fixed root node.
- 25 layers
- 6 children nodes per parent / AP.
- 98 nodes distributed across 9 locations in the building (single floor).
- distance is too far (signal is too weak) for nodes from one end of the building to directly communicate with nodes on the other end.

[image]

1. Power up the devices with root node placed at one end of the mesh (location A) and let the mesh network establish.
2. Root node is powered down and moved to the other end of the mesh (location B)
3. After a while (hours), some nodes are still not connected to the root.
3.1 Unavailable nodes are located near location A.
4. Root node is powered down and moved to the original location (location A).
5. After a while, the missing nodes are back, but some nodes that were available after the first switch are now unavailable.
5.1 Unavailable nodes are now located near location B.

## Debug Logs

Log from one of the device that was connected directly on the root (at location A)
[image]

shenjun7, 2022-07-27:
After the fix-root network root disappears, the layer 2 nodes will always scan and wait for the root to appear, and the identity of the layer 2 nodes will not be invalidated. 
The long-distance position adjustment of the root leads to the appearance of layer 2 nodes in the A and B area. Layer 2 nodes across regions cannot give up their identities, resulting in loss of connection. 
We can make the layer 2 nodes give up their identity to connect to lower-layer nodes to solve the problem of unavailable nodes.

sinope-db, 2022-07-27:
@shenjun7  

I think it would be great if layer 2 nodes would give up their identities after loosing the root for a certain period of time.

Q1. does this behavior only applies to layer 2 nodes?  In other words, would a layer node that loose its parent be able to reconnect to a layer 4+ node?

Q2. Is the connection status with the root available for potential parents when a node scan the mesh?

shenjun7, 2022-07-29:
A1. only applies to layer 2 nodes. For the fix-root network, the network topology will not change after the root disappears. If you want a flexible root node, you can use self-organizing network.
A2. node scanning mesh will receive `<MESH_EVENT_NETWORK_STATE>is_rootless: ?` event.

sinope-db, 2022-07-29:
@shenjun7 
For my second question I meant does a device that performs a WiFi scan for mesh neighbors / parents have the rootless status for each entry in the results?

MisterScience5, 2022-08-14:
Hi, I also support letting layer 2 nodes disconnect and retry connecting after a timeout, as this design is creating a lot of trouble with our way of running a mesh. We run the mesh intermittently, and then restart the controllers after the mesh phase is complete, but if some controllers have dropped off the root they will not be ordered to restart, and then start "dead" branches of the mesh which tend to survive over the long term until the controllers are manually rebooted at the same time.

I just recently discovered this post, and tried to add a disconnect on the is_rootless event, however my devices don't seem to receive this event when connecting to an existing dead branch.

I will have to add some spaghetti to verify that a mesh connection actually means that there is a root connection, however this is a lot of uneccessary work if you are considering changing this design.

shenjun7, 2022-08-17:
If the location of the router or root node is not fixed, it is better to use a self-organized network to avoid dead branch.

MisterScience5, 2022-08-17:
In my case this is complicated as I'm using a modem only on the root device to synchronize data from the full mesh.
However, I don't believe that the current design that allows connections to orphaned layer 2 nodes indefinitely in a fixed root network is ideal, although I may be missing something.
Also the is_rootless event does not seem to propagate properly in a fixed root network, is it supposed to do so?

sinope-db, 2022-08-22:
Similar use case here, where the network is self organized except for the root node because the application requires more memory. This allowed us to use existing hardware and keep the cost down on the nodes.

Alvin1Zhang, 2022-08-31:
Thanks for reporting, feel free to reopen.

MisterScience5, 2022-08-31:
I don't believe this is done, the way mesh works with a fixed root is clearly a poor/bugged implementation, as well as not being reflected in the docs.

sinope-db, 2022-08-31:
@Alvin1Zhang Why close the issue? What is the resolution? Will there be change done or does Espressif reject the request?

Alvin1Zhang, 2022-09-01:
@MisterScience5 @sinope-db Sorry for closing. Have reopened just now.

shenjun7, 2022-09-01:
Sorry for the late reply. We will change the behavior of the layer2 node in the fixed-root scenario. The layer2 node will become idle state when the fixed-root disappears for a while and then avoid dead branch. The fixed-root documentation will be updated later.

MisterScience5, 2022-09-25:
Any progress update on this? We have a lot of units in production suffering from the problem, so I am hoping not to have to make a temporary workaround:)

KonssnoK, 2022-09-28:
interested in this issue too

MisterScience5, 2022-10-17:
I am in the process of making a workaround for this now, as we are pressed for time, and I have made a routine that immediately reboots a controller that is layer 2 when it loses root connection, which should solve this for the future.

I am wondering though how I can detect on an individual node that it is connecting to a dead branch? The nodes do not receive the is_rootless event, causing every device to have to be simultaneously shut down to remove dead branches. (I just did a shutdown of 45 units, and forgot one, causing every single node on the network to connect to the dead branch...)

I need to fix this somehow, as our devices are deployed remotely and it is very costly and impractical to do simultaneous shutdowns, so any ideas are very welcome.

KonssnoK, 2022-10-17:
in theory there is the event `MESH_EVENT_TODS_REACHABLE` which should report if the external network is reachable, but i haven't seen it working properly as far as i can remember

MisterScience5, 2022-10-17:
That's true, although I don't have an active connection, I did notice that the dead branches do not give me this event, while the live branch does indicating that this message originates from the root - I will try and use it for a flag to indicate that there is an active root on the branch:)

KonssnoK, 2022-10-17:
yeah but i was not able to see it switching between"reachable" and not whenever i switch on and off the hotspot of my network...
So probably there's a bug somewhere (or i didn't do the test correctly)

MisterScience5, 2022-10-17:
My solution so far has been to implement a task that sends a minor message to root with esp_mesh_send_block_time(10000), if it times out 3 times in a row I presume we're in a dead branch and reboot. Since this runs on all nodes, I think it should clear out all the dead branches fairly effectively.

MisterScience5, 2022-10-20:
A quick update for anyone working on the same issue, my proposed workaround didn't work, as it seems the function sometimes returns ESP_OK even in a dead branch, so I now do a broadcast from the root every 10 seconds when the mesh is on, and reboot any units that haven't seen that broadcast in 60 seconds, this seems to work to keep dead branches from forming, but does sometimes kick off units that are in the correct mesh, presumably because of signal propagation issues.

ClaesIvarsson, 2023-01-23:
Any update on this issue?

KonssnoK, 2023-01-24:
@ClaesIvarsson 
afaik (as a client) there are currently more pressing issues on the mesh library.
Right now there is Chinese new year, so you should wait next week.

zhangyanjiaoesp, 2023-01-30:
@MisterScience5 @ClaesIvarsson @sinope-db will provide the solution ASAP.

zhangyanjiaoesp, 2023-02-06:
@sinope-db Based on ESP-IDF v3.3.4, provide a solution for this issue. Please check the attachment.
wifi firmware version: 3214e4b
[wifi_lib_v3.3.4_0206.zip](https://github.com/espressif/esp-idf/files/10658144/wifi_lib_v3.3.4_0206.zip) replace these in `idf/components/esp32/lib`.

Please have a test and feedback whether the issue can be solved, thanks.

zhangyanjiaoesp, 2023-02-06:
@ClaesIvarsson You are in v4.4.3, which chip do you use? ESP32 or ESP32S3 or any others?

ClaesIvarsson, 2023-02-06:
> @ClaesIvarsson You are in v4.4.3, which chip do you use? ESP32 or ESP32S3 or any others?

I am using ESP32 using v4.4.3.

zhangyanjiaoesp, 2023-02-06:
@ClaesIvarsson Based on IDF v4.4.3(6407ecb3f8d2cc07c4c230e7e64f2046af5c86f7), provide a solution for this issue. Please check the attachment.
wifi firmware version: d4c5866
[wifi_lib_v4.4.3_0206.zip](https://github.com/espressif/esp-idf/files/10661148/wifi_lib_v4.4.3_0206.zip)

Please have a test and feedback whether the issue can be solved, thanks.

ClaesIvarsson, 2023-02-06:
Now I have been running some initial test with this fix on my system.
I have seen multiple occasions where a node (*n1*) thinks it is connected to another node (*n2*, not root) but the *n2* is not recognizing the *n1* as a child. *n1* is then not trying to connect to another and *n2* is not aware of *n1*, meaning it gets stuck in unreachable state.

sinope-db, 2023-02-06:
@zhangyanjiaoesp  thank you for providing a 3.3.4 lib to test, we'll test it as soon as we can.

zhangyanjiaoesp, 2023-02-07:
> I have seen multiple occasions where a node (_n1_) thinks it is connected to another node (_n2_, not root) but the _n2_ is not recognizing the _n1_ as a child. _n1_ is then not trying to connect to another and _n2_ is not aware of _n1_, meaning it gets stuck in unreachable state.

@ClaesIvarsson Does this happens when the fixed root is moved? Please provide the complete logs of the two nodes, thanks.

ClaesIvarsson, 2023-02-07:
> @ClaesIvarsson Does this happens when the fixed root is moved? Please provide the complete logs of the two nodes, thanks.

Log from node that used to be connected to the root:
```
I (309186) mesh: 5118<assoc>parent layer:1, channel:1, rssi:-35, assoc:6, rssi threshold<-78,-82,-85>
W (313140) mesh: [mesh_schedule.c,3130] [WND-RX]max_wnd:9, 1200 ms timeout, seqno:157, xseqno:132, no_wnd_count:94, timeout_count:0
W (314340) mesh: [mesh_schedule.c,3130] [WND-RX]max_wnd:9, 1200 ms timeout, seqno:157, xseqno:132, no_wnd_count:94, timeout_count:1
I (315186) wifi:state: run -> init (6c0)
I (315187) wifi:pm stop, total sleep time: 0 us / 92753959 us

I (315188) wifi:new:<1,1>, old:<1,1>, ap:<1,1>, sta:<1,1>, prof:1
I (315190) mesh: [wifi]disconnected reason:6(non-authenticated STA), continuous:1/max:12, non-root, vote(,)<><>
I (315194) wifi:primary chan differ, old=1, new=1, start CSA timer
I (315197) wifi:new:<1,1>, old:<1,1>, ap:<1,1>, sta:<1,1>, prof:1
I (315200) wifi:state: init -> auth (b0)
I (315203) wifi:state: auth -> init (6c0)
I (315204) wifi:new:<1,1>, old:<1,1>, ap:<1,1>, sta:<1,1>, prof:1
W (315540) mesh: [mesh_schedule.c,3130] [WND-RX]max_wnd:9, 1200 ms timeout, seqno:157, xseqno:132, no_wnd_count:94, timeout_count:2
W (365164) wifi:Haven't to connect to a suitable AP now!

I (482447) wifi:station: fc:f5:c4:3c:fb:70 leave, AID = 2, bss_flags is 131171, bss:0x3f83a208
I (482447) wifi:new:<1,1>, old:<1,1>, ap:<1,1>, sta:<1,1>, prof:1

W (545164) wifi:Haven't to connect to a suitable AP now!
```
This node reports that the current layer is -1 but it used to be layer 2

Layer 3 nodes still stuck in dead state reports very little in the log. I see this one such node:
```
I (732755) wifi:station: c4:dd:57:b8:19:28 leave, AID = 1, bss_flags is 131171, bss:0x3f84dc60
I (732756) wifi:new:<1,1>, old:<1,1>, ap:<1,1>, sta:<1,1>, prof:1
I (744938) wifi:station: c4:dd:57:b8:18:f4 leave, AID = 2, bss_flags is 131171, bss:0x3f850708
I (744939) wifi:new:<1,0>, old:<1,1>, ap:<1,1>, sta:<1,1>, prof:1

W (905830) wifi:Haven't to connect to a suitable AP now!
```
This node also reports that the current layer is -1 but it used to be layer 3

ClaesIvarsson, 2023-03-06:
Any news on this topic ?

zhangyanjiaoesp, 2023-03-07:
@ClaesIvarsson According to your logs, I still can't find the root cause for your issue. Can you describe  your problem in details? Are your application scenarios exactly same as the first reporter ？Please provide a demo for us to reproduce this problem, then we can debug it on our side.

MisterScience5, 2023-03-21:
Thank you for the proposed fix @zhangyanjiaoesp, I am sorry I have been unresponsive, I have not been able to test this due to a high workload. I should be able to test it during this/next week and will provide feedback ASAP.

Leadrive, 2023-04-29:
@zhangyanjiaoesp 
我用4.4.4，做了初步测试。
1、起了两个root，root连的是wifi，root轮流关停，子节点能顺利切换，有时候也会出现切换不了。
2、如果root一个是wifi, 另外一个是ethernet，轮流关停root，那肯定是无法自动切换的。
这两种情况下，子节点重启，都可以顺利连上root。

另外，就是用esp_mesh_disconnect(), esp_mesh_connect(), esp_mesh_stop()之类，都无法切换到另外的root，只有重启。

请教一下。在根节点丢失了，mesh网络还存在着另外一个或多个root的情况下，layer2的节点，有什么办法不通过重启也能切换过去？
这个需求还是很实用的，希望能考虑一下。谢谢。

zhangyanjiaoesp, 2023-09-12:
@Leadrive @ClaesIvarsson @MisterScience5 
For release v4.4, https://github.com/espressif/esp-idf/commit/8756d9545f2208ce4a56718eceea6f4d91b27502 this MR has fixed the layer2 connect issue when fixed root disappear, please update your branch and have a try.

ClaesIvarsson, 2023-09-12:
> @Leadrive @ClaesIvarsson @MisterScience5 For release v4.4, [8756d95](https://github.com/espressif/esp-idf/commit/8756d9545f2208ce4a56718eceea6f4d91b27502) this MR has fixed the layer2 connect issue when fixed root disappear, please update your branch and have a try.
Thanks for the feedback. Will try it!

Leadrive, 2023-09-23:
@zhangyanjiaoesp
我想测试，但不知道如何把这个分支的内容拉下来。
git clone -b 8756d9545f2208ce4a56718eceea6f4d91b27502 https://github.com/espressif/esp-idf
正克隆到 'esp-idf'...
fatal: 远程分支 8756d9545f2208ce4a56718eceea6f4d91b27502 在上游 origin 未发现

如果你们测试没问题，麻烦更新到主分支吧，谢谢！

zhangyanjiaoesp, 2023-09-25:
@Leadrive The fix has already been merged into master, release 5.1/5.0/4.4/4.3

MaxwellAlan, 2023-11-17:
Hi @Leadrive Plz have a try and feel free to reopen if issue still exist.

closed 2023-11-17 by MaxwellAlan