espressif/esp-idf #6521: "PPP module function not supported" error occured during stopping https server and softAP

wwll123, 2021-02-10:
## Environment

- Module or chip used:  esp32-wrover-b
- IDF version: v4.1
- Build System: Make
- Compiler version: esp-2020r2
- Operating System: Linux
- Using an IDE?: No
- Power Supply: USB

## Problem Description
ESP32 would start softAP mode and https server with DHCP after power-up. After user's laptop connecting to AP of ESP32, user would need to use postman to communicate with the https server using RESTful API. ESP32 would stop https server and softAP after receiving stop command from user.

During my testing, the first part (connect to AP and RESTful API communication) is working fine. But sometimes the "PPP module function not supported" error would be occured during stopping https server and softAP. I would like to ask if there is error in my init and deinit code.

### Expected Behavior
Https server and softAP mode would be stopped without error.

### Actual Behavior
Sometimes the "PPP module function not supported" error would be occured during stopping https server and softAP.

### Code to reproduce this issue

Here is the init flow for SoftAp + Https server with DHCP. 
```cpp
ESP_ERROR_CHECK(esp_netif_init());
ESP_ERROR_CHECK(esp_event_loop_create_default());

// Init ap mode and dhcp
wifi_init_config_t cfg = WIFI_INIT_CONFIG_DEFAULT();
ESP_ERROR_CHECK(esp_wifi_init(&cfg));
ESP_ERROR_CHECK(esp_event_handler_register(WIFI_EVENT, ESP_EVENT_ANY_ID, &EvtHandler, NULL));
ESP_ERROR_CHECK(esp_event_handler_register(IP_EVENT, IP_EVENT_STA_GOT_IP, &EvtHandler, NULL));
ESP_ERROR_CHECK(esp_event_handler_register(IP_EVENT, IP_EVENT_AP_STAIPASSIGNED, &EvtHandler, NULL));

psApHandler = esp_netif_create_default_wifi_ap();
esp_netif_dhcps_stop(psApHandler);
esp_netif_ip_info_t IP;
    IP4_ADDR(
        &IP.ip,
		DEFAULT_WIFI_AP_IP_1,
		DEFAULT_WIFI_AP_IP_2,
		DEFAULT_WIFI_AP_IP_3,
		DEFAULT_WIFI_AP_IP_4);
    IP4_ADDR(
        &IP.gw,
		DEFAULT_WIFI_AP_GW_1,
		DEFAULT_WIFI_AP_GW_2,
		DEFAULT_WIFI_AP_GW_3,
		DEFAULT_WIFI_AP_GW_4);
    IP4_ADDR(
        &IP.netmask,
		DEFAULT_WIFI_AP_NM_1,
		DEFAULT_WIFI_AP_NM_2,
		DEFAULT_WIFI_AP_NM_3,
		DEFAULT_WIFI_AP_NM_4);

esp_netif_set_ip_info(psApHandler, &IP);
esp_netif_dhcps_start(psApHandler);
esp_wifi_set_mode(WIFI_MODE_AP);
esp_wifi_set_config(ESP_IF_WIFI_AP, &WifiApConfig);
esp_wifi_start();

// Init https server
httpd_ssl_config_t Config = HTTPD_SSL_CONFIG_DEFAULT();
Config.transport_mode = HTTPD_SSL_TRANSPORT_SECURE;
Config.httpd.max_uri_handlers = HTTPS_SERVER_MAX_URI_HANDLERS;
Config.cacert_pem = https_server_cert_pem_start;
Config.cacert_len = https_server_cert_pem_end - https_server_cert_pem_start;
Config.prvtkey_pem = https_server_prvt_key_pem_start;
Config.prvtkey_len = https_server_prvt_key_pem_end - https_server_prvt_key_pem_start;
httpd_ssl_start(&HttpsServerHandler, &Config);
httpd_register_uri_handler(HttpsServerHandler, &GetSysUri);
```

Here is the code for stopping https server and ap mode.
```cpp
httpd_ssl_stop(HttpsServerHandler);
esp_netif_dhcps_stop(psApHandler);
esp_netif_destroy(psApHandler);
esp_wifi_stop();
esp_wifi_deinit();
ESP_ERROR_CHECK(esp_event_handler_unregister(WIFI_EVENT, ESP_EVENT_ANY_ID, &EvtHandler));
ESP_ERROR_CHECK(esp_event_handler_unregister(IP_EVENT, IP_EVENT_STA_GOT_IP, &EvtHandler));
ESP_ERROR_CHECK(esp_event_handler_unregister(IP_EVENT, IP_EVENT_AP_STAIPASSIGNED, &EvtHandler));
```

Sometimes above code would trigger following error code. 
```cpp
E (457925) esp-netif_lwip-ppp: esp_netif_stop_ppp not supported, please enable PPP in lwIP component configuration
I (457926) Wifi: Wifi event: WIFI_EVENT_AP_STADISCONNECTED
I (457926) Wifi: Wifi event: WIFI_EVENT_AP_STOP
E (457927) esp-netif_lwip-ppp: esp_netif_stop_ppp not supported, please enable PPP in lwIP component configuration
```

david-cermak, 2021-02-12:
Hi @wwll123 

Please move the `esp_netif_destroy()` down, below the `esp_wifi_deinit()`. You can refer to the correct deinit of wifi in IDF examples:
https://github.com/espressif/esp-idf/blob/master/examples/common_components/protocol_examples_common/connect.c#L281-L294

What's happening in your scenario, is that the network interface handle gets deleted and, after that, the `WIFI_EVENT_AP_STOP` event is posted (upon `esp_wifi_stop()`) with an invalid (already destroyed) netif.

wwll123, 2021-02-19:
Hi David,
Thanks. The issue is fixed.

closed 2021-02-19 by wwll123