espressif/esp-idf #6522: Support for WiFi direct (P2P)

DavidJRobertson, 2021-02-10:
There have been mentions a few years ago that wifi direct was on the to-do list, but it is still not implemented. 

This is surprising because there are very clear reasons why developers would want to use wifi direct in an IOT project. Specifically, it would greatly improve the user experience of using wifi to connect to the ESP32 device from an Android phone without the need for network infrastructure or provisioning.

Right now the alternative is to either use bluetooth (which is low-bandwidth and has licensing/certification cost issues) or to use soft AP on the ESP32, which provides poor user experience due to the requirement for the smartphone wifi connection to be manually switched, interrupting the phone's normal network connection. At least on some Android devices, it is supported to connect to a normal AP and a wifi direct group at the same time.

I note that the upstream wpa_supplicant project includes support for wifi direct/p2p so I presume this could be made to work on the ESP32.

Yugnat4Github, 2021-02-12:
Permit me to agree with what is said in this feature request.
If application is mobile and range not short enough for bluetooth, the only way to share data with a device is creating soft AP. That means  users will loose internet connectivity on their devices which is not acceptable... Unless there are other ways I don't know about.
Please consider implementing this.

sagb2015, 2021-02-28:
@DavidJRobertson - Yes, this is on our TODO list, but it may not be available any time soon since our focus is currently on the protocols that are more important from security perspective (WPA3, PMF, DPP) or that cover more IoT/wireless scenarios (FTM, 11kv, 11r etc). However, depending on the demand we may consider to increase the priority of implementing this feature as well.

kaedros, 2021-05-25:
We have also been waiting for this feature for many years and it would bring a huge improvement in the experience already described very well by @DavidJRobertson.

We cannot imagine that this demand is not important for everyone who develops for IoT.

wagner-rebello, 2021-06-17:
This feature can help a lot for IoT usage.
The network configuration using a smartphone app would be very simple as the network do not need to change.
Changing networks on Android or iOS is painful. There are many factors that implies in the network refuses to change.
Using wifi direct "kills" all that factors.

mauriciofarina, 2021-06-17:
@DavidJRobertson and @sagb2015 
We are also on standby for this feature for a log time!

itamardv, 2021-06-21:
It would be great

jensolsson, 2021-06-28:
Need this too. Now we are looking for another chip for our project but the ESP32 would be perfect if it only supported WiFi Direct / P2P

TeschRenan, 2021-07-12:
I agree that it's not extremely important, but I support inclusion.

FlynnMa, 2021-07-23:
> Need this too. Now we are looking for another chip for our project but the ESP32 would be perfect if it only supported WiFi Direct / P2P

What is the another chip are you talking about?
...I would be happy to use an alternative if if provides both BT and wifi with wifi-direct.

wuujiann, 2021-07-30:
I too am looking for this feature.  Hopefully, this feature can be impelmented soon.

Torarqui, 2021-11-05:
This feature would be decisive in our IoT products! As @wagner-rebello pointed out, developing an App that can connect via Wifi to an device without Wifi-Direct is really hard, and even if you manage to do that, the end result is not user friendly as @DavidJRobertson said. Looking forward to see this implemented in ESP-IDF!

writetosumeet, 2021-12-05:
Any timeline for Wifi Direct for ESP32?
We are considering using ESP32 for a new product but inability to provide WIFI direct is a big hinderance!

maleiter, 2021-12-16:
We'd need WiFi Direct too. The lack of WiFi Direct support forces us to use another chip with the next version of our product.

jack0c, 2022-01-06:
Does anyone try to use DPP, which is more official for IOT device provisioning. There are already some Android phones support this feature, and I think new Android phones is likely to support this feature, just like Android support WPS and P2P.

ZantsuRocks, 2022-01-07:
> Does anyone try to use DPP, which is more official for IOT device provisioning. There are already some Android phones support this feature, and I think new Android phones is likely to support this feature, just like Android support WPS and P2P.

As iOS don't support this technology and the same for old android phones (the majority in Brazil where our product is sold) i can't even try to lose time testing this.

DavidJRobertson, 2022-01-07:
> As iOS don't support this technology and the same for old android phones (the majority in Brazil where our product is sold) i can't even try to lose time testing this.

Yeah, nice thing about WiFi direct is that it is backwards compatible with WiFi devices that don't support it 'directly' (just appears as access point, so much the same as existing soft-AP provisioning approaches).

adamoskaranikas, 2022-03-03:
Hi, are there any updates regarding WiFi Direct support for ESP32 ? Is there a way to maybe adapt ESP-NOW to allow for communication with Wifi-Direct enabled devices other than ESP32's (i.e. ESP32 to Android mobile device). This would be a game changer for us, otherwise we will have to look at other MCU options supporting this feature

benjamin-shapiro, 2022-03-12:
+1 on this -- current soft-AP implementation with Android is highly undesirable. The ability to directly send a video stream to a mobile device on the ESP32 platform would be transformational for our product.

ArnyminerZ, 2022-04-06:
Hope this gets added ASAP. I agree that it's not more important than security, but platform-wise I think it would help users and developers quite a lot.

AxelLin, 2022-04-19:
> Does anyone try to use DPP, which is more official for IOT device provisioning. There are already some Android phones support this feature, and I think new Android phones is likely to support this feature, just like Android support WPS and P2P.

@jack0c 
Just want to clarify that to use DPP the Access Point needs to support WPA3, right?
That is a problem because may users are still using access point without WPA3 support.

AxelLin, 2022-04-19:
> @DavidJRobertson - Yes, this is on our TODO list, but it may not be available any time soon since our focus is currently on the protocols that are more important from security perspective (WPA3, [...]

@sagb2015 
It has been a year since your reply. Is there any update for WiFi direct? (May users are asking this feature).
Can we expect v5.0 to include wifi direct support?

romatetemadze, 2022-05-03:
Any news on the WIFI-Direct?

sagb2015, 2022-05-06:
Hi All, 
No we do not have support for WiFi-Direct yet, neither we are considering it for IDF 5.0. However, given the demand we would like to evaluate which chips you would like this features to be on. Given the memory size implication we may target limited subset of the features. WiFi-Direct is a very big feature to support.

AxelLin, 2022-05-06:
@sagb2015 
Do you have a rough estimated number of the memory that will be taken by WiFi-Direct?

jensolsson, 2022-05-06:
@sagb2015 any of the esp chips would work for us. 

Kind regards
Jens

jensolsson, 2022-05-06:
@sagb2015 I started playing with WiFi Direct a while ago, the goal was to make the ESP32 appear as a wifi direct device on an Android phone so I hand crafted a vendor IE with P2P info and used esp_wifi_set_vendor_ie to enable it. I finally got it to work so my esp32 device would show up under WiFi Direct in my Android phone. But it was of course not possible to connect to it yet.

I think the next steps would be to enable WPS and do some kind of p2p handshake in order to determine which of the devices will be master. But as it is possible to set a hint on Android to force it to be either master or not, i do not really know if this handshake is necessary or if it would just work if WPS would be enabled?

Is it really that complicated to get a basic WiFi Direct up and running?

Kind regards
Jens

nicklasb, 2023-04-29:
This would be very valuable. 
As it stands now, the only implemented and broadly used peer to-peer connectivity is BLE with all its complexities. 
WiFi Direct would enable not only peer-to-peer communication with normal PC:s, and possibly with some other MCU:s and devices. 

I really like ESP-NOW, but it is currently only possible to run on espressif hardware and it seems as if it requires driver-level changes on PC:s which just won't happen on a broader scale. Especially when WiFi Direct exists.

I think that it would be in Espressifs interests to open up this avenue of communication, there are so many use cases where you don't want to have to depend on some wonky WiFi router, hassle with passwords and expose yourself to the very well-developed WiFi hacking world to do some communication. 

(And as BLE is implemented, the cat is sort of out of the vendor-lock-in bag anyway. )

andresbrocco, 2023-04-29:
+1 Up! Great Feature request! Would be really useful

johnnyb, 2023-06-19:
Would love to have Wifi Direct on an ESP32.  Would have been a game-changer for my current project.  Still might be for the future of the project.  Right now we are having to do a full Linux Single-board computer to get a worthwhile Wifi Direct.

nicklasb, 2023-06-19:
> Would love to have Wifi Direct on an ESP32. Would have been a game-changer for my current project. Still might be for the future of the project. Right now we are having to do a full Linux Single-board computer to get a worthwhile Wifi Direct.

Exactly! There are a huge number of possible solutions that could benefit from this being implemented.

jasta, 2023-07-24:
I think the real issue here is that wpa_supplicant would need to be updated, then new APIs added to use the features it exposes.  There appear to be a great many reasons why this would be desirable, such as possibly fixing #10615.  Unfortunately there are a lot of patches on top of esp-idf's fork of wpa_supplicant so unmerging those forks and rebasing on top of the latest will probably not be realistic, at least not without potential breakage.  I wonder if perhaps espressif is open to a strategy in which we experimentally bring in another version of wpa_supplicant and try to just get things building and let the community tinker with the unstable support to collect bug reports.

nicklasb, 2023-07-24:
> Unfortunately there are a lot of patches on top of esp-idf's fork of wpa_supplicant so unmerging those forks and rebasing on top of the latest will probably not be realistic, at least not without potential breakage. 

If there's already a lot of patches and those patches aren't moving upstream, then i would rather try and be inspired by, rather than try to merge, the upstream code. There is always that point where you have to make that decision.
 
So If I were Espressif, I would rather just try to port it into their fork. The business case would be quite uniquely interoperable hardware. It would probably pays to use those extra resources for maintaining their own supplicant.
Or maybe I am just trying to remind Espressif of the hidden costs of non-reuse. :-)

jasta, 2023-07-24:
> If there's already a lot of patches and those patches aren't moving upstream, then i would rather try and be inspired by, rather than try to merge, the upstream code. There is always that point where you have to make that decision.

Certainly an option, but still a good amount of work.  Given the age of this p2p support request, I'm guessing espressif isn't going to suddenly prioritize deepening their fork just for p2p support.

> So If I were Espressif, I would rather just try to port it into their fork. The business case would be quite uniquely interoperable hardware. It would probably pays to use those extra resources for maintaining their own supplicant.
Or maybe I am just schrewdly trying to remind Espressif of the hidden costs of non-reuse. :-)

I mean, they're spending a lot of time fixing issues I'd bet money upstream has already fixed, so I think they're probably already aware of the costs: https://github.com/espressif/esp-idf/commits/master/components/wpa_supplicant/src

It's a hard spot for them to be in though, I empathize.  I do think though that forking wpa_supplicant and leaving the ability to merge with upstream was probably the wrong call given how much Wi-Fi is still actively developing by the Wi-Fi Alliance.  They're getting very far behind very fast...

Thothore, 2023-10-12:
Hello, I would love to have this feature for my project (android auto wireless adapter), the chip I have is a wemos lolin32 v1

blaueente, 2023-10-20:
Might Wifi Aware / NAN be a suitable replacement?
https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/network/esp_nan.html

wagner-rebello, 2023-10-20:
> Might Wifi Aware / NAN be a suitable replacement? https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/network/esp_nan.html

No it's not, many OS do not have this functionality (AKA iOS)

blaueente, 2023-10-20:
> > Might Wifi Aware / NAN be a suitable replacement? https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/network/esp_nan.html
> 
> No it's not, many OS do not have this functionality (AKA iOS)

True, but iOS also never actually supported WiFi direct.

nicklasb, 2023-10-20:
> > > Might Wifi Aware / NAN be a suitable replacement? https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/network/esp_nan.html
> > 
> > 
> > No it's not, many OS do not have this functionality (AKA iOS)
> 
> True, but iOS also never actually supported WiFi direct.

Hm. TBH, this weakens the case for Wi-Fi Direct significantly. 
The usual Apple-for-purely-Apple-reasons wielding their proprietary lock-in sword as much as they possibly can. 
:-(

johnnyb, 2023-11-15:
Wifi direct is native compatible with Android.  Thus, making devices that would auto-connect to Android and use IP-based protocols would be greatly simplified with Wifi Direct support.

nicklasb, 2023-11-16:
> Wifi direct is native compatible with Android. Thus, making devices that would auto-connect to Android and use IP-based protocols would be greatly simplified with Wifi Direct support.

Agreed. And I just realized that I have an Android phone now. :-)
Seriously though, it would be actually be quite helpful. 
Bluetooth is a bit cumbersome and overengineered, so a more straightforward protocol would be nice.

johnnyb, 2023-11-27:
I think if an ecosystem for Wifi Direct were built up inside enough secondary devices, it would force Apple to either go along with the crowd to participate in the ecosystem, or publish their spec openly.

AxelLin, 2023-12-17:
> Hi All, No we do not have support for WiFi-Direct yet, neither we are considering it for IDF 5.0. However, given the demand we would like to evaluate which chips you would like this features to be [...]

@sagb2015 
It has been a long time, do you have any progress to share?

nachiketkukade, 2024-01-05:
@AxelLin , we're prioritising some other features and improvements over P2P at this moment (
e.g. - WiFi Aware). So there is no major update on this from our side.

artem-pdus, 2024-04-10:
Support of wifi-direct would be really great step towards improving user experiences with diy/embedded devices, looking forward to it.

AxelLin, 2024-11-05:
@nachiketkukade 

This was marked as "Status:Reviewing" since Feb 27, but there is no more update.
How is the status now? (It does not make sense to review for such long time).

nachiketkukade, 2024-11-28:
> we're prioritising some other features and improvements over P2P at this moment (
e.g. - WiFi Aware). So there is no major update on this from our side.

@AxelLin , there is no change in our roadmap since this update was given. Bot had incorrectly moved this to `Reviewing`.

finger563, 2025-05-28:
👍 for this feature

AHWang2, 2025-12-28:
I recently finish the first step of P2P. I let my android phone find out my ESP32S3 in its P2P list, only by adding some information element to ESP's probe response frame. You can have a look at my project.

nicklasb, 2025-12-28:
Cool, I won't have time for now (until after mew years), but can't help but
thinking that implementing the component would be a good Copilot job if
done stepwise, the standard seems well-defined.

AHWang2, 2025-12-30:
Hey guys, if WiFi P2P is implemented on ESP, in your application scenario, do you want ESP devices to become GO or Android phones to become GO? And why?

nicklasb, 2025-12-31:
> > > Hey guys, if WiFi P2P is implemented on ESP, in your application scenario, do you want ESP devices to become GO or Android phones to become GO? And why?
> > 
> > 
> > Both
> 
> What do you mean "both"? Could you please share your scenario?

If this was implemented, I would usually have the ESP32 as a group owner to be able for a phone to use its services. But eventually I would like mix them to have ESPs using the same services.

So if you are asking which to begin with implementing, it would be GO first, then Client.  At least after my personal priorities. 
But others may of course have other use cases.

(Main key for me is not having to have a router and/or be separate of the wifi)

AHWang2, 2026-01-01:
> ESP devices can as both of GO and client

Thanks for your replying. May I ask your scenario for ESP development?

AHWang2, 2026-01-01:
> If this was implemented, I would usually have the ESP32 as a group owner to be able for a phone to use its services. But eventually I would like mix them to have ESPs using the same services.
>
> So if [...]

Thanks for your replying. May I ask your scenario for ESP development?

jesses2025smith, 2026-01-04:
> > ESP devices can as both of GO and client
> 
> Thanks for your replying. May I ask your scenario for ESP development?

Maybe smart home

AHWang2, 2026-01-04:
> > > ESP devices can as both of GO and client
> > 
> > 
> > Thanks for your replying. May I ask your scenario for ESP development?
> 
> Maybe smart home

I have a doubt here. I think IoT needs low power while WiFi-Direct does cost much power. I was considering if implementing WiFi-Direct based on ESP-IDF was a really valuable requirement. Will there be many ESP developers choosing to use this function? And what's their scenario?