espressif/esp-idf #6524: ASSERT_ERR(0), in coex_schm.c at line 189

formus14, 2021-02-11:
- Module or chip used: ESP32-WROVER-B
- Board type: Custom PCB
- IDF version: v4.1.1
- ToolChain :  esp-2020r3
- Build System: CMake
- Operating System: Windows
- (Windows only) environment type: VisualGDB

- Problem Description:
GATT server example code crashes continuously due to assertation failure in the WiFi/BT Co-existence.

- Steps to reproduce
Just flash the GATT server demo example with the [SDK ](https://files.fm/u/nerh3ubw7)config attached.

- Here's the log :

```
I (31) boot: ESP-IDF v4.1.1-dirty 2nd stage bootloader
I (31) boot: compile time 20:29:40
I (31) boot: chip revision: 1
I (34) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (41) boot.esp32: SPI Speed      : 40MHz
I (46) boot.esp32: SPI Mode       : DIO
I (50) boot.esp32: SPI Flash Size : 2MB
I (55) boot: Enabling RNG early entropy source...
I (60) boot: Partition Table:
I (64) boot: ## Label            Usage          Type ST Offset   Length
I (71) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (79) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (86) boot:  2 factory          factory app      00 00 00010000 00100000
I (94) boot: End of partition table
I (98) boot_comm: chip revision: 1, min. application chip revision: 0
I (105) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x1c0b4 (114868) map
I (158) esp_image: segment 1: paddr=0x0002c0dc vaddr=0x3ffbdb60 size=0x02bb4 ( 11188) load
I (163) esp_image: segment 2: paddr=0x0002ec98 vaddr=0x40080000 size=0x00404 (  1028) load
I (165) esp_image: segment 3: paddr=0x0002f0a4 vaddr=0x40080404 size=0x00f74 (  3956) load
I (175) esp_image: segment 4: paddr=0x00030020 vaddr=0x400d0020 size=0x762a8 (484008) map
I (367) esp_image: segment 5: paddr=0x000a62d0 vaddr=0x40081378 size=0x1296c ( 76140) load
I (411) boot: Loaded app from partition at offset 0x10000
I (411) boot: Disabling RNG early entropy source...
I (411) cpu_start: Pro cpu up.
I (415) cpu_start: Application information:
I (420) cpu_start: Project name:     gatt_server_demos
I (426) cpu_start: App version:      1
I (430) cpu_start: Compile time:     Feb 10 2021 20:28:51
I (436) cpu_start: ELF file SHA256:  c43af6513091d6fd...
I (442) cpu_start: ESP-IDF:          v4.1.1-dirty
I (448) cpu_start: Starting app cpu, entry point is 0x40081284
I (439) cpu_start: App cpu up.
I (458) heap_init: Initializing. RAM available for dynamic allocation:
I (465) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM
I (471) heap_init: At 3FFB6388 len 00001C78 (7 KiB): DRAM
I (477) heap_init: At 3FFB9A20 len 00004108 (16 KiB): DRAM
I (483) heap_init: At 3FFBDB5C len 00000004 (0 KiB): DRAM
I (489) heap_init: At 3FFC81D8 len 00017E28 (95 KiB): DRAM
I (496) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (502) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (508) heap_init: At 40093CE4 len 0000C31C (48 KiB): IRAM
I (514) cpu_start: Pro cpu start user code
I (533) spi_flash: detected chip: gd
I (534) spi_flash: flash io: dio
W (534) spi_flash: Detected size(8192k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (544) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (579) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (589) BTDM_INIT: BT controller compile version [af40726]
I (589) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (689) phy: phy_version: 4500, 0cd6843, Sep 17 2020, 15:37:07, 0, 0
ASSERT_ERR(0), in coex_schm.c at line 189
Guru Meditation Error: Core  0 panic'ed (Interrupt wdt timeout on CPU0)
Core 0 register dump:
PC      : 0x40082e0c  PS      : 0x00060f34  A0      : 0x80112c91  A1      : 0x3ffb5f20
A2      : 0x00000001  A3      : 0x00000000  A4      : 0x60008048  A5      : 0x3ffbdbe4
A6      : 0x3ffbdbe4  A7      : 0x00000000  A8      : 0x80082e0c  A9      : 0x3ffb5f00
A10     : 0x0000002a  A11     : 0x0000002a  A12     : 0x00000010  A13     : 0xffffffff
A14     : 0x00000000  A15     : 0xfffffffc  SAR     : 0x00000004  EXCCAUSE: 0x00000005
EXCVADDR: 0x00000000  LBEG    : 0x40082db4  LEND    : 0x40082dbb  LCOUNT  : 0x00000000

ELF file SHA256: c43af6513091d6fd

Backtrace: 0x40082e09:0x3ffb5f20 0x40112c8e:0x3ffb5f40 0x40112ce6:0x3ffb5f80 0x401120d4:0x3ffb5fb0 0x4008df3d:0x3ffb5fe0

Core 1 register dump:
PC      : 0x40145bc6  PS      : 0x00060634  A0      : 0x800d3a1a  A1      : 0x3ffbc3f0
A2      : 0x00000000  A3      : 0x00000000  A4      : 0x00000001  A5      : 0x80000001
A6      : 0x00000003  A7      : 0x00060023  A8      : 0x800d34ee  A9      : 0x3ffbc3c0
A10     : 0x00000000  A11     : 0x00060623  A12     : 0x00060620  A13     : 0x00000001
A14     : 0x00060620  A15     : 0x00000000  SAR     : 0x00000000  EXCCAUSE: 0x00000005
EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000

ELF file SHA256: c43af6513091d6fd

Backtrace: 0x40145bc3:0x3ffbc3f0 0x400d3a17:0x3ffbc410 0x4008fa55:0x3ffbc430 0x4008df3d:0x3ffbc450

Rebooting...
ets Jun  8 2016 00:22:57
```
Obviously, there is an issue with the BT/WiFi coexistence, but as the demo example is for BT functionality only coexistence wouldn't make sense to be enabled.

Any help is appreciated..

projectgus, 2021-02-12:
Hi @formus14,

Thanks for reporting this. I was able to reproduce the same using the default config, by disabling the option `CONFIG_ESP32_WIFI_SW_COEXIST_ENABLE`. As a workaround, re-enabling this option should prevent the assert.

Most of the Bluetooth engineers are currently on holiday for Chinese New Year, so it may be a week or two before someone gets back to you with more details. Thanks for your patience.

formus14, 2021-02-25:
@projectgus any updates on this ?!

blueMoods, 2021-02-26:
Please test with this patch, (can't upload file, just paste it here):
And we will fix it on release version as soon as possible.
```
diff --git a/components/bt/controller/bt.c b/components/bt/controller/bt.c
index c506290..5602d10 100644
--- a/components/bt/controller/bt.c
+++ b/components/bt/controller/bt.c
@@ -1113,7 +1113,7 @@ static uint8_t coex_schm_curr_period_get_wrapper(void)
 #if CONFIG_SW_COEXIST_ENABLE
     return coex_schm_curr_period_get();
 #else
-    return 0;
+    return 1;
 #endif
 }
 
@@ -1131,7 +1131,7 @@ static int coex_wifi_channel_get_wrapper(uint8_t *primary, uint8_t *secondary)
 #if CONFIG_SW_COEXIST_ENABLE
     return coex_wifi_channel_get(primary, secondary);
 #else
-    return 0;
+    return -1;
 #endif
 }
 
@@ -1140,7 +1140,7 @@ static int coex_register_wifi_channel_change_callback_wrapper(void *cb)
 #if CONFIG_SW_COEXIST_ENABLE
     return coex_register_wifi_channel_change_callback(cb);
 #else
-    return 0;
+    return -1;
 #endif
 }
 
@@ -1328,7 +1328,7 @@ esp_err_t esp_bt_controller_init(esp_bt_controller_config_t *cfg)
     if (btdm_dram_available_region[0].mode == ESP_BT_MODE_IDLE) {
         return ESP_ERR_INVALID_STATE;
     }
-    
+
     osi_funcs_p = (struct osi_funcs_t *)malloc_internal_wrapper(sizeof(struct osi_funcs_t));
     if (osi_funcs_p == NULL) {
         return ESP_ERR_NO_MEM;
```

sebirdman, 2021-03-03:
@blueMoods this appears to have resolved the issue for me

formus14, 2021-03-26:
@blueMoods the patch fixes the bug, would the proposed fix be added to IDF 4.3?

closed 2021-03-26 by formus14

demik, 2021-04-18:
Just hit the same issue. Looks like the fix is not on IDF 4.2. Is it pushed somewhere yet ?