espressif/esp-idf #6525: esp32 how to detect terminal(putty etc.) connected?

drony, 2021-02-11:
Logging decreases processing power. If the usb is not connected I want to turn off logging.
How to detect usb attached and terminal(putty etc.) connected?

projectgus, 2021-02-12:
Hi @drony,

I'm afraid there's no way to detect the connection by itself, at least not a normal dev board. If you're designing your own hardware then you could probably connect a signal from the USB/serial chip directly to the ESP32 for this purpose.

One thing you could do is listen for serial input and only start logging in this case. Then when you open your terminal you type a character to enable logging, and you could use a different character to end logging.

Increasing the baud rate from the default is also a good way to decrease the overhead of logging in a task, although there is always some overhead as you note.

drony, 2021-02-15:
thanks for advice

igrr, 2021-02-15:
If you know that the serial terminal is expected to support VT-100 escape sequences, you can use `linenoiseProbe` function. This function will return 0 if the connected terminal does support VT-100, and non-zero if the terminal doesn't support VT-100 or is absent.
This is used in the "system/console" example: https://github.com/espressif/esp-idf/blob/73db142403c6e5b763a0e1c07312200e9b622673/examples/system/console/main/console_example_main.c#L164-L178.

drony, 2021-02-15:
i will try. thanks

drony, 2021-02-18:
thanks it's worked

Alvin1Zhang, 2021-02-18:
@drony Thanks for sharing the updates, could the issue be closed now? Thanks.

closed 2021-02-18 by drony