espressif/esp-idf #6526: SSL Error - error 104, error no - Connection Reset  By Peer

Vbansal1996, 2021-02-11:
Dev : ESP32 Wroom Devkit
ENV: Linux
IDF version: 4.2  - release on Dec 2020

When I am trying to make the device connection over MQTT for Azure IOT HUB.
Device is able to get connected to the HUB.
Able to subscribe the few messages but after that it throws this particular error and get disconnected.
I had gone through google but didn't able to find any solution on this till now.

Adding error logs for the reference:

```
D (18189) MQTT_CLIENT: mqtt_enqueue id: 55875, type=3 successful
D (18189) OUTBOX: ENQUEUE msgid=55875, msg_type=3, len=180, size=359
D (18349) MQTT_CLIENT: mqtt_message_receive: first byte: 0x90
D (18359) MQTT_CLIENT: mqtt_message_receive: read "remaining length" byte: 0x3
D (18359) MQTT_CLIENT: mqtt_message_receive: total message length: 5 (already read: 2)
D (18369) MQTT_CLIENT: mqtt_message_receive: read_len=3
D (18369) MQTT_CLIENT: mqtt_message_receive: transport_read():5 5
D (18379) MQTT_CLIENT: msg_type=9, msg_id=1320
D (18379) MQTT_CLIENT: pending_id=55875, pending_msg_count = 5
D (18389) OUTBOX: DELETED msgid=1320, msg_type=8, remain size=280
D (18389) MQTT_CLIENT: Subscribe successful
I (18399) [cloud_mode]: SUBSCRIBED, msg_id=1320
D (18399) event: running post HIFE_MQTT_EVENTS:18 with handler 0x400dfa3c and context 0x3ffcb020 on loop 0x3ffb58c8
0x400dfa3c: NetworkEventHandler at /home/vaibhav/esp/HIFEThings/build/../components/hife_hscp/src/hife_hscp_manager.c:81

D (18409) event: running post HIFE_MQTT_EVENTS:18 with handler 0x400df1a4 and context 0x3ffd156c on loop 0x3ffb58c8
0x400df1a4: CloudEventHandler at /home/vaibhav/esp/HIFEThings/build/../components/message_interface/src/messages_manager.c:110

D (18419) [messages_manager]: Unhandled event 18
E (18659) TRANS_SSL: ssl_poll_read select error 104, errno = Connection reset by peer, fd = 55
E (18659) MQTT_CLIENT: mqtt_message_receive: transport_read() error: errno=0
E (18669) [cloud_mode]: ERROR Code: 0
E (18669) MQTT_CLIENT: mqtt_process_receive: mqtt_message_receive() returned -1
D (18679) MQTT_CLIENT: Reconnect after 10000 ms
E (18679) [cloud_mode]: DISCONNECTED
D (18689) event: running post HIFE_MQTT_EVENTS:22 with handler 0x400dfa3c and context 0x3ffcb020 on loop 0x3ffb58c8
0x400dfa3c: NetworkEventHandler at /home/vaibhav/esp/HIFEThings/build/../components/hife_hscp/src/hife_hscp_manager.c:81

I (18699) ReportModuleWiFiMode: WiFi Working State: 3
D (18729) event: running post HIFE_MQTT_EVENTS:22 with handler 0x400df1a4 and context 0x3ffd156c on loop 0x3ffb58c8
0x400df1a4: CloudEventHandler at /home/vaibhav/esp/HIFEThings/build/../components/message_interface/src/messages_manager.c:110

D (18729) [messages_manager]: Unhandled event 22
D (28689) MQTT_CLIENT: Reconnecting...
I (28689) [cloud_mode]: MQTT_EVENT_BEFORE_CONNECT
D (28689) event: running post HIFE_MQTT_EVENTS:16 with handler 0x400dfa3c and context 0x3ffcb020 on loop 0x3ffb58c8
0x400dfa3c: NetworkEventHandler at /home/vaibhav/esp/HIFEThings/build/../components/hife_hscp/src/hife_hscp_manager.c:81

D (28689) esp-tls: host:AZR-QA-IoTHub.azure-devices.net: strlen 31
D (28689) event: running post HIFE_MQTT_EVENTS:16 with handler 0x400df1a4 and context 0x3ffd156c on loop 0x3ffb58c8
0x400df1a4: CloudEventHandler at /home/vaibhav/esp/HIFEThings/build/../components/message_interface/src/messages_manager.c:110

D (28709) [messages_manager]: Unhandled event 16
D (28799) esp-tls: handshake in progress...
D (30649) MQTT_CLIENT: Transport connected to mqtts://AZR-QA-IoTHub.azure-devices.net:8883
D (30649) MQTT_CLIENT: Sending MQTT CONNECT message, type: 1, id: 0000
D (30729) MQTT_CLIENT: mqtt_message_receive: first byte: 0x20
D (30739) MQTT_CLIENT: mqtt_message_receive: read "remaining length" byte: 0x2
D (30739) MQTT_CLIENT: mqtt_message_receive: total message length: 4 (already read: 2)
D (30749) MQTT_CLIENT: mqtt_message_receive: read_len=2
D (30749) MQTT_CLIENT: mqtt_message_receive: transport_read():4 4
D (30759) MQTT_CLIENT: Connected
I (30759) [cloud_mode]: CONNECTED
D (30759) event: running post HIFE_MQTT_EVENTS:17 with handler 0x400dfa3c and context 0x3ffcb020 on loop 0x3ffb58c8
0x400dfa3c: NetworkEventHandler at /home/vaibhav/esp/HIFEThings/build/../components/hife_hscp/src/hife_hscp_manager.c:81

D (30759) MQTT_CLIENT: mqtt_enqueue id: 29572, type=8 successful
I (30779) ReportModuleWiFiMode: WiFi Working State: 3
D (30779) OUTBOX: ENQUEUE msgid=29572, msg_type=8, len=79, size=359
D (30799) MQTT_CLIENT: Sent subscribe topic=devices/63DA0043EC70649000910C63000013C71BF60EC10/messages/devicebound/#, id: 29572, type=8 successful
D (30809) MQTT_CLIENT: mqtt_enqueue id: 30325, type=8 successful
D (30809) OUTBOX: ENQUEUE msgid=30325, msg_type=8, len=29, size=388
D (30819) event: running post HIFE_MQTT_EVENTS:17 with handler 0x400df1a4 and context 0x3ffd156c on loop 0x3ffb58c8
0x400df1a4: CloudEventHandler at /home/vaibhav/esp/HIFEThings/build/../components/message_interface/src/messages_manager.c:110

D (30819) MQTT_CLIENT: Sent subscribe topic=$iothub/methods/POST/#, id: 30325, type=8 successful
I (30839) [messages_manager]: Listening to cloud
D (30839) MQTT_CLIENT: mqtt_enqueue id: 12857, type=8 successful
D (30849) OUTBOX: ENQUEUE msgid=12857, msg_type=8, len=25, size=413
D (30859) MQTT_CLIENT: Sent subscribe topic=$iothub/twin/res/#, id: 12857, type=8 successful
D (30859) event: running post APPLICATION_MESSAGES:2 with handler 0x400e01dc and context 0x3ffd16c4 on loop 0x3ffb58c8
0x400e01dc: _event_handler at /home/vaibhav/esp/HIFEThings/build/../components/application/smart_ac/src/smart_ac.c:46

D (30859) MQTT_CLIENT: mqtt_enqueue id: 52111, type=8 successful
I (30879) [smart_ac]: NEW_DEVICE_CONNECTED
D (30889) OUTBOX: ENQUEUE msgid=52111, msg_type=8, len=46, size=459
I (30889) [smart_ac]: [APP] Free memory: 135872 bytes
D (30889) MQTT_CLIENT: Sent subscribe topic=$iothub/twin/PATCH/properties/desired/#, id: 52111, type=8 successful
D (30909) event: running post HSCP_EVENT:8 with handler 0x400df9b8 and context 0x3ffbdb50 on loop 0x3ffb58c8
0x400df9b8: HSCPIncomingEventsHandler at /home/vaibhav/esp/HIFEThings/build/../components/hife_hscp/src/hife_hscp_manager.c:57

E (30989) [messages_manager]: Empty context...
I (30999) [cloud_mode]: {"DEVID":"63DA0043EC70649000910C63000013C71BF60EC10","PRODUCTCODE":"AirConditionerV1","H2101":0,"H2100":[]}
D (30999) MQTT_CLIENT: mqtt_enqueue id: 25981, type=3 successful
D (31009) OUTBOX: ENQUEUE msgid=25981, msg_type=3, len=180, size=639
D (31159) MQTT_CLIENT: mqtt_message_receive: first byte: 0x90
D (31159) MQTT_CLIENT: mqtt_message_receive: read "remaining length" byte: 0x3
D (31159) MQTT_CLIENT: mqtt_message_receive: total message length: 5 (already read: 2)
D (31169) MQTT_CLIENT: mqtt_message_receive: read_len=3
D (31169) MQTT_CLIENT: mqtt_message_receive: transport_read():5 5
D (31179) MQTT_CLIENT: msg_type=9, msg_id=29572
D (31189) MQTT_CLIENT: pending_id=25981, pending_msg_count = 9
D (31189) OUTBOX: DELETED msgid=29572, msg_type=8, remain size=560
D (31199) MQTT_CLIENT: Subscribe successful
I (31199) [cloud_mode]: SUBSCRIBED, msg_id=29572
D (31209) event: running post HIFE_MQTT_EVENTS:18 with handler 0x400dfa3c and context 0x3ffcb020 on loop 0x3ffb58c8
0x400dfa3c: NetworkEventHandler at /home/vaibhav/esp/HIFEThings/build/../components/hife_hscp/src/hife_hscp_manager.c:81

D (31219) event: running post HIFE_MQTT_EVENTS:18 with handler 0x400df1a4 and context 0x3ffd156c on loop 0x3ffb58c8
0x400df1a4: CloudEventHandler at /home/vaibhav/esp/HIFEThings/build/../components/message_interface/src/messages_manager.c:110

D (31229) [messages_manager]: Unhandled event 18
E (31459) TRANS_SSL: ssl_poll_read select error 104, errno = Connection reset by peer, fd = 55
E (31459) MQTT_CLIENT: mqtt_message_receive: transport_read() error: errno=9
E (31469) [cloud_mode]: ERROR Code: 0
E (31469) MQTT_CLIENT: mqtt_process_receive: mqtt_message_receive() returned -1
D (31479) MQTT_CLIENT: Reconnect after 10000 ms
E (31479) [cloud_mode]: DISCONNECTED
```

If someone can help in this that will be grateful.
I tried passing certificates but still getting same issue.

Any help!

david-cermak, 2021-02-12:
Hi @Vbansal1996 

The error means that the remote endpoint actively terminated the connection, i.e. your server closed the connection. Could you please check if the server side logs show anything, which we could interpret as a potential reason for disconnection?
I haven't found anything in the log above which might cause this disconnection.

Alvin1Zhang, 2021-03-30:
Thanks for reporting, will close due to short of feedback, feel free to reopen with more details. Thanks.

closed 2021-03-30 by Alvin1Zhang