espressif/esp-idf #6527: ESP-MESH not possible to power down WiFi without reinitializing entire mesh stack

MisterScience5, 2021-02-11:
**Is your feature request related to a problem? Please describe.**

I am designing a MESH-connected light which is battery powered. To keep power consumption low, MESH is only enabled periodically to synchronize measurements and setup. However, to power down WiFi and successfully reconnect to MESH, it is necessary to initialize MESH nearly from scratch every time, otherwise MESH will be unable to connect. An additional problem arises as esp_mesh_start() seems to duplicate a queue every time it is started, leading to a memory leak. This has been reported as a seperate issue.

**Describe the solution you'd like**

I feel esp_mesh_stop() and esp_mesh_start() should be more equivalent to esp_wifi_stop() and esp_wifi_start(), i.e. that esp_mesh_start() can be successfully rerun after esp_mesh_stop() is called, and any functions that require reinit of the mesh should happen in esp_mesh_deinit().

Any other solutions that allow WiFi to be powered down intermittently while using MESH would be very helpful.

Ideally I would like to be able to power down WiFi while using mesh in one of the following manners (examples below do not work):

`Example 1: (Does not work, as it will not connect to other AP's on enable - I have been unable to figure out why, as it very rarely does successfully connect. My feeling is that this sometimes happens after it has not connected successfully for a minimum of approx. 15 minutes, but I do not have any good data.)
Disable:
esp_mesh_set_self_organized(false, false);
esp_mesh_disconnect();
esp_wifi_stop();

Enable:
esp_wifi_start();
esp_mesh_set_self_organized(true, false);
esp_mesh_connect();

Example 2: (Does not work as esp_mesh_stop() deinits the mesh stack, and requires esp_mesh_init() and full configuration to be run again)
Disable:
esp_mesh_stop();
esp_wifi_stop();

Enable:
esp_wifi_start();
esp_mesh_start();
esp_mesh_set_self_organized(true, false);
esp_mesh_connect();`

**Describe alternatives you've considered**

I have tried hundreds of permutations to get the MESH to reliably restart after it has been disconnected and WiFi powered down, but so far my only working alternative has been to reinit pretty much everything every time I want to set up the MESH again.

These are my current functions that do work, but were very difficult to get working and also suffer from the aforementioned memory leak:

`esp_err_t mesh_disable(void) {  // LAST_VERSION
  if (mesh_on) {
    is_mesh_connected = false;
    if (esp_mesh_is_root()) {
      node_status[0].connected = false;
    }
    if (!esp_mesh_is_root()) {
      esp_wifi_scan_stop();
    }
    vTaskDelay(100 / portTICK_PERIOD_MS);
    ESP_ERROR_CHECK_WITHOUT_ABORT(esp_mesh_flush_scan_result());
    ESP_ERROR_CHECK_WITHOUT_ABORT(esp_mesh_flush_upstream_packets());
    vTaskDelay(500 / portTICK_PERIOD_MS);
    ESP_ERROR_CHECK_WITHOUT_ABORT(esp_mesh_stop());
    vTaskDelay(100 / portTICK_PERIOD_MS);
    vTaskSuspend(get_message_handle);
    if (esp_mesh_is_root()) {
      vTaskSuspend(send_ota_to_node_handle);
      vTaskSuspend(send_ota_chunk_handle);
    }
    ESP_ERROR_CHECK_WITHOUT_ABORT(esp_mesh_deinit());
    ESP_ERROR_CHECK_WITHOUT_ABORT(esp_wifi_stop());
    mesh_on = false;
  } else {
    ESP_LOGE(MESH_TAG, "Mesh network already disabled!");
  }
  return ESP_OK;
}

esp_err_t mesh_enable(void) {
  esp_err_t err = ESP_OK;
  if (!mesh_on) {
    if (mesh_init_done == false) {
      ESP_LOGW(MESH_TAG, "Initializing mesh network");
    } else {
      ESP_LOGW(MESH_TAG, "Enabling mesh network");
    }
    if (mesh_init_done == false) {
      disableCore0WDT();
      ESP_ERROR_CHECK_WITHOUT_ABORT(esp_netif_init());
      ESP_ERROR_CHECK_WITHOUT_ABORT(esp_event_loop_create_default());
      ESP_ERROR_CHECK_WITHOUT_ABORT(esp_netif_create_default_wifi_mesh_netifs(&netif_sta, NULL));
      wifi_init_config_t config = WIFI_INIT_CONFIG_DEFAULT();
      ESP_ERROR_CHECK_WITHOUT_ABORT(esp_wifi_init(&config));
      ESP_ERROR_CHECK_WITHOUT_ABORT(esp_event_handler_register(IP_EVENT, IP_EVENT_STA_GOT_IP, &ip_event_handler, NULL));
      ESP_ERROR_CHECK_WITHOUT_ABORT(esp_wifi_set_storage(WIFI_STORAGE_RAM));
      ESP_ERROR_CHECK_WITHOUT_ABORT(esp_wifi_set_mode(WIFI_MODE_APSTA));
      ESP_ERROR_CHECK_WITHOUT_ABORT(esp_wifi_set_ps(WIFI_PS_NONE));
    }
    ESP_ERROR_CHECK_WITHOUT_ABORT(esp_wifi_set_protocol(WIFI_IF_AP, WIFI_PROTOCOL_LR));
    ESP_ERROR_CHECK_WITHOUT_ABORT(esp_wifi_set_protocol(WIFI_IF_STA, WIFI_PROTOCOL_LR));

    ESP_ERROR_CHECK_WITHOUT_ABORT(esp_wifi_start());

    ESP_ERROR_CHECK_WITHOUT_ABORT(esp_wifi_set_max_tx_power(78));

    ESP_ERROR_CHECK_WITHOUT_ABORT(esp_mesh_init());
    ESP_ERROR_CHECK_WITHOUT_ABORT(setup_root());
    if (mesh_init_done == false) {
      ESP_ERROR_CHECK_WITHOUT_ABORT(esp_event_handler_register(MESH_EVENT, ESP_EVENT_ANY_ID, &mesh_event_handler, NULL));
    }
    ESP_ERROR_CHECK_WITHOUT_ABORT(esp_mesh_set_topology(MESH_TOPO_CHAIN));
    ESP_ERROR_CHECK_WITHOUT_ABORT(esp_mesh_set_max_layer(mesh_settings.max_layer));
    ESP_ERROR_CHECK_WITHOUT_ABORT(esp_mesh_set_vote_percentage(1));
    ESP_ERROR_CHECK_WITHOUT_ABORT(esp_mesh_set_xon_qsize(64));
    if (mesh_settings.pwr_save) {
      ESP_ERROR_CHECK_WITHOUT_ABORT(esp_mesh_enable_ps());
      ESP_ERROR_CHECK_WITHOUT_ABORT(esp_mesh_set_ap_assoc_expire(60));
      ESP_ERROR_CHECK_WITHOUT_ABORT(esp_mesh_set_announce_interval(5000, 30000));
    } else {
      ESP_ERROR_CHECK_WITHOUT_ABORT(esp_mesh_disable_ps());
      ESP_ERROR_CHECK_WITHOUT_ABORT(esp_mesh_set_ap_assoc_expire(10));
    }
    mesh_cfg_t cfg = MESH_INIT_CONFIG_DEFAULT();
    memcpy((uint8_t *)&cfg.mesh_id, mesh_settings.id, 6);
    cfg.channel = mesh_settings.channel;
    cfg.allow_channel_switch = true;
    ESP_ERROR_CHECK_WITHOUT_ABORT(esp_mesh_set_ap_authmode(mesh_settings.auth_mode));
    esp_mesh_set_ie_crypto_key(mesh_settings.crypt_key, strlen(mesh_settings.crypt_key));
    cfg.mesh_ap.max_connection = mesh_settings.max_conn;
    memcpy((uint8_t *)&cfg.mesh_ap.password, mesh_settings.ap_pass, strlen(mesh_settings.ap_pass));
    ESP_ERROR_CHECK_WITHOUT_ABORT(esp_mesh_set_config(&cfg));
    ESP_ERROR_CHECK_WITHOUT_ABORT(esp_mesh_start());
    mesh_on = true;
    if (mesh_init_done == false) {
      ESP_LOGI(MESH_TAG, "Creating mesh receive task, mesh_init_done now %d", mesh_init_done);
      xTaskCreate(&get_message, "get_message", 9216, NULL, 5, &get_message_handle);
      if (esp_mesh_is_root()) {
        ESP_LOGI(MESH_TAG, "Creating root mesh tasks");
        ota_send_queue = xQueueCreate(10, sizeof(mesh_ota_message_t));
        xTaskCreate(&send_ota_to_node, "send_ota_to_node", 2184, NULL, 5, &send_ota_to_node_handle);
        xTaskCreate(&send_ota_chunk, "send_ota_chunk", 4980, NULL, 5, &send_ota_chunk_handle);
      }
      enableCore0WDT();
      ESP_ERROR_CHECK_WITHOUT_ABORT(esp_read_mac(mesh_self_addr.addr, 0));
    } else {
      ESP_ERROR_CHECK_WITHOUT_ABORT(esp_mesh_comm_p2p_start());
    }
    ESP_ERROR_CHECK_WITHOUT_ABORT(esp_mesh_set_self_organized(true, false));
    if (mesh_init_done == false) {
      if (esp_mesh_is_root()) {
        memcpy(&mesh_root_addr, &mesh_self_addr, sizeof(mesh_addr_t));
        memcpy(&node_status[0].address, &mesh_self_addr, sizeof(mesh_addr_t));
        node_status[0].exists = true;
        node_status[0].connected = true;
        node_status[0].current_version = SOFTWARE_VERSION;
        node_status[0].uptime = (uint32_t)(esp_timer_get_time() / 1000000ULL);
        node_status[0].mesh_layer = esp_mesh_get_layer();
      }
      mesh_init_done = true;
      ESP_LOGI(MESH_TAG, "Set mesh_init_done to %d", mesh_init_done);
    }
  } else {
    ESP_LOGW(MESH_TAG, "Mesh network already enabled!");
  }
  return err;
}`

**Additional context**

Add any other context or screenshots about the feature request here.

MisterScience5, 2022-10-15:
I've been looking into this some more, and it seems the reason method #1 from above doesn't work is that it leaves to opposing flags in place so the mesh handler believes it is both connected and disconnected, disallowing both, this is with the following code run with a mesh connected:

```
ESP_ERROR_CHECK(esp_wifi_start());
vTaskDelay(3 * 1000 / portTICK_PERIOD_MS);
ESP_ERROR_CHECK(esp_mesh_set_self_organized(true, true));
```

Event handler subclause for stop_reconnect:

```
case MESH_EVENT_STOP_RECONNECTION:
    {
        ESP_LOGI(MESH_TAG, "<MESH_EVENT_STOP_RECONNECTION>");
        ESP_ERROR_CHECK(esp_mesh_disconnect());
        vTaskDelay(1000 / portTICK_PERIOD_MS);
        ESP_ERROR_CHECK(esp_mesh_connect());
    }
```
Output:
```
I (62036) wifi:mode : sta (30:c6:f7:2f:e0:fc) + softAP (30:c6:f7:2f:e0:fd)
I (62037) wifi:enable tsf
W (62038) wifi:<MESH AP>adjust channel:1, secondary channel offset:1(40U)
I (62041) wifi:Total power save buffer number: 16
I (65073) mesh: [IO]enable self-organizing, search parent<adaptive>
I (65684) mesh: [SCAN][ch:1]AP:4, other(ID:0, RD:0), MAP:1, idle:0, candidate:1, root:1, topMAP:0[c:2,i:2][00:00:00:00:00:00][FIXED-ROOT]<>
I (65685) mesh: 7369[selection]try rssi_threshold:-78, backoff times:0, max:5<-78,-82,-85>
I (65694) mesh: [DONE]connect to parent:ESPM_715EA0, channel:1, rssi:-42, ac:67:b2:71:5e:a1[layer:1, assoc:0], my_vote_num:0/voter_num:0, rc[00:00:00:00:00:00/-120/2]
I (65735) mesh: <mesh_parent_select_done,1662>parent is set<stop reconnect>g_is_wifi_connecting:1, g_is_wifi_disconnecting:0, g_mesh_stop_reconnection:2
I (00:01:02.323) mesh: <MESH_EVENT_STOP_RECONNECTION>
I (65743) mesh: <MESH_NWK_PARENT_DISCONNECTED>already disconnected, ignore it
I (66743) mesh: <MESH_NWK_PARENT_DISCONNECTED>already reconnecting, ignore it, stop_reconnection:2
```

Leadrive, 2023-04-29:
I want to know how to switch layer 2 nodes to another root, when the current root is down or disappear.
I tried more method, but it did not work ok.
So I must restart the layer 2 nodes to swtich to another root.

I use esp_mesh_fix_root(true);
In sample mesh, maybe include some root(wifi or ethernet or usb modem), if the current root is down, I want the layer2 can swtich to another automatic.

Leadrive, 2023-04-29:
I want to know how to switch layer 2 nodes to another root, when the current root is down or disappear.
I tried more methods, but it did not work ok.
So I must restart the layer 2 nodes to swtich to another root.

I use esp_mesh_fix_root(true);
In sample mesh, maybe include some root(wifi or ethernet or usb modem), if the current root is down, I want the layer2 can swtich to another automatic.

zhangyanjiaoesp, 2023-09-12:
@Leadrive For release v4.4, 8756d9545f2208ce4a56718eceea6f4d91b27502 this MR has fixed the layer2 connect issue when fixed root disappear, please update your branch and have a try.

MaxwellAlan, 2023-11-17:
Hi @Leadrive Plz have a try and feel free to reopen if issue still exist.

MisterScience5, 2023-11-17:
Hi, this one did diverge a bit from the original, but the feature request is still valid:)

zhangyanjiaoesp, 2024-03-19:
@MisterScience5 we will consider feature request, but the priority is not high.

zhangyanjiaoesp, 2025-01-13:
@MisterScience5 Since there is considerable overlap between mesh_deinit and mesh_stop, it is challenging to separate them in the same way as wifi_stop and wifi_deinit. Therefore, we have decided to keep the existing functions unchanged.

closed 2025-01-22 by Alvin1Zhang