espressif/esp-idf #6528: esp_event_loop_delete_default()  LOOP BEHAVIOR (NOT RESPONSE)

JosueSanchezEZM, 2021-02-11:
Hi everyone, I am testing the pppos_client example from esp-idf v4.2 . I am changing a little bit the structure of the code but without affecting the functionality. I say this because I already have a project structured in a different way but running the same behavior as the example. 

I was trying to close ppp connection and start it again, so for that, I needed to use esp_event_loop_delete_default() to clean s_default_loop variable that host the loop which notifies esp IP_EVENTS events related with PPP automatically, this in order to use esp_event_loop_create_default() again to establish ppp  initial configurations (nic attachment and so on). 

My problems is that when using "esp_event_loop_delete_default()", having already used esp_event_loop_create_default, my program gets stuck at "esp_event_loop_delete_default()" specifically in the vTaskDelete(loop->task) function, inside of itself.

I would like to know if there is any correct way or order  to use this function, because even if y run esp_event_loop_create_default() and esp_event_loop_delete_default() consecutively, my program gets stuck. 

Thanks so much for your comments and help.

projectgus, 2021-02-12:
Hi @JosueSanchezEZM,

What version of ESP-IDF are you using?

Rather than a description of the code that triggers this problem, is it possible to please provide a sample program that triggers it?

Thanks,

Angus

JosueSanchezEZM, 2021-02-12:
Yes of course. I am using esp-idf v4.2 release. As an example project, I attach pppos_client from esp-idf examples stack in which I replicate my issue. What I was trying to do is to close PPP session immediately after PPP interface get IP, this only to validate the possibility to attach again the netif interface for PPP and start and close PPP session several times, but to achieve this error using this example project now I am getting a new error about timeout with esp_modem_stop_ppp() as you can see below:
[image]

My specific problem is about the use of esp_event_loop_delete_default() as I show you below ( also present in  .zip project file I attached here)

:
[image]

Attached file ( project ): https://drive.google.com/file/d/1LBLzXp3SNNmlMeGeNdfS-iBqx9oq22KL/view?usp=sharing

JosueSanchezEZM, 2021-02-12:
Regarding esp_event_loop_delete_default(), what happens to me is that when I used this function anywhere, of course after creating the default loop with esp_event_loop_create-default(), inside the delete function I get stuck and then and error appears to me:
Loop structure:
[image]

vTasKDelete part inside esp_event_loop_delete_default():
[image]

Error:
[image]

david-cermak, 2021-02-16:
@JosueSanchezEZM Could you please update to the `release/v4.2` [branch](https://github.com/espressif/esp-idf/commits/release/v4.2)? There has been a fix on the deinitializetion of pppos client in 84039dc77d which unfortunately didn't make it to `v4.2` tag. The errors you've posted above, caused by the `esp_modem_stop_ppp()`  exiting the PPP mode should be fixed by 84039dc77d.

As for the `esp_event_loop_delete_default()` and the related crashes, please do not destroy the default event loop, not *until* any of the network/wifi component is up and running, since they rely on the system events. It should be called *after* the wifi and all the `esp-netif`s has been stopped and deleted.

JosueSanchezEZM, 2021-02-22:
Thanks, @david-cermak I updated my esp-idf folder. Now all events are watched by this handler thanks.

closed 2021-02-22 by JosueSanchezEZM

acidclear, 2024-08-18:
This is currently happening on the ESP-WiFi-Mesh example after spinning up the default netifs for mesh.