espressif/esp-idf #6529: Using SPIFFS and SDSPI together at the same time results in a lockup

pyjamasam, 2021-02-11:
## Environment

- Development Kit: none
- Kit version: N/A
- Module or chip used:  ESP32-WROOM-32D
- IDF version: v4.2-238-g8cd16b60f and v4.4-dev-4-g73db14240
- Build System: idf.py
- Compiler version: For both esp-idf v4.2 and v4.4-dev I am using xtensa-esp32-elf-gcc (crosstool-NG esp-2020r3) 8.4.0
- Operating System: macOS
- Using an IDE?: VSCode and Command line.  Doesn't change the outcome of the issue
- Power Supply: USB

## Problem Description

When writing to a SDSPI connected SD card (Using the VSPI bus with the standard pins via IOMUX) in one task and trying to read a file from a SPIFFS partition in another task from the same core an exception occurs.

The current example code is using fatfs directly vs through the VFS layer as I was attempting to trim down the code that caused the problem.  But this problem also occurs when using the vfs layer for SD card access.

### Expected Behavior

Data should be written to the SD card, and data should be read from the SPIFFS partition

### Actual Behavior

When running with my project sdkconfig file settings the chip locks up and eventually resets due to the watchdog with a DoubleException

When running the a default sdkconfig file a core panic occurs due to 'Cache disabled but cached memory region accessed'

### Steps to reproduce

A repository with code to reproduce the error
https://github.com/pyjamasam/SPIFFS_SDSPI_Test

Connect a SD card via the VSPI standard pins:
  MOSI = GPIO23
  MISO = GPIO19
  CLK = GPIO18
  CS = GPIO5

I have tried with a 10k pull up to 3.3v on MOSI.

Running the referenced code will result in an error (when the SD card initializes correctly - once the error occurs the esp will restart and then the SD card will not initialize correctly.  It then takes a few restart cycles until the SD card is ready to be used again).

### Code to reproduce this issue

https://github.com/pyjamasam/SPIFFS_SDSPI_Test

## Debug Logs

```
rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:6924
ho 0 tail 12 room 4
load:0x40078000,len:14336
load:0x40080400,len:3672
0x40080400: _init at ??:?

entry 0x40080678
I (29) boot: ESP-IDF v4.4-dev-4-g73db14240-dirty 2nd stage bootloader
I (29) boot: compile time 16:23:44
I (30) boot: chip revision: 1
I (33) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (41) boot.esp32: SPI Speed      : 40MHz
I (45) boot.esp32: SPI Mode       : DIO
I (50) boot.esp32: SPI Flash Size : 4MB
I (54) boot: Enabling RNG early entropy source...
I (60) boot: Partition Table:
[... 103 lines cut ...]

0x400d9e33: spiffs_page_data_check at /Users/chris/esp-idf/components/spiffs/spiffs/src/spiffs_nucleus.c:20

0x400dc0f9: spiffs_object_read at /Users/chris/esp-idf/components/spiffs/spiffs/src/spiffs_nucleus.c:2035

0x400d8d48: spiffs_hydro_read at /Users/chris/esp-idf/components/spiffs/spiffs/src/spiffs_hydrogen.c:414

0x400d92ad: SPIFFS_read at /Users/chris/esp-idf/components/spiffs/spiffs/src/spiffs_hydrogen.c:426

0x400d88f2: vfs_spiffs_read at /Users/chris/esp-idf/components/spiffs/esp_spiffs.c:504

0x400d203a: esp_vfs_read at /Users/chris/esp-idf/components/vfs/vfs.c:449 (discriminator 3)

0x400e405d: _fread_r at /builds/idf/crosstool-NG/.build/HOST-x86_64-apple-darwin12/xtensa-esp32-elf/src/newlib/newlib/libc/stdio/fread.c:228

0x400e40ed: fread at /builds/idf/crosstool-NG/.build/HOST-x86_64-apple-darwin12/xtensa-esp32-elf/src/newlib/newlib/libc/stdio/fread.c:266

0x400d646d: processing_core0_1(void*) at /Users/chris/Documents/Projects/SPIFFS_SDSPI_Test/build/../main/main.cpp:83

0x4008923d: vPortTaskWrapper at /Users/chris/esp-idf/components/freertos/port/xtensa/port.c:168
```

## Other items if possible

- Multiple sdkconfig files are in the referenced repository which result in slightly differnt lockup scenarios.  One (sdkconfig.project) results in a silent lockup, and the other (sdkconfig) produces the Cache panic.

igrr, 2021-02-11:
@pyjamasam For the cache issue, could you please try modifying `spi_bus_intr_disable` function in /Users/chris/esp-idf/components/driver/spi_master.c:512 as follows:
```c++
static void SPI_MASTER_ISR_ATTR spi_bus_intr_disable(void *host)
```
(add SPI_MASTER_ISR_ATTR).

For the lockup issue, it seems you have interrupt watchdog disabled in sdkconfig. Could you please try enabling it and see if this yields any additional information (e.g. a backtrace)?

pyjamasam, 2021-02-11:
Thanks for the quick response Ivan,

Enabling the interrupt watchdog did not result in any change to the lockup observed when using my sdkconfig.project file.  

It still resulted in a silent lockup and then the following message after restarting
```W (196) boot.esp32: PRO CPU has been reset by WDT.
W (202) boot.esp32: WDT reset info: PRO CPU PC=0x400803c3
0x400803c3: _DoubleExceptionVector at /Users/chris/esp-idf/components/freertos/port/xtensa/xtensa_vectors.S:457

W (208) boot.esp32: WDT reset info: APP CPU PC=0x40083586
0x40083586: spi_flash_op_block_func at /Users/chris/esp-idf/components/spi_flash/cache_utils.c:115 (discriminator 1)
```

And the default sdkconfig that is also in the repository had that flag enabled and produced the original stack trace included in the issue.

BUT 

Adding `SPI_MASTER_ISR_ATTR` to `spi_master.c` for the `spi_bus_intr_disable` function results the correct behaviour of being able to read from SPIFFS and write the SD card at the same time.  In both IDF v4.2 and master (v4.4-dev).

I will test the fix in my larger application and report back the status of that.

pyjamasam, 2021-02-12:
I have been running at test for about 20 hours now, and I can successfully write the the SD card and continue to read from SPIFFS.  This is using the IDF v4.2 tag with the modification to spi_master.c.

Thank you very much for the help with with.  From my standpoint I consider this to be fixed, but I'd like to see it as part of the official repo.
Is there anything I need to do further to help this get added in to an upcoming release?

igrr, 2021-02-13:
Thanks for the confirmation @pyjamasam, we will fix this in the master branch and also backport to release/v4.2. This issue will be closed once the fix is merged.

AxelLin, 2021-08-05:
> Thanks for the confirmation @pyjamasam, we will fix this in the master branch and also backport to release/v4.2. This issue will be closed once the fix is merged.

Hi @igrr 
The issue is reported again by https://github.com/espressif/esp-idf/issues/7368
How is the status of the fix?

ginkgm, 2021-08-05:
Hi @AxelLin ,

It's being reviewed internally. There's no other considerations about it, so I think the reviewing will be fast.

closed 2021-08-06 by espressif-bot

Alvin1Zhang, 2021-08-09:
Thanks for reporting, fix on master branch is available https://github.com/espressif/esp-idf/commit/de0f24e373729dc7375dbc697c89d63d4626dfe9, we are back porting the fix to release/4.2, thanks.