espressif/esp-idf #6530: esp_mesh_waive_root() stops AP mode of new root

RajSinghGaur, 2021-02-12:
**Environment:**
idf version: 4.2
modules: esp32
OS: linux
wifi router: mobile hotspot

**Description:**
I flashed two esp32 with waive root node code, let them be N1 and N2, and followed the following prodcedure:
1. kept mobile near N1 to make it root
2. waited for the waive function to start
3. ket the mobile to N2 to make it the new root

**Expected Behaviour**
N2 become the new root and N1 connect to it.

**Observed Behaviour**
N2 becomes the new root but N1 is not able to connect to it. When the N1 node was restarted it connected to the mobile hotspot instead of the root, i.e. N2. This occurance occurs once after 3-5 times running the above procedure.

**Code Repository**
https://github.com/RajSinghGaur/Mesh-Waive-Root.git

**Logs**
Logs for both N1 and N2 are in repository by names N1.txt and N2.txt repectively.

**Possible Reasons**
During waive the AP of N2, i.e. the new root, was not working.

Alvin1Zhang, 2021-02-18:
Thanks for reporting.

zhangyanjiaoesp, 2023-06-29:
@RajSinghGaur Do you still have this issue ? I have tested with your code locally, but I didn't reproduce your problem.

> This occurance occurs once after 3-5 times running the above procedure.

Do you mean after performing a root switch successfully, then restart the board and executing it again?

closed 2024-03-06 by Alvin1Zhang