espressif/esp-idf #6531: Can not initiate ppp connection with SIM800L using pppos_client

GunashekarG, 2021-02-12:
OS: Ubuntu Mate(20.04)
MCU: ESP Wroom 32
ESP-IDF version: ESP-IDF v4.3-dev-2586-g526f68239
Example: pppos_client example

SDK config file: 
[sdkconfig.txt](https://github.com/espressif/esp-idf/files/5971780/sdkconfig.txt)

I am trying to interface SIM 800L with ESP32 and trying to initiate ppp connection so that I can be able to reach out the server and upload files later. The connections are typical UART connection and module is working fine. I have tried to run the code that is provided inbuilt and I am getting error in sending AT command via a function. I am attaching the error image for your referance.
[image]
[Here I have added few log statements for debugging purpose, please ignore that].

If anyone has successfully used SIM800L with ESP32 and successfully initiated the ppp connection or do anyone knows what exactly is the problem, please let me know. Any lead will do good. Thanks in advnce.

GunashekarG, 2021-02-15:
Also, I am trying to use Cavli C10AM LTE module on the similar lines mentioned and getting same error as the screenshot attached along the post above. If anyone has the lead for how to go further by resolving this issue, that would be great. Thanks in advance.

david-cermak, 2021-02-22:
Hi @GunashekarG 

The `sync failed` error you are experiencing above is just an initial communication failure, meaning that the ESP32 sends `AT\r` on TX UART line and expects a `OK` on the RX line, which never comes. I would suggest double checking the wiring and making sure the device has properly initialized and is not kept in reset. you can also check the Rx and Tx lines with a logic analyzer.

Although the SIM800L device will probably not work (depending on your internet provider and G2 support in your country), this initial sequence should work nonetheless and the device would at least respond to these basic AT commands.

Alvin1Zhang, 2021-08-31:
@GunashekarG Thanks for reporting. Would you please help share if any updates for this issue? Thanks.

Alvin1Zhang, 2021-09-07:
Thanks for reporting, would close due to short of feedback, feel free to reopen.

closed 2021-09-07 by Alvin1Zhang