espressif/esp-idf #6533: ESP32-C3 missing complete interrupts list

imheresamir, 2021-02-13:
Hi,

I believe the complete "Interrupt hardware source table" is missing in the appropriate esp32c3 `soc.h` file. Also the TRM is currently missing from Espressif documentation site. Would really love an official SVD (tracking issue: #2214).

Thank you,
Samir

igrr, 2021-02-13:
Hi @imheresamir, you can find the interrupt list in https://github.com/espressif/esp-idf/blob/master/components/soc/esp32c3/interrupts.c or https://github.com/espressif/esp-idf/blob/master/components/soc/esp32c3/include/soc/interrupt_core0_reg.h. Although there is no TRM for ESP32-C3 yet, the "Interrupt Matrix" peripheral in ESP32-C3 is very similar to that of ESP32-S2 and ESP32, you may consult ESP32-S2 TRM for the description of the Interrupt Matrix.
The part which is different in ESP32-C3 is the additional interrupt controller between the interrupt matrix and the CPU; the registers for this module are listed after all the interrupt matrix registers here: https://github.com/espressif/esp-idf/blob/73db142403c6e5b763a0e1c07312200e9b622673/components/soc/esp32c3/include/soc/interrupt_core0_reg.h#L545-L848. Will see if we our documentation colleagues can make some preliminary description of this part available ahead of the TRM release.

Esp-Doc, 2021-03-01:
Hi @imheresamir , just a bit more information: the base address of interrupt matrix (interrupt controller) is 0x600C_2000.

closed 2021-03-02 by Alvin1Zhang