espressif/esp-idf #6534: TIMEOUT ERROR occurred for TCP MODBUS

harshalAcc, 2021-02-13:
## Environment

- Development Kit:      ESP32_Ethernet_kit_A_V1_1
- Kit version (for WroverKit/PicoKit/DevKitC): V1.1
- Module or chip used:  ESP32-WROVER-B
- IDF version (run ``git describe --tags`` to find it): V4.1.1
- Build System:        idf.py
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
    // 1.22.0-80-g6c4433a
- Operating System:     Windows
- (Windows only) environment type: ESP Command Prompt
- Using an IDE?: No
- Power Supply:   USB

## Problem Description
We are not able to get the response of MODBUS TCP request from PLC.

Basically we are using ethernet for TCP MODBUS. We are using TCP sample code which is provided into the example. Using this TCP MODBUS example we are trying to read the "Holding register (8960)" from Shindler PLC. But we are not able to get the response. If we test this code with other PC (Using Modscan) It is working fine. (Hardware wise PLC are compatible) 

Once more observation which I have is, If we are going read any other register like coil register which are not configured then we will get the exception response.

### Expected Behavior
We have to get the response for Holding Register.
### Actual Behavior
We are getting TIMOUT ERROR
### Steps to reproduce

1. step1 : Connect our board with Shindler PCL
2. step2 : Change the server IP address in to the code
3. step3 : Build and program our device
4. step4 : Our board are able connect with PLC and send the command.

// If possible, attach a picture of your setup/wiring here.

alisitsyn, 2021-02-15:
Hello @harshalAcc,

Thanks for reporting. 
The information provided is not enough to reproduce and find the reason for issues. Could you provide more information:
1. The IDF version v4.1.1 showed above does not support the Modbus TCP communication! Please clarify how you use the Modbus TCP in your project.
2. Are you using the modbus TCP master example from IDF and try to read the registers of PLC configured as a Modbus TCP Slave Device? If so, how do you configure the data dictionary for CID corresponding to "Holding register (8960)" from Shindler PLC in the master example? Please provide the changed code and send your existing log file of the example when you try to read the registers.
3. Please set `Default log verbosity` to Debug  in the `Component config - > Log Output` kconfig menu and then recompile example then repeat the connection from PLC and save log.
4. I need your example code with modifications and sdkconfig file to reproduce the issue.

Thanks.

harshalAcc, 2021-02-22:
Hello @alisitsyn ,

Thank you very much for your inputs,

I had configured the logging as you mentioned. I found the issue which is related to UID set by me.

Thanks

closed 2021-02-22 by harshalAcc

alisitsyn, 2021-02-23:
@harshalAcc,

Feel free to reopen if you have further issues.