espressif/esp-idf #6535: Websocket client does not send payload data to WEBSOCKET_EVENT_CLOSED event handler

abobija, 2021-02-13:
## Environment

- Development Kit: ESP32-Wrover-Kit
- Module or chip used: ESP32-WROOM-32
- IDF version: ESP-IDF v4.4-dev-4-g73db14240-dirty
- Build System: idf.py
- Compiler version: 8.4.0
- Operating System: Linux
- Power Supply: USB

## Problem Description

I'm using [`esp_websocket_client`](https://github.com/espressif/esp-idf/tree/master/components/esp_websocket_client) to communicate with websocket server. I have registered websocket event handler with `esp_websocket_register_events` and everything works awesome. In event handler I'm receiving instance of `esp_websocket_event_data_t` struct, and on `WEBSOCKET_EVENT_DATA` event I can read received websocket frame data. But, problem is when the server close connection (cleanly - op_code 8) with some data in websocket frame, event  `WEBSOCKET_EVENT_CLOSED` has been invoked, but I do not have access to the received data because `data_len` property of received `esp_websocket_event_data_t` is set to zero, and `data_ptr` is `NULL`.

As you can see in this line, websocket client dispatch the `WEBSOCKET_EVENT_CLOSED` event with `data` set to `NULL` and `data_len` set to zero:

https://github.com/espressif/esp-idf/blob/73db142403c6e5b763a0e1c07312200e9b622673/components/esp_websocket_client/esp_websocket_client.c#L701

### Expected Behavior

It would be nice to have access to websocket frame data in `WEBSOCKET_EVENT_CLOSED` event, because my websocket server send more details about closing reason in frame data.

### Code to reproduce this issue

```c
static void gw_websocket_event_handler(void* handler_arg, esp_event_base_t base, int32_t event_id, void* event_data) {
    esp_websocket_event_data_t* data = (esp_websocket_event_data_t*) event_data;

    switch (event_id) {
        case WEBSOCKET_EVENT_CLOSED:
            ESP_LOGD(TAG, "WEBSOCKET_EVENT_CLOSED (WS_OP: %d, payload_len=%d, data_len=%d)", data->op_code, data->payload_len, data->data_len);
            break;
    }
}
```

## Debug Logs

```
D (6811) application: WEBSOCKET_EVENT_CLOSED (WS_OP: 8, payload_len=24, data_len=0)
```

abobija, 2021-02-15:
Hi @david-cermak @ESP-Marius  sorry for tag but I see that you are most active recently on ws_client. Just want to ask you what do you think will this be possible to implement, and will it at all? My ws client really need to have ability to read websocket data on clean connection close.  I've never contribute to esp-idf, but if you want I can try to implement this and open PR?

david-cermak, 2021-02-16:
Hi @abobija 

First of all, please, do not hesitate to post a PR (just note that we are a bit slow in getting back to you with feedback and reviews, but all contributions are more than welcome. we might amend the changes a little, though, to be in line with the internal pipeline of reviews and tests).

About the close message payload: I believe it should be already available in the application code, but as a separate event (`WEBSOCKET_EVENT_DATA`). In order to receive the close packet data, you could just modify the event handler here

https://github.com/espressif/esp-idf/blob/master/examples/protocols/websocket/main/websocket_example.c#L72-L73

to read not only the close message code, but also any applicable data after that.
```c
if (data->op_code == 0x08 && data->data_len >= 2) {
```

abobija, 2021-02-16:
Thank you @david-cermak

I was thinking all the time that I need to read closing data inside of `WEBSOCKET_EVENT_CLOSED` event, but now I see that on clean close first will be raised `WEBSOCKET_EVENT_DATA` with data, and then `WEBSOCKET_EVENT_CLOSED`, so I need to record data (and closing code) in first event and than use recorded data later in second event if I wish. That makes sense, so this issue can be closed.

p.s: Thanks for info about PRs

Best regards.

closed 2021-02-16 by abobija

hitecSmartHome, 2024-06-27:
My websocket client code just freezes and does nothing after I got a close data from the server.

```cpp
if (data->op_code == 0x08 && data->data_len == 2) {
    printf("[WebsocketClient] - Got Close data\n");
    return;
}
```

When my esp gets this data, the ws client does nothing after that. Can't receive or send frames. What should I do with this? Even a reinit does not help with these

```cpp
void WebsocketClient::init() {
    client = esp_websocket_client_init(&config);
    esp_websocket_register_events(
        client,
        WEBSOCKET_EVENT_ANY,
        reinterpret_cast<esp_event_handler_t>(eventHandler),
        static_cast<void*>(this));
    esp_websocket_client_start(client);
}

bool WebsocketClient::reInit(){
    esp_err_t err = esp_websocket_client_destroy(client);
    if(err != ESP_OK){ return false; }
    init();
    return true;
}
```

It just stuck