espressif/esp-idf #6536: esp_eth_mac_enc28j60.c:24:10: fatal error: esp_rom_sys.h: No such file or directory

mntolia, 2021-02-14:
## Environment

- Development Kit:  ESP32-DevKitC
- IDF version: 4.2
- Platform: 3.4
- Build System:CMake
- Operating System: Windows
- Using an IDE?: PlatformIO over VSCode

## Problem Description

I am trying to get the Ethernet example working : https://github.com/espressif/esp-idf/tree/master/examples/ethernet/enc28j60

But I am getting this error while compiling:

`src/esp_eth_mac_enc28j60.c:24:10: fatal error: esp_rom_sys.h: No such file or directory`

### Code to reproduce this issue

Same as https://github.com/espressif/esp-idf/tree/master/examples/ethernet/enc28j60

## Debug Logs

```
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32dev.html
PLATFORM: Espressif 32 (3.0.0) > Espressif ESP32 Dev Module
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (esp-prog) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES:
 - framework-espidf 3.40200.210118 (4.2.0)
 - tool-cmake 3.16.4
 - tool-esptoolpy 1.30000.201119 (3.0.0)
 - tool-idf 1.0.1
 - tool-mconf 1.4060000.20190628 (406.0.0)
 - tool-ninja 1.9.0
 - toolchain-esp32ulp 1.22851.191205 (2.28.51) 
 - toolchain-xtensa32 2.80400.210114 (8.4.0)
Reading CMake configuration...
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 0 compatible libraries
Scanning dependencies...
No dependencies
Building in release mode
Compiling .pio\build\esp32dev\src\esp_eth_mac_enc28j60.o
src/esp_eth_mac_enc28j60.c:24:10: fatal error: esp_rom_sys.h: No such file or directory
```

negativekelvin, 2021-02-14:
Did you copy the example code from a more recent branch and try to build it against 4.2?

https://github.com/espressif/esp-idf/blob/release/v4.2/examples/ethernet/enc28j60/main/esp_eth_mac_enc28j60.c

mntolia, 2021-02-19:
Thanks @negativekelvin . That was the issue. 

I am facing another issue. There is some issue while loading the config:

```
rst:0xc (SW_CPU_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:4
load:0x3fff0034,len:7284
load:0x40078000,len:13240
load:0x40080400,len:4568
entry 0x400806f4
␛[0;32mI (29) boot: ESP-IDF 3.40200.210118 2nd stage bootloader␛[0m
␛[0;32mI (29) boot: compile time 16:13:18␛[0m
␛[0;32mI (29) boot: chip revision: 1␛[0m
␛[0;32mI (33) boot_comm: chip revision: 1, min. bootloader chip revision: 0␛[0m
␛[0;32mI (40) boot.esp32: SPI Speed      : 40MHz␛[0m
␛[0;32mI (44) boot.esp32: SPI Mode       : DIO␛[0m
␛[0;32mI (49) boot.esp32: SPI Flash Size : 4MB␛[0m
␛[0;32mI (53) boot: Enabling RNG early entropy source...␛[0m
␛[0;32mI (59) boot: Partition Table:␛[0m
␛[0;32mI (62) boot: ## Label            Usage          Type ST Offset   Length␛[0m
␛[0;32mI (70) boot:  0 nvs              WiFi data        01 02 00009000 00006000␛[0m
␛[0;32mI (77) boot:  1 phy_init         RF data          01 01 0000f000 00001000␛[0m
␛[0;32mI (85) boot:  2 factory          factory app      00 00 00010000 00100000␛[0m
␛[0;32mI (92) boot: End of partition table␛[0m
␛[0;32mI (96) boot_comm: chip revision: 1, min. application chip revision: 0␛[0m
␛[0;32mI (103) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x1063c ( 67132) map␛[0m
␛[0;32mI (138) esp_image: segment 1: paddr=0x00020664 vaddr=0x3ffb0000 size=0x019d8 (  6616) load␛[0m
␛[0;32mI (141) esp_image: segment 2: paddr=0x00022044 vaddr=0x40080000 size=0x00404 (  1028) load␛[0m
␛[0;32mI (145) esp_image: segment 3: paddr=0x00022450 vaddr=0x40080404 size=0x0b008 ( 45064) load␛[0m
␛[0;32mI (173) esp_image: segment 4: paddr=0x0002d460 vaddr=0x00000000 size=0x02bb8 ( 11192) ␛[0m
␛[0;32mI (178) esp_image: segment 5: paddr=0x00030020 vaddr=0x400d0020 size=0x31064 (200804) map␛[0m
␛[0;32mI (262) boot: Loaded app from partition at offset 0x10000␛[0m
␛[0;32mI (262) boot: Disabling RNG early entropy source...␛[0m
␛[0;32mI (263) cpu_start: Pro cpu up.␛[0m
␛[0;32mI (266) cpu_start: Application information:␛[0m
␛[0;32mI (271) cpu_start: Project name:     tusb_sample_descriptor␛[0m
␛[0;32mI (278) cpu_start: App version:      1␛[0m
␛[0;32mI (282) cpu_start: Compile time:     Feb 19 2021 16:11:27␛[0m
␛[0;32mI (288) cpu_start: ELF file SHA256:  2eb7b437b9fcf832...␛[0m
␛[0;32mI (294) cpu_start: ESP-IDF:          3.40200.210118␛[0m
␛[0;32mI (300) cpu_start: Single core mode␛[0m
␛[0;32mI (304) heap_init: Initializing. RAM available for dynamic allocation:␛[0m
␛[0;32mI (311) heap_init: At 3FF80000 len 00002000 (8 KiB): RTCRAM␛[0m
␛[0;32mI (317) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM␛[0m
␛[0;32mI (323) heap_init: At 3FFB3070 len 0002CF90 (179 KiB): DRAM␛[0m
␛[0;32mI (329) heap_init: At 3FFE0440 len 0001FBC0 (126 KiB): D/IRAM␛[0m
␛[0;32mI (336) heap_init: At 40078000 len 00008000 (32 KiB): IRAM␛[0m
␛[0;32mI (342) heap_init: At 4008B40C len 00014BF4 (82 KiB): IRAM␛[0m
␛[0;32mI (348) cpu_start: Pro cpu start user code␛[0m
␛[0;32mI (366) spi_flash: detected chip: generic␛[0m
␛[0;32mI (367) spi_flash: flash io: dio␛[0m
␛[0;32mI (367) cpu_start: Starting scheduler on PRO CPU.␛[0m
␛[0;32mI (375) enc28j60: revision: 0␛[0m
␛[0;31mE (375) enc28j60: enc28j60_verify_id(479): wrong chip ID␛[0m
␛[0;31mE (375) enc28j60: emac_enc28j60_init(853): vefiry chip ID failed␛[0m
␛[0;32mI (385) gpio: GPIO[4]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 ␛[0m
␛[0;31mE (395) esp_eth: esp_eth_driver_install(198): init mac failed␛[0m
ESP_ERROR_CHECK failed: esp_err_t 0xffffffff (ESP_FAIL) at 0x400844a4
file: "src/main.c" line 399
func: app_main
expression: esp_eth_driver_install(&eth_config, &eth_handle)

abort() was called at PC 0x400844a7 on core 0

Backtrace:0x40084927:0x3ffaf4a0 0x40084f39:0x3ffaf4c0 0x40088646:0x3ffaf4e0 0x400844a7:0x3ffaf550 0x400d4083:0x3ffaf570 0x400d5eaf:0x3ffaf640 0x40084f41:0x3ffaf670
```
[... 41 lines cut ...]
    mac_config.smi_mdc_gpio_num = -1; // ENC28J60 doesn't have SMI interface
    mac_config.smi_mdio_gpio_num = -1;
    esp_eth_mac_t *mac = esp_eth_mac_new_enc28j60(&enc28j60_config, &mac_config);

    eth_phy_config_t phy_config = ETH_PHY_DEFAULT_CONFIG();
    phy_config.autonego_timeout_ms = 0; // ENC28J60 doesn't support auto-negotiation
    phy_config.reset_gpio_num = -1;     // ENC28J60 doesn't have a pin to reset internal PHY
    esp_eth_phy_t *phy = esp_eth_phy_new_enc28j60(&phy_config);

    esp_eth_config_t eth_config = ETH_DEFAULT_CONFIG(mac, phy);
    esp_eth_handle_t eth_handle = NULL;
    ESP_ERROR_CHECK(esp_eth_driver_install(&eth_config, &eth_handle));

    /* ENC28J60 doesn't burn any factory MAC address, we need to set it manually.
       02:00:00 is a Locally Administered OUI range so should not be used except when testing on a LAN under your control.
    */
    mac->set_addr(mac, (uint8_t[]){
                           0x02, 0x00, 0x00, 0x12, 0x34, 0x56});

    /* attach Ethernet driver to TCP/IP stack */
    ESP_ERROR_CHECK(esp_netif_attach(eth_netif, esp_eth_new_netif_glue(eth_handle)));
    /* start Ethernet driver state machine */
    ESP_ERROR_CHECK(esp_eth_start(eth_handle));
```
Do you know why its not loading?

suda-morris, 2021-02-22:
See from your log:
␛[0;32mI (375) enc28j60: revision: 0␛[0m
␛[0;31mE (375) enc28j60: enc28j60_verify_id(479): wrong chip ID␛[0m
Looks like the SPI timing is not matched somehow with your hardware.

Alvin1Zhang, 2021-08-04:
@mntolia Thanks for reporting, please help check suggestion by @suda-morris , feel free to reopen.

closed 2021-08-04 by Alvin1Zhang