espressif/esp-idf #6537: "ESP-IDF in Custom CMake Projects" CMake example doesn't build

resetnow, 2021-02-14:
## Environment

- Development Kit: esp32-s2-saola-1
- IDF version: `v4.3-dev-2586-g526f68239`
- Build System: CMake
- Operating System: macOS
- Using an IDE?: VSCode

## Problem Description

The following example doesn't build out of the box:

https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/build-system.html#using-esp-idf-in-custom-cmake-projects

When running `idf.py` build, CMake will attempt to build the files with system compiler instead of a cross compiler. The reason is missing call to `__target_set_toolchain`.

I had to add the following lines before calling `project()`:

```
set(IDF_TARGET esp32s2)
set(CMAKE_TOOLCHAIN_FILE $ENV{IDF_PATH}/tools/cmake/toolchain-${IDF_TARGET}.cmake)

project(myprojectname C CXX)
```

projectgus, 2021-02-15:
Hi @resetnow,

Thanks for reporting this. Just to check: you mean the "idf_as_lib" example in the code, not the sample CMakeLists.txt file shown in the documentation page?

When building the `idf_as_lib` example, we provide two wrapper scripts - [build.sh](https://github.com/espressif/esp-idf/blob/526f682397a8cfb74698c601fd2c5b30e1433837/examples/build_system/cmake/idf_as_lib/build.sh) and [build-esp32.sh](https://github.com/espressif/esp-idf/blob/526f682397a8cfb74698c601fd2c5b30e1433837/examples/build_system/cmake/idf_as_lib/build-esp32.sh). `build.sh` is intended to build on the host, and `build-esp32.sh` builds for the embedded target. `build-esp32.sh` passes the `CMAKE_TOOLCHAIN_FILE` to the cmake command line and also sets the target.

You're right that it's possible to move these settings into the CMakeLists.txt file, however if doing this then you lose one of the benefits of using "normal" CMake - reusing the same cmake files on multiple target platforms. This is why in the example we demonstrate it with the wrappers. 

Is there something else in the examples or the docs that's unclear and should be changed?

resetnow, 2021-02-15:
Alright, thank you @projectgus, I understand the idea now. Yes, it would be great to mention it in documentation - just the fact that the CMake file is targeted for multiple platforms and thus toolchain need to be specified explicitly, as it is not immediately obvious there's a difference between overridden `project()` and `idf_build_process()`.

Another difference is the lack of project_description.json generation - it would be good if it was exported in a public API, so that users could write something like:

```
if (CMAKE_CROSSCOMPILING)
    idf_generate_project_json(OUTPUT ${CMAKE_BINARY_DIR})
endif()
```

As far as I understand, "monitor mode" needs this file (at least when launched from VS Code).

Thank you for your help :)

zikalino, 2023-11-28:
closing as an old issue and already explained by @projectgus

closed 2023-11-28 by zikalino