espressif/esp-idf #6538: httpd websocket callback does receive valid user_ctx

harbaum, 2021-02-14:
## Environment

- IDF version: v4.4-dev-4-g73db14240

## Problem Description

When working with websockets under httpd the user_ctx provided by the user in httpd_uri_t is not passed to the httpd_req_t returned by the handler callback.

### Expected Behavior

The data provided via httpd_uri_t.user_ctx shows up in httpd_req_t.user_ctx as it does with regular http requests.

### Actual Behavior

httpd_req_t.user_ctx is NULL

yuanjianmin, 2021-03-04:
@harbaum Please take this patch to test the modify whether could meet your requirement. Thanks.
[0001-httpd-Support-user_ctx-in-websocket-handler-callback.zip](https://github.com/espressif/esp-idf/files/6080524/0001-httpd-Support-user_ctx-in-websocket-handler-callback.zip)

harbaum, 2021-03-04:
I can confirm, that this patch does what it's intended to. My Micropython websocket setup now works nicely.

BTW: Is there a particular reason why the "user_data" is named "user_ctx" in this particular case?

yuanjianmin, 2021-03-04:
> user_data

There is no special reason, it may be the naming style and habit.

harbaum, 2021-03-04:
Micropython/LVGL use automatic parsers to make use of the user_data fields. That does not work for user_ctx. Is it possible that these be renamed to user_data for consitency?

yuanjianmin, 2021-03-04:
> 
> 
> Micropython/LVGL use automatic parsers to make use of the user_data fields. That does not work for user_ctx. Is it possible that these be renamed to user_data for consitency?

I didn't understand that Micropython/LVGL's use of user_data parsing is related to user_ctx in the code. Can you point out where user_ctx is displayed specifically? Can give an simple example.

harbaum, 2021-03-04:
The espidf bindings for Micropython as used by LVGL are automatically generated by a script. To carry the micropython context the user_data field is automatically being used.

This is the line where the generator script finds the user_data and uses it:
https://github.com/lvgl/lv_binding_micropython/blob/b0f3b7ede698bb095512795315a9b93d03f62f2f/gen/gen_mpy.py#L1856

yuanjianmin, 2021-03-04:
I think you could change `user_data` to `user_ctx` in the line you mentioned. Modifying the IDF code is not easy, and there are many places to modify. You can choose a simple plan to modify and test first. Hope that can help you.

closed 2021-03-12 by espressif-bot