espressif/esp-idf #6539: wake up from deep sleep by ULP

chegewara, 2021-02-14:
Im not sure if it is esp-idf or arduino issue, or just esp-idf example, but this is not working:
```
I_WR_REG_BIT(RTC_CNTL_STATE0_REG, RTC_CNTL_ULP_CP_SLP_TIMER_EN_S, 0),
```
instead this is working:
```
I_WR_REG_BIT(RTC_CNTL_STATE0_REG, RTC_CNTL_ULP_CP_SLP_TIMER_EN, 0),
```
https://github.com/espressif/esp-idf/blob/master/examples/system/deep_sleep/main/deep_sleep_example_main.c#L361

zikalino, 2022-02-24:
@sudeep-mohanty please take a look

chegewara, 2022-02-24:
Wow, nice timing, one year later.

ESP-Marius, 2022-09-01:
@chegewara Yeah, sometimes issues slips through some cracks, sorry!

Do you remember why you thought it wasn't working? A simple test from my side (just doing a `I_WR_REG_BIT(RTC_CNTL_STATE0_REG, RTC_CNTL_ULP_CP_SLP_TIMER_EN_S, 0)` and reading the results from the main CPU seems to show that it is behaving as expected.

chegewara, 2022-09-01:
As far as i remember it just didnt wake up device at all.
I had some app with deep sleep and reading analog pin, then wake up procedure when value is ...

ESP-Marius, 2022-09-01:
Hmm, my suspicion is that: the ULP failed to wake-up the main CPU for some reason (possibly related to https://github.com/espressif/esp-idf/issues/8341), and by changing `RTC_CNTL_ULP_CP_SLP_TIMER_EN` instead you actually made it **not** stop the timer. 

This allowed the ULP to wake-up and run again, and this time it successfully woke up the main CPU, "solving" the issue.

Not sure if we can do much more to debug it more at this point, if you think my explanation sounds reasonable then l suggest we close this issue, as `I_WR_REG_BIT` seemed to work fine according to my tests.

I'm planning to follow up on https://github.com/espressif/esp-idf/issues/8341 as well, so if that is the root cause then we should be able to improve upon it.

chegewara, 2022-09-01:
My ULP expertise level is so low, so any explanation is reasonable.

Thanks

closed 2022-09-01 by chegewara

ESP-Marius, 2022-09-01:
Thanks for quick replies as always @chegewara, we are working on improving our gitlab issue handling to avoid delays and "no responses" like this. So hopefully you should see less of it in the future :)