espressif/esp-idf #6543: ESP-IDF build error

SNSIOTP, 2021-02-15:
Seeking assistance please. Tried to build Blink example and got the following error: 

Loading defaults file D:/esp/esp-idf/examples/get-started/blink/sdkconfig.defaults...
CMake Error at ../../../tools/cmake/component.cmake:305 (message):
  Include directory 'D:/esp/esp-idf/components/mbedtls/mbedtls/include' is
  not a directory.
Call Stack (most recent call first):
  ../../../tools/cmake/component.cmake:477 (__component_add_include_dirs)
  ../../../components/mbedtls/CMakeLists.txt:4 (idf_component_register)

-- Configuring incomplete, errors occurred!
See also "D:/esp/esp-idf/examples/get-started/blink/build/CMakeFiles/CMakeOutput.log".
cmake failed with exit code 1

georgik, 2021-02-16:
@SNSIOTP Thank you for posting the issue.
How was the esp-idf installed?
Do you use Windows Installer http://dl.espressif.com/dl/esp-idf-tools-setup-online-2.5.exe ?
It seems that the copy of esp-idf does not contain all submodules. Suggested workaround:
```
cd D:/esp/esp-idf/
git submodule update --init --recursive
```

SNSIOTP, 2021-02-16:
@georgik Ths issue is resolved. As you mentioned there were some missing pieces which is fixed by fresh download.

Thank you. Issue can be closed now.

closed 2021-02-16 by SNSIOTP