espressif/esp-idf #6544: xEventGroupWaitBits triggers watchdog (wdt)

mdvorak, 2021-02-15:
## Environment

- Development Kit:      LOLIN32
- Module or chip used:  ESP32
- IDF version: master-73db142, v4.2
- Build System:         idf.py
- Compiler version:  xtensa-esp32-elf-gcc (crosstool-NG esp-2020r3) 8.4.0
- Operating System:     Windows
- (Windows only) environment type: Plain Command Prompt
- Using an IDE?: vscode, cli
- Power Supply:         USB

## Problem Description

When using `xEventGroupWaitBits` in the task, that has WDT enabled, and the timeout is higher than WDT timeout, WDT is triggered. That is, unlike `vTaskDelay`, it does not feed a watchdog. 

Ideally, this should be fixed, but if it can't be, it should be at least explicitly stated in the function documentation. 

_Note that it might be encountered even in nested component code, and since it is a combination of timeout and sdkconfig, it makes it hard to predict. Documenting it instead of solving it should be therefore last resort._

### Expected Behavior

`xEventGroupWaitBits` function will behave same as `vTaskDelay`, feeding watchdog timer.

### Actual Behavior

`xEventGroupWaitBits` triggers WDT.

### Steps to reproduce

1. Enable WDT via `esp_task_wdt_add` for given task
2. Call `xEventGroupWaitBits` with timeout higher than `CONFIG_ESP_TASK_WDT_TIMEOUT_S`

### Code to reproduce this issue

```cpp
#include <esp_log.h>
#include <freertos/FreeRTOS.h>
#include <freertos/event_groups.h>
#include <esp_task_wdt.h>

void app_main()
{
  esp_err_t err = esp_task_wdt_add(NULL);
  ESP_LOGI("main", "esp_task_wdt_add: %d %s", err, esp_err_to_name(err));

  EventGroupHandle_t event_group = xEventGroupCreate();
  xEventGroupWaitBits(event_group, BIT0, pdFALSE, pdTRUE, portMAX_DELAY);
}

```
Buildable version is here https://github.com/mdvorak/esp-event-group-wait-wdt

## Debug Logs

```
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:7056
load:0x40078000,len:14336
load:0x40080400,len:3672
0x40080400: _init at ??:?

entry 0x40080678
I (27) boot: ESP-IDF v4.4-dev-4-g73db14240 2nd stage bootloader
I (27) boot: compile time 12:04:31
I (27) boot: chip revision: 1
I (31) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (47) boot.esp32: SPI Speed      : 40MHz
I (47) boot.esp32: SPI Mode       : DIO
I (47) boot.esp32: SPI Flash Size : 4MB
I (52) boot: Enabling RNG early entropy source...
I (57) boot: Partition Table:
I (61) boot: ## Label            Usage          Type ST Offset   Length
I (68) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (75) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (83) boot:  2 factory          factory app      00 00 00010000 00100000
I (90) boot: End of partition table
I (94) boot_comm: chip revision: 1, min. application chip revision: 0
I (102) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=05ea4h ( 24228) map
I (119) esp_image: segment 1: paddr=00015ecc vaddr=3ffb0000 size=0289ch ( 10396) load
I (123) esp_image: segment 2: paddr=00018770 vaddr=40080000 size=078a8h ( 30888) load
I (139) esp_image: segment 3: paddr=00020020 vaddr=400d0020 size=13278h ( 78456) map
I (167) esp_image: segment 4: paddr=000332a0 vaddr=400878a8 size=03494h ( 13460) load
I (179) boot: Loaded app from partition at offset 0x10000
I (179) boot: Disabling RNG early entropy source...
I (191) cpu_start: Pro cpu up.
I (191) cpu_start: Starting app cpu, entry point is 0x40080f4c
0x40080f4c: call_start_cpu1 at C:/Users/mikee/.espressif/esp-idf-master/components/esp_system/port/cpu_start.c:143
I (0) cpu_start: App cpu up.
I (205) cpu_start: Pro cpu start user code
I (205) cpu_start: cpu freq: 160000000
I (205) cpu_start: Application information:
I (210) cpu_start: Project name:     esp-event-group-wait-wdt
I (216) cpu_start: App version:      5c96a00
I (221) cpu_start: Compile time:     Feb 15 2021 12:04:25
I (227) cpu_start: ELF file SHA256:  e50fbdbe6c939199...
I (233) cpu_start: ESP-IDF:          v4.4-dev-4-g73db14240
I (240) heap_init: Initializing. RAM available for dynamic allocation:
I (247) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (253) heap_init: At 3FFB30C8 len 0002CF38 (179 KiB): DRAM
I (259) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (265) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (272) heap_init: At 4008AD3C len 000152C4 (84 KiB): IRAM
I (279) spi_flash: detected chip: winbond
I (283) spi_flash: flash io: dio
I (288) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (297) main: esp_task_wdt_add: 0 ESP_OK
I (297) main: esp_task_wdt_status: 0 ESP_OK
E (5307) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (5307) task_wdt:  - main (CPU 0)
E (5307) task_wdt: Tasks currently running:
E (5307) task_wdt: CPU 0: IDLE
E (5307) task_wdt: CPU 1: IDLE
E (5307) task_wdt: Print CPU 0 (current core) backtrace

Backtrace:0x400D5368:0x3FFB07E0 0x40082289:0x3FFB0800 0x400E2AEB:0x3FFB64B0 0x400D391B:0x3FFB64D0 0x40086242:0x3FFB64F0 0x40087B1D:0x3FFB6510
0x400d5368: task_wdt_isr at C:/Users/mikee/.espressif/esp-idf-master/components/esp_common/src/task_wdt.c:189
0x40082289: _xt_lowint1 at C:/Users/mikee/.espressif/esp-idf-master/components/freertos/port/xtensa/xtensa_vectors.S:1105
0x400e2aeb: cpu_ll_waiti at C:/Users/mikee/.espressif/esp-idf-master/components/hal/esp32/include/hal/cpu_ll.h:183
 (inlined by) esp_pm_impl_waiti at C:/Users/mikee/.espressif/esp-idf-master/components/esp_pm/pm_impl.c:778
0x400d391b: esp_vApplicationIdleHook at C:/Users/mikee/.espressif/esp-idf-master/components/esp_common/src/freertos_hooks.c:63
0x40086242: prvIdleTask at C:/Users/mikee/.espressif/esp-idf-master/components/freertos/tasks.c:3835 (discriminator 1)
0x40087b1d: vPortTaskWrapper at C:/Users/mikee/.espressif/esp-idf-master/components/freertos/port/xtensa/port.c:168

E (5307) task_wdt: Print CPU 1 backtrace

Backtrace:0x400839F5:0x3FFB0DE0 0x40082289:0x3FFB0E00 0x4000BFED:0x3FFB6BB0 0x40087D7E:0x3FFB6BC0 0x400D55AC:0x3FFB6BE0 0x400D55B7:0x3FFB6C10 0x400D390D:0x3FFB6C30 0x40086242:0x3FFB6C50 0x40087B1D:0x3FFB6C70
0x400839f5: esp_crosscore_isr at C:/Users/mikee/.espressif/esp-idf-master/components/esp32/crosscore_int.c:77
0x40082289: _xt_lowint1 at C:/Users/mikee/.espressif/esp-idf-master/components/freertos/port/xtensa/xtensa_vectors.S:1105
0x40087d7e: vPortExitCritical at C:/Users/mikee/.espressif/esp-idf-master/components/freertos/port/xtensa/port.c:469
0x400d55ac: esp_task_wdt_reset at C:/Users/mikee/.espressif/esp-idf-master/components/esp_common/src/task_wdt.c:336
0x400d55b7: idle_hook_cb at C:/Users/mikee/.espressif/esp-idf-master/components/esp_common/src/task_wdt.c:88
0x400d390d: esp_vApplicationIdleHook at C:/Users/mikee/.espressif/esp-idf-master/components/esp_common/src/freertos_hooks.c:51 (discriminator 1)
0x40086242: prvIdleTask at C:/Users/mikee/.espressif/esp-idf-master/components/freertos/tasks.c:3835 (discriminator 1)
0x40087b1d: vPortTaskWrapper at C:/Users/mikee/.espressif/esp-idf-master/components/freertos/port/xtensa/port.c:168
```

abobija, 2021-02-15:
Hi.

This behaviour looks ok for me, because `xEventGroupWaitBits` function blocks the code while the bits are not set.
And you set endless timeout `portMAX_DELAY` but your  BIT0 will never be set. Which means that your main task will go to that line and stay there forever, that is why whatchdog is triggered, and in my opinion it should be triggered in that case.

I suggest you to put  `xEventGroupWaitBits` in while loop, and set some reasonable timeout, for example 1 sec.

igrr, 2021-02-15:
@mdvorak maybe the confusion you are facing is related to the task that is being watched by task watchdog. 
vTaskDelay function yields to lower priority tasks. So if the task watchdog is enabled for the Idle task (default behavior), then yielding from main will let the idle task run and feed its watchdog.

In the example you have posted, the task watchdog is added to the `main` task. Yielding from `main` doesn't feed `main` task watchdog, nor does `vTaskDelay` do it. If you register task watchdog to watch one of your application tasks, you need to explicitly call `esp_task_wdt_reset()` function from that specific task.

mdvorak, 2021-02-15:
My apologies, as usual the problem is between chair and computer.
For some reason, I had a notion, that `vTaskDelay` resets TWDT on its own. But that is not the case, dunno why I did not test that first.
With default idle task WDT, both works the same (both puts CPU on IDLE).

Thanks for both fast replies.

PS: Original example provided was too simplified, of course I had separate task where I used it. But that does not matter, problem was elsewhere.

closed 2021-02-15 by mdvorak